esphome/feature-requests #651: Binary sensor: Motion detected by camera

antonio-fiol, 2020-03-25:
**Describe the problem you have/What new integration you would like**

I would like to be able to act (in HA) upon anything moving in front of the camera.

**Please describe your use case for this integration and alternatives you've tried:**

I have checked the ESPHome and HA documentation and did not find anything that helps: the camera implementation in HA is really limited, and besides I don't think this belongs outside the ESP32.
I have seen some people exploring the concept: https://eloquentarduino.github.io/2020/01/motion-detection-with-esp32-cam-only-arduino-version/
I have seen someone comment on the binary sensor being there: https://www.reddit.com/r/homeassistant/comments/bf1zrw/my_experience_with_esphome_on_home_assistant/elfrfrk/
...but I can't find any way to enable such a binary sensor neither in ESPHome or HA.

**Additional context**

I am aware that other (advanced) processing features such as face detection or QR reading have been discarded in #341 , but firstly motion detection was not covered in that request, and secondly motion detection seems (naively) a less complex problem to deal with, and one that would justify (to avoid using bandwidth when not needed) the local processing.

Technical challenge: Would it be possible to do motion detection on the encoded MJPEG stream without fully decoding it?

sergeolkhovik, 2020-03-25:
I checked the first link - looks really interesting. I tried to build custom solution with ESP32 and PIR sensor, and it was really unstable. PIR generated many false events, so I decided to postpone that idea.

antonio-fiol, 2020-03-26:
Oh, I was definitely not talking about using a PIR, but just using the
image data. PIR requires extra hardware and it seems that with the image
alone it would be doable.

sergeolkhovik, 2020-03-26:
Exactly, with PIR I didn't get right result, and would like to have exactly single camera implementation.

jeroenterheerdt, 2020-11-24:
has anyone ever looked into this? I found this online https://smartsolutions4home.com/ss4h-sd-smart-doorbell/ and it does everything I need to get rid of my Ring, except for the motion detection...

sokolsok, 2020-12-17:
You can use the camera as a motion sensor. I haven't checked but it should work

fkoteam, 2022-03-24:
it would be great. 
a new algoritm: 
https://eloquentarduino.github.io/2020/05/easier-faster-pure-video-esp32-cam-motion-detection/