esphome/feature-requests #658: A new release

staraxis, 2020-03-30:
**Describe the problem you have/What new integration you would like**

Would love for @OttoWinter to release a new version of esphome
**Please describe your use case for this integration and alternatives you've tried:**

There hasn't been a release in a while.
**Additional context**

Would be happy to help out if @OttoWinter or @glmnet need it.

Misiu, 2020-03-30:
I also would like to see a new version released.
The last one was released on 17 Nov 2019.
ESP Home in one of THE BEST addition for Home Assistant and a must-have for DIY.
More recent releases will surely help to make ESP Home even better and much popular.

glmnet, 2020-03-30:
I believe I cannot do it by myself, 
a) I don't know how
b) I suspect I need permissions (passwords, etc) I don't have.

I can however push to dev, and it is lagging in the dependencies, e.g. Platformio is already 4.3 and ESPHome is using 4.1 

So long it still works I'm not too worried about releasing a new stable release, the dev channel, which you can use as a Home Assistant plugin (hass.io) is always automatically updated with all features, you just need to rebuild the docker for it.

Also up to date docs are in http://next.esphome.io

So, ESPHome releases are more like a picture of dev at some time and branding it, you can always get the latest bits automatically, no need a tag for it.

brandond, 2020-03-30:
I think @OttoWinter set it up to autobuild and push things like the docker images on tag, but I'll admit I haven't confirmed that *everything* works - for example I'm not sure if that includes the hassio addons?

wizu, 2020-04-01:
@glmnet Unfortunatelly there is no autobuild dev docker armv7 image - only stable.

glmnet, 2020-04-01:
> @glmnet Unfortunatelly there is no autobuild dev docker armv7 image - only stable.

I don't know what a "autobuild docker" means.

What does this means for you? I can use latest ESPHome dev on my HA which runs on a RPI 3 B plus.

[image]

wizu, 2020-04-01:
@glmnet I'm not using HASSIO. I use standalone esphome installation via docker-compose on my RPI 4 using https://hub.docker.com/r/esphome/esphome-armv7 and there is no dev tag there.
I know I can build my own container but then I cannot update it by simply pulling new container.

randybb, 2020-04-01:
Have you try to pull a new dev image?
```bash
cube:~ # docker images | grep esphome
esphome/esphome                  dev                 eae7af0cbbd4        2 weeks ago         734MB
cube:~ # docker image pull esphome/esphome:dev
dev: Pulling from esphome/esphome
22e816666fd6: Already exists 
079b6d2a1e53: Already exists 
11048ebae908: Already exists 
c58094023a2e: Already exists 
c571577bda77: Already exists 
c646052a12db: Already exists 
3bfb1c7861ad: Already exists 
9807a8ff20d1: Already exists 
b8b77146804d: Pull complete 
3e99f1c0404d: Pull complete 
Digest: sha256:3b265aa90ad6facb377cff4b8a73b178a7ff4534b153c33bbf63d9d424921196
Status: Downloaded newer image for esphome/esphome:dev
docker.io/esphome/esphome:dev
cube:~ # docker images | grep esphome
esphome/esphome                  dev                 3cfd5af083fd        2 days ago          734MB
esphome/esphome                  <none>              eae7af0cbbd4        2 weeks ago         734MB
```

wizu, 2020-04-01:
@randybb unfortunatelly esphome/espohome won't work on ARM - it results in
`standard_init_linux.go:211: exec user process caused "exec format error"`

There is always a way to run it - I could create a local Dockerfile and build it that way. But it would be great if ARM dev build was up to date.

randybb, 2020-04-01:
Yea, it is for different platform.
When you check [Dockerfile for HASSIO](https://github.com/esphome/hassio/blob/master/esphome-dev/Dockerfile) it just pulls new package from github and installs it via pip3. It should be possible to do it via docker-compose exec, but as you see the old dev image still uses pip2, so it need a gently touch from @OttoWinter

staraxis, 2020-04-06:
Thanks everyone! I enjoy esphome immensely and was just hoping we could get some clarity on the direction. Thanks @glmnet for being active here and on discord. Your expertise is greatly appreciated!

fjufju, 2020-07-22:
Any news about 1.15 release?
Latest is 1.14.5 but it not include new features.
@OttoWinter do you plan to release it soon? Any roadmap? Thank you for great work and for the info 👍

glmnet, 2020-07-22:
@OttoWinter is there any part I could help? Like generating change logs from GH PRs, etc. 

I'm following your bump scripts but a guide will be needed too

closed 2020-09-24 by glmnet