esphome/feature-requests #659: Sensors - make correction of values possible

klatvo, 2020-03-30:
**Describe the problem you have/What new integration you would like**

When using cheap sensors the measured values are not always correct. Right now there is no option to add a corrected value (reference point) to the system. Please add that.

**Please describe your use case for this integration and alternatives you've tried:**

E.g. Tasmota provides an option to add a reference point. E.g. you install a sensor (power measurement CSE7766 Power Sensor), than you measure the real voltage with a extra voltmeter. This corrected value you write with "voltageset" to the system and the cheap CSE7766 values get corrected. It would be VERY helpful to have the option to add a "offset" between measured and real value to the ESPHome sensor declarations.

**Additional context**

ESPHome for ever! Such a cool product!

glmnet, 2020-03-30:
I must ask before going on this thread, doesn't [this](https://esphome.io/components/sensor/index.html#offset-multiply) do what you need here? then why?

klatvo, 2020-03-30:
> I must ask before going on this thread, doesn't this do what you need here? then why?

That's exactly what I was looking for and what I couldn't find. Thanks for the hint!!!

closed 2020-03-30 by klatvo

brandond, 2020-03-30:
There are also the calibration filters that are specifically meant for this sort of thing:
https://esphome.io/components/sensor/index.html#calibrate-linear