esphome/feature-requests #661: Listen light color updates even if light is off

iBobik, 2020-03-31:
**Describe the problem you have/What new integration you would like**

ssieb, 2020-04-05:
I know where this came from so I'll provide more of an explanation.
When a light is turned off, color changes don't take effect, so when the light is turned back on, the color is still the same as before.  I haven't tested this myself, so I don't know if HA is even sending updates, so I don't know if it's something that can be fixed on the esphome side.
@iBobik can you check the logs on the device to see if it's even receiving updates from that HA plugin?

iBobik, 2020-04-05:
Thank you for more info.

Yes, I saw in logs of esphome incoming updates.

nagyrobi, 2022-07-07:
Closed as this is by design in HA.

closed 2022-07-07 by nagyrobi