esphome/feature-requests #663: Option for RGBWW light to switch between WW and RGB leds depending on color

TECHNOFAB11, 2020-04-01:
**Describe the problem you have/What new integration you would like**
RGBWW light should have an option to merge color and white data. What i mean is that when i turn it to white, it should use the white leds. If i change the color to lets say green, it should turn off the white leds and turn on the green part of the rgb leds. So with this option there always should be only one of the two (color and white) on. 

**Please describe your use case for this integration and alternatives you've tried:**
Alternative: Tasmota. I dont have any white brightness slider there which means when i change the color to white in HA it turns on the white leds. If i change to green it turns the white off and the colored on. But i had various issues with Tasmota, the biggest one being MQTT, so i would love to switch to ESPHome

**Additional context**
If nobody wants to do it or so, i would help to code it, but i dont understand the esphome framework yet like where the config gets put in the C code

wizu, 2020-04-02:
@TECHNOFAB11 there is something like this already merged in dev https://github.com/esphome/esphome/pull/967 but it turns RGB to 100% if you change WW so it isn't ideal.

GauthamVarmaK, 2020-04-15:
The current handling of ESPHome's RGBWW bulbs has many drawbacks:

1. They are people who reported their bulb burnt when using both white and RGB channels on full brightness.
2. It doesn't really work well with voice/other integrations like google assistant and Alexa.
3. It also adds a little extra complexity curve to using the controls in Home Assistant with different sliders for both white and RGB. In fact, few users split the bulb into 2 bulbs (one rgb bulb and one CCT bulb).

I feel ESPHome should switch to the way Tasmota handles RGBWW bulbs which works well with Home Assistant, google assistant, Alexa and all. One key difference how Tasmota and ESPHome RGBWW bulbs work in Home Assistant is Tasmota allows sending only either RGB colour or WW colour data in light.turn_on service call. while ESPHome simply executes both without throwing an exception.

If this is fixed it will be really helpful for us to use ESPHome with RGBWW bulbs a lot more effectively.

pinkeen, 2020-08-24:
Hello all! 

Actually I've almost finished slight refactoring of the light core which fixes many issues (I think all RGBW bugs) with the current dev version and adds new features:
- Proper handling of configurable color/white interlock (currently it's imperfect IMO) 
- Ability to mix white into color with RGBW(W) to improve color spectrum of desaturated bulbs with option to keep constant brightness (which roughly equals power draw). If color interlock is disabled in this mode then white slider adds extra white bias in this mode. 
- Ability to control color temperature with RGBW (optional) having only RGBW channel which works very well if white led is warm and light is properly mixed/diffuses

Expect a relatively large PR shortly.

ryanjohnsontv, 2020-09-03:
> Hello all!
> 
> Actually I've almost finished slight refactoring of the light core which fixes many issues (I think all RGBW bugs) with the current dev version and adds new features:
> 
> * Proper handling of configurable color/white interlock (currently it's imperfect IMO)
> * Ability to mix white into color with RGBW(W) to improve color spectrum of desaturated colors (the less saturated the color the stronger the effect) with option to keep constant brightness (which roughly equals power draw). If color interlock is enabled in this mode then white slider adds extra white bias.
> * Optional color temperature adjustment for RGBW (only one white channel) which works very well if white led is warm and light is properly mixed/diffused
> 
> Expect a relatively large PR shortly.

If you need any assistance testing, I have a wide mix of RGB, RGBW, and RGBWW bulbs that would benefit from your work!

Tuckie, 2022-01-22:
@pinkeen any chance you could share your code for this? This is exactly the problem I'm trying to solve right now!

> * Ability to mix white into color with RGBW(W) to improve color spectrum of desaturated colors (the less saturated the color the stronger the effect) with option to keep constant brightness (which [...]

nagyrobi, 2022-07-01:
Fixed in https://github.com/esphome/esphome/pull/2012

closed 2022-07-01 by nagyrobi