esphome/feature-requests #664: Time-based Cover with tilt function

InToSSH, 2020-04-01:
**Describe the problem you have/What new integration you would like**
I have tilting venetian exterior blinds, it is the kind with one motor, has built in endstops. When they move up, they first tilt open, then start moving up, when they move down, they first tilt close, then start moving down. It would be nice to be able to set not only open_duration and close_duration, but also tilt_close_duration and tilt_open_duration. In my case this is around 0.9sec. This time is included in the total close/open time, since when they are fully open, they first need close before moving down, and vice versa, when they are fully down, when moving up, they first need to open, then they start moving. 

**Example**
- close_duration set to 60sec
- tilt_open/close_duration set to 1sec
- I command to blind to close to 50%
- Blind starts moving down, counts to 30sec, then stops. 30s is saved as current position
- I set the tilt to open, motor starts moving the blind up for 1sec, but also subtracting the position from 30sec, so now only 29sec left for the position.

The tilt could really be stateless, either close or open, perhaps also 45deg.
I know there is a problem with synchronizing the states of tilt, because it would have to set the tilt state to closed when the blind was previously moving down, and set the tilt state to open if the blind was previously moving up for longer than tilt_open/close_duration, so the tilt could be stateless and it would just run the motor down for 1sec to close then, in worst case it would move the blind down slightly if it the tilt was previously closed, but counting this to the overall position time, so it wouldn't lose the track of position.

**Please describe your use case for this integration and alternatives you've tried:**
For tilting blinds with single motor with up/down direction.
I have tried setting up internal time_based cover, just to keep track of the position based on time, then setting up template cover, manually counting the time for the tilt, and this template cover was controlling the internal time based cover, this sorta worked, but is a bit clumsy.

**Additional context**
The blind uses Somfy WT motor, which has internal position tracking and has endpoints set.
Thank you

kvn1351, 2020-04-24:
This is exactly what I'm trying to accomplish as well!

What do you think about it @OttoWinter? It it possible to do this via the API?

InToSSH, 2020-04-24:
@kvn1351 I actually managed to do this using combination of template cover and time-based cover after not seeing a response on this topic. So, it is possible to do, but it is a bit too complicated and having a tilt directly on the time-based cover would be much easier.
I just didn't bother with actually calculating the current tilt position, instead I am resynchronizing the tilt position each time, since it takes 1s to fully close/open, it might in some cases move the blind up or down a little, but it doesn't bother me.

If you would like to use my code, [I posted it in the community](https://community.home-assistant.io/t/project-controlling-exterior-venetian-blind-with-tilt-with-shelly2-5/184531)

kvn1351, 2020-04-25:
> @kvn1351 I actually managed to do this using combination of template cover and time-based cover after not seeing a response on this topic. So, it is possible to do, but it is a bit too complicated [...]

Thanks a lot. It would be nice if they added better support for this within ESPHome.

kvn1351, 2020-05-14:
> @kvn1351 I actually managed to do this using combination of template cover and time-based cover after not seeing a response on this topic. So, it is possible to do, but it is a bit too complicated [...]

I have tested your config. Does it happen to you too that the tilt action gets activated twice sometimes?

muraj238, 2020-11-18:
Doesn't have the [Cover component](https://esphome.io/components/cover/index.html) the tilt setting? I thought the Time-based Cover is derived. I plan to use 'dumb' motors as well and I really need the tilt function.

InToSSH, 2020-11-18:
@muraj238 no, time-based doesn't have tilt control, this is why I used Template cover, which gets position synced with the time-based cover and then allows me to use tilt aswell, which I scripted myself using lambda.
I guess if you want to use it with 'dumb' motors you can use my script, which is posted in the link above, or in my gists.
Just be aware that I am using Shelly2.5 with power monitoring, which is used to determine if the motor has already reached an endpoint based on the decrease of power draw (my motors have virtual endstops built-in). I use this to synchronize the time-based position in case of power loss for example.

muraj238, 2020-11-18:
@InToSSH Thanks. I've already looked at it, but it's a little bit complex for my tasting. :)

muraj238, 2020-11-20:
@InToSSH So I implemented it myself, but I's not really any less complex than your solution. It supports the tilt, as well as the position to be set from the gui. So far it seems working correctly, at least at my desk. :)

[image]
https://hastebin.com/wumegerofa.yaml

florok2005, 2021-02-11:
> @InToSSH So I implemented it myself, but I's not really any less complex than your solution. It supports the tilt, as well as the position to be set from the gui. So far it seems working correctly, on least at my desk. :)
> 
> [image]
> https://hastebin.com/wumegerofa.yaml

Hello,

could you please share your code. The link you have provided, does not contain it.

btw. aj ja som zo Slovenska :)

muraj238, 2021-02-11:
https://wtools.io/paste-code/b36Q

neyc0, 2021-06-10:
Hey there!
Can you  guys please make an example exactly the same, but everything time-based - positioning and tilt?
I don't have shelly2.5's, just basic relays to switch motors up/down.
Time-based positioning is working fine, but the tilt is confusing me

nagyrobi, 2022-07-05:
alternative solution from @muraj238 solution (pasted here for reference):
```yaml
# BASIC
esphome:
  name: rollers
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret esp_wifi_ssid
  password: !secret esp_wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Rollers Fallback Hotspot"
    password: "hn5Yuc6TwEOo"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esp_api_password

ota:
  password: !secret esp_ota_password

globals:
  # a global variable to store the tilt
  - id: global_roller_1_tilt
    type: float
    restore_value: true
    initial_value: '0'

  - id: global_roller_1_tilt_request
    type: float
    restore_value: true
    initial_value: '0'

  # a global variable to store the absolute position
  - id: global_roller_1_pos
    type: float
    restore_value: true
    initial_value: '0'

  - id: global_roller_1_pos_request
    type: float
    restore_value: true
    initial_value: '0'

# switches to operate the blinds
switch:
  - platform: gpio
    pin: 13
    id: roller_1_up_switch
    interlock: [roller_1_down_switch]
    restore_mode: always off

  - platform: gpio
    pin: 12
    id: roller_1_down_switch
    interlock: [roller_1_up_switch]
    restore_mode: always off

# maintain the global variables with tilt and position
script:
  - id: roller_1_pos
    mode: restart
    then:
      - if:
          condition:
            - lambda: return id(global_roller_1_pos_request) > id(global_roller_1_pos) + 0.01;
          then:
            - switch.turn_on: roller_1_up_switch
            - while:
                condition:
                  - lambda: return id(global_roller_1_tilt) + 0.01 < 1;
                then:
                  - lambda: id(global_roller_1_tilt) += 0.01;
                  - delay: 1ms
            - while:
                condition:
                  - lambda: return id(global_roller_1_pos) + 0.01 < id(global_roller_1_pos_request);
                then:
                  - lambda: id(global_roller_1_pos) += 0.01;
                  - delay: 1ms
            - switch.turn_off: roller_1_up_switch
          else:
            - if:
                condition:
                  - lambda: return id(global_roller_1_pos_request) < id(global_roller_1_pos) - 0.01;
                then:
                  - switch.turn_on: roller_1_down_switch
                  - while:
                      condition:
                        - lambda: return id(global_roller_1_tilt) - 0.01 > 0;
                      then:
                        - lambda: id(global_roller_1_tilt) -= 0.01;
                        - delay: 1ms
                  - while:
                      condition:
                        - lambda: return id(global_roller_1_pos) - 0.01 > id(global_roller_1_pos_request);
                      then:
                        - lambda: id(global_roller_1_pos) -= 0.01;
                        - delay: 1ms
                  - switch.turn_off: roller_1_down_switch
      - lambda: ESP_LOGD("main", "Roller 1 moved to position %f", id(global_roller_1_pos));

  - id: roller_1_tilt
    mode: restart
    then:
      - if:
          condition:
            - lambda: return id(global_roller_1_tilt_request) > id(global_roller_1_tilt) + 0.01;
          then:
            - switch.turn_on: roller_1_up_switch
            - while:
                condition:
                  - lambda: return id(global_roller_1_tilt) + 0.01 < id(global_roller_1_tilt_request);
                then:
                  - lambda: id(global_roller_1_tilt) += 0.01;
                  - delay: 1ms
            - switch.turn_off: roller_1_up_switch
          else:
            - if:
                condition:
                  - lambda: return id(global_roller_1_tilt_request) < id(global_roller_1_tilt) - 0.01;
                then:
                  - switch.turn_on: roller_1_down_switch
                  - while:
                      condition:
                        - lambda: return id(global_roller_1_tilt) - 0.01 > id(global_roller_1_tilt_request);
                      then:
                        - lambda: id(global_roller_1_tilt) -= 0.01;
                        - delay: 1ms
                  - switch.turn_off: roller_1_down_switch
      - lambda: ESP_LOGD("main", "Roller 1 tilted to position %f", id(global_roller_1_tilt));

cover:
  - platform: template
    name: "Roller 1"
    id: roller_1
    has_position: true
    assumed_state: true
    lambda: return id(global_roller_1_pos);
    tilt_lambda: return id(global_roller_1_tilt);
    open_action:
      - script.stop: roller_1_pos
      - lambda: id(global_roller_1_pos_request) = 1;
      - script.execute: roller_1_pos
    close_action:
      - script.stop: roller_1_pos
      - lambda: id(global_roller_1_pos_request) = 0;
      - script.execute: roller_1_pos
    stop_action:
      - script.stop: roller_1_pos
      - switch.turn_off: roller_1_up_switch
      - switch.turn_off: roller_1_down_switch
    position_action:
      - script.stop: roller_1_pos
      - lambda: id(global_roller_1_pos_request) = pos;
      - lambda: ESP_LOGD("main", "Roller 1 setting position to %f", pos);
      - script.execute: roller_1_pos
    tilt_action:
      - script.stop: roller_1_pos
      - lambda: id(global_roller_1_tilt_request) = tilt;
      - lambda: ESP_LOGD("main", "Roller 1 setting tilt to %f", tilt);
      - script.execute: roller_1_tilt
```

ssieb, 2023-12-28:
This is still a valid feature request, even if there is a workaround, especially when the workaround is so ugly.