esphome/feature-requests #667: Add rate of change filter (or sensor)

petersandersen, 2020-04-03:
**Describe the problem you have/What new integration you would like**

Ability to convert raw values to the rate of change over some window size. I.e. current value minus last value or an average of previous x values.

**Please describe your use case for this integration and alternatives you've tried:**

I've made a bed occupancy sensor using a load cell but there is a lot of drift even after calibration. Since I don't really need the exact weight I want to count the bed as occupied if either of these conditions have been met in the last 5 minutes: 1) Current weight is above the average weight in the last 24 hours or 2) the absolute value of the current value minus the last value is above 1kg. 

The automation rules can be handled in home assist but I need to be able to create a rate of change filter (or sensor? i.e. it could be a copy of an existing sensor with filters to facilitate multiple values) in esphome.

**Additional context**

Thanks again for creating esphome, its been an awesome piece of software to improve my home automation and tinker with electronics and components I'd never have otherwise played with.

rudshep, 2020-04-13:
I'm looking for something similar but for energy monitoring. Detect if energy consumption increased by between 340 - 360 watts over 3 seconds which could be an indicator that a certain appliance has started up.

I don't know how to solve it either but let's exchange a few ideas...

- Use the sliding_window_moving_average to smooth out the absolute reading over 5 minutes.
- Set the delta filter to, say, 10kg to detect big changes. 1kg might be triggered by a dog or movement.

Thats still not your "Rate of Change Sensor" though. Home Assistant has a Trend integration but I don't think there is an equivalent in ESPHome. https://www.home-assistant.io/integrations/trend/

petersandersen, 2020-04-20:
Thanks for the reply @rudshep, it looks like the existing lambda functionality can solve both our issues.

For my hx711 based bed presence sensor I'm now using the following code:
`sensor:
  - platform: hx711
    name: "Bed HX711 Value"
    id: bed_weight
    dout_pin: GPIO1
    clk_pin: GPIO3
    gain: 128
    update_interval: 1s
    filters:
      - lambda: |-
          static int last_value = 0;
          static int bed_change = 0;
          bed_change =  x - last_value;
          last_value = x;
          return abs(bed_change);
      - sliding_window_moving_average:
          window_size: 60
          send_every: 10
`
This returns the average absolute change in the raw sensor value over the last minute every 10 seconds. Due to sensor noise it sits around 140+/- 20 but is consistently above 1000 when I'm on the bed. Will report back if its working tomorrow. 

References used ( https://esphome.io/guides/automations.html & https://esphome.io/components/sensor/index.html ). 

Still doesn't solve the issue of creating copies of sensors that each have their own lambdas / filters for different indicators. Please report back how your implementation goes :)

nebulous, 2022-10-21:
I've often wanted for something like the inverse of a median filter - where instead of removing outliers, outliers are detected. Would be useful for a couple things in my HA/ESP projects. I'm thinking the lambda filter idea above could solve that problem by having a median filter, then subtracting the current value from it. Thanks for the tip - maybe this comment will serve to trigger another new component or better idea :)

rcxrbt, 2022-11-30:
> I've often wanted for something like the inverse of a median filter - where instead of removing outliers, outliers are detected. Would be useful for a couple things in my HA/ESP projects. I'm [...]

very interesting!
also me trying to find peaks on sensor output, can you share your code? :)
This is my case:
https://github.com/esphome/feature-requests/issues/1980

nebulous, 2022-11-30:
I went ahead and tried this since this thread prompted it today. Here's a synopsis which may work for your case @rcxrbt - or might cause someone who actually knows what they're doing to be so annoyed they post TheRightWay® to handle this 😄 

There are four components:
- the actual signal sensor (in my test case this was a temperature sensor, but in yours would be particulate)
- a copy of said sensor with a long-term median filter applied which should approximate the steady state
- a delta sensor which returns the difference between the current value and the median
- a binary threshold sensor which turns on or off based on the delta sensor

```yaml
sensor:
  - platform: bme280
    temperature:
      name: "$name BME280 Temperature"
      id: bme280_temperature

  - platform: template
    name: "$name BME280 Temperature median"
    id: bme280_temperature_median
    lambda: |-
      return id(bme280_temperature).state;
    filters:
        - median:
            window_size: 7

  - platform: template
    name: "$name BME280 Temperature signal"
    id: bme280_temperature_signal
    lambda: |-
      float tempnow = id(bme280_temperature).state;
      float tempmed = id(bme280_temperature_median).state;
      return fabs(tempnow - tempmed);

binary_sensor:
  - platform: analog_threshold
    name: "$name rapid change"
    sensor_id: bme280_temperature_signal
    threshold: 1.5

```

rcxrbt, 2022-12-02:
> I went ahead and tried this since this thread prompted it today. Here's a synopsis which may work for your case @rcxrbt - or might cause someone who actually knows what they're doing to be so annoyed they post TheRightWay® to handle this 😄
> 
> There are four components:
> 
> * the actual signal sensor (in my test case this was a temperature sensor, but in yours would be particulate)
> * a copy of said sensor with a long-term median filter applied which should approximate the steady state
> * a delta sensor which returns the difference between the current value and the median
> * a binary threshold sensor which turns on or off based on the delta sensor
> 
> ```yaml
> sensor:
>   - platform: bme280
>     temperature:
>       name: "$name BME280 Temperature"
>       id: bme280_temperature
> 
>   - platform: template
>     name: "$name BME280 Temperature median"
>     id: bme280_temperature_median
>     lambda: |-
>       return id(bme280_temperature).state;
>     filters:
>         - median:
>             window_size: 7
> 
>   - platform: template
>     name: "$name BME280 Temperature signal"
>     id: bme280_temperature_signal
>     lambda: |-
>       float tempnow = id(bme280_temperature).state;
>       float tempmed = id(bme280_temperature_median).state;
>       return fabs(tempnow - tempmed);
> 
> binary_sensor:
>   - platform: analog_threshold
>     name: "$name rapid change"
>     sensor_id: bme280_temperature_signal
>     threshold: 1.5
> ```
> 
> oh, and @petersandersen the above is an example of making a copy of an existing sensor using a template sensor.

NIce!
I will try!!!!

Tnx :)

ssieb, 2022-12-03:
> oh, and @petersandersen the above is an example of making a copy of an existing sensor using a template sensor.

Are you aware of the [copy sensor](https://esphome.io/components/copy.html)?

rcxrbt, 2022-12-03:
It seems working, but need more tuning.
Threshold is = 15 (red line).

[image]

nebulous, 2022-12-03:
>> or might cause someone who actually knows what they're doing to be so annoyed they post TheRightWay® to handle this 😄

> Are you aware of the [copy sensor](https://esphome.io/components/copy.html)?

⬆️ Like that! Thanks @ssieb 

But nice to see you have something working, @rcxrbt. Seems like ultimately what you want is some shorter term signal divided by some longer term signal, with values for each tweaked until your microcontroller is annoyed by smoke at exactly the same times you are.

AJolly, 2025-09-19:
> This has the advantage of preserving the original data from the sensor, since the derivative sensor is a separate entity.

The disadvantage is you have to set it up in HomeAssistant, so not great if you want to roll out multiple devices.  And if you wanted to preserve the original data you could have had two esphome sensors.  

Potentially forces more HA database writes, not sure you can have a derivative sensor while disabling the original sensor.