esphome/feature-requests #668: SGP30: report "baseline" as a sensor as well as it's already done in the log

PieBru, 2020-04-03:
**Describe the problem you have/What new integration you would like**

After evaluating CCS811 vs SGP30, I decided to use the SGP30 for my Portable Air Quality useful toy.
The SGP30 re-calibrates its baseline value frequently, as reported in the ESP_LOGI() but it is reset at every boot.

**Please describe your use case for this integration and alternatives you've tried:**

The quality of the SGP30 returned values greatly depends on a baseline calibration updated as much as possible. The manufacturer suggests every 7 days, but initially (in my experience for the first couple of months or so) it should be updated almost daily.
It would be nice to have the SGP30 baseline value reported as a sensor, so one can periodically save the last value and then restore it after the next boot (which in my case is frequent, as the wifi is weak and the connection is lost frequently).

**Additional context**

As my build can also work offline from my home HASSIO server (i.e. at my sons or nephews home), it would be very nice for me to be able to save it also locally as a `globals: restore_value: yes` if some `save_globals` action could be called (not often!) in some way. Otherwise I will need to bind the Air Quality ESP32 probe with a dedicated RasPI-Zero, only to have a "portable" HassIO server.

PieBru, 2020-04-03:
Forgot to attach a SGP30 log sample.
```
[20:01:58][D][sgp30:237]: Got eCO2=400.0ppm TVOC=0.0ppb
[20:01:58][D][sgp30:131]: External compensation data received: Humidity 38.11%
[20:01:58][D][sgp30:141]: External compensation data received: Temperature 14.91°C
[20:01:58][D][sgp30:150]: Calculated Absolute humidity: 4.851 g/m³ (0x04D9)
[20:01:58][I][sgp30:112]: Current eCO2 baseline: 0x8C64, TVOC baseline: 0x86AC
```

PieBru, 2020-04-08:
As reported by the SGP30 datasheet:

> The SGP30 also provides the possibility to read and write the baseline values of the baseline correction algorithm. This feature is used to save the baseline in regular intervals on an external non-volatile memory and restore it after a new power-up or soft reset of the sensor.

Otherwise one MUST look at the logs to update the baseline, and eventually rebuild the ESPHome firmware for that device.

Thanks,
Piero

nagyrobi, 2022-07-07:
Fixed in https://github.com/esphome/esphome-docs/pull/982

closed 2022-07-07 by nagyrobi