esphome/feature-requests #670: Modbus component for epsolar charge controller

petermh87, 2020-04-07:
**Describe the problem you have/What new integration you would like**

A generic modbus component that I can configure to read the registers of a modbus component and spit into a sensor without having to utilise hassio.

**Please describe your use case for this integration and alternatives you've tried:**

I am trying to read the registers of a ls1024b solar controller (same as an epsolar tracer) using modbus, and using the output of these registers to toggle switches or enable/inhibit functions without having to relay on hassio.

I have tried configuring a custom component to no avail, but have noticed a modbus component I can add into the configuration, but with only a few parameters.

modbus:
  id:
  uart_id:

I have not been able to find any further documentation or guidance on this.

**Additional context**

martgras, 2020-07-30:
Fyi I started implementing this component. 
The code is at https://github.com/martgras/esphome/tree/epever/esphome/components/epever

So far only reading values is implemented and I'm not sure exactly sure how to return status values like the current charging mode for example in a nicer way right now the raw number is returned

martgras, 2020-08-13:
My component is now working quite well for me.
I'm planning to do some clean up and submit a PR. 
I already split the code into a generic modbus component and then derived a Epever senor from it. 
I'm struggling a bit how to completely separate my modbus component from the epever specialization so that modbus component can be reused for other devices. 

Also it would be great to get a bit of feedback from anyone using a epever device 
A very basic doc is at
https://github.com/martgras/esphome/blob/epever/esphome/components/epsolar/readme.md

petermh87, 2020-09-19:
Hi,

Thanks for your hard work on this so far.

I am trying to test the epever component with my LS1024B solar controller and can see that the selected named variable appear in my hassio instance once I set up the UART, Modbud, EPever and MQTT components as per your sample config.

However I am currently not pulling down any values. I am not sure whether this is due to the following:-
- I cannot get the rtc_sync to work with my time component being set to homeassistant rather than sntp server. 
- The address of my device is different then your example "0x1" 
-  The LS1024B has a common positive rail, so I am unable to use the ground connection from the RS485 port as this will keep the "load" permanently and if any high current devices are present this could potentially destroy the controller.

Any advice would be greatly appreciated.

Pete

martgras, 2020-09-20:
If you have a different modbus device address configure it with the address property (6 in this example)

sensor:
  - platform: epsolar
    modbus_id: modbus_epsolar
    id: traceran
    ## the Modbus device addr
    address: 0x6

(According to the documentation the default address is 0x1 )

I don't have that device so I can't tell you exactly how to hook it up exactly to rs485 but from I saw there is no difference.
All I can find is from docs https://www.developpez.net/forums/attachments/p196506d1451307310/systemes/autres-systemes/automation/probleme-com-modbus-pl7-pro/controllerprotocolv2.3.pdf/
…
(1)To improve the communication quality, the Gound pins(connected with the
negative terminal of the battery) could be used if necessary. However, the user must
care the common ground problem of the connected devices.

Martin

silicoon84, 2021-02-09:
Any idea how i can get this going with esphome on homeassistant OS? Obviously don't have access to the underlying OS to run pip3 commands etc. Been following your progress (https://github.com/esphome/esphome/pull/1256) but keen to get something working sooner if possible!

I only really need to read, not write. My other option is rebuilding my current arduino codebase and using mqtt, but obviously rather use this! Appreciate any help

petermh87, 2021-02-09:
> Any idea how i can get this going with esphome on homeassistant OS? Obviously don't have access to the underlying OS to run pip3 commands etc. Been following your progress ([esphome/esphome#1256](https://github.com/esphome/esphome/pull/1256)) but keen to get something working sooner if possible!
> 
> I only really need to read, not write. My other option is rebuilding my current arduino codebase and using mqtt, but obviously rather use this! Appreciate any help

See this guide below to run on esphome....you can then perform automations, data logging etc in homeassistant

https://github.com/martgras/esphome/tree/epever/esphome/components/epsolar

Hope this helps

martgras, 2021-02-09:
I assume you have Windows 10.
All you need is python 3  (https://www.python.org/ftp/python/3.8.7/python-3.8.7-amd64.exe) . Make sure to add the python directory to PATH
You can then either download and unpack the zipped sources from https://github.com/martgras/esphome/archive/epever.zip  and skip the git clone command or install git for windows as well

From there the steps from https://github.com/martgras/esphome/tree/epever/esphome/components/epsolar should work

https://github.com/martgras/esphome/tree/epever/esphome/components/epsolar

silicoon84, 2021-02-09:
Sorry, i should have given more detail. 
I've seen the main readme etc, but I'm running home assistant OS image/appliance on raspberry pi, so no access to the OS to run any commands (apart from the hass cli) unless I'm missing something. 
Somewhat new to Hassio and esphome, but because its an appliance, it doesnt look like i'm able to run commands like i would if it was running straight on debian, docker etc.
Its an sd image install, which i just added esphome to, via the supervisor addons.

martgras, 2021-02-11:
Not sure if this is the right place for this discussion so I keep it brief. 
You don't have to use your esphome appliance to build the esphome firmware.  Any Windows or Linux computer can be used 
Feel free to open an issue in my repro if you need more

## Here are the steps on Windows 10

You need a working Python 3 setup on your machine. 
You can either install the package from python.org https://www.python.org/downloads/ or using powershell 

## Steps for powershell

Open Powershell as an administrator: right click Windows Start Button - Windows Powershell (admin) 

Install choco
````
Set-ExecutionPolicy Bypass -Scope Process -Force; [System.Net.ServicePointManager]::SecurityProtocol = [System.Net.ServicePointManager]::SecurityProtocol -bor 3072; iex ((New-Object System.Net.WebClient).DownloadString('https://chocolatey.org/install.ps1'))
````
Install Python and git using choco
````
choco install -y python3
choco install -y git.install
````
close the admin powershell window and open it again
Create directory (name doesn't matter but avoid spaces in the path to make things easier) 

````
mkdir \esphome-dev
cd \esphome-dev
````
Now install the forked version of esphome 

````
git clone https://github.com/martgras/esphome.git
cd esphome/
pip3 install -r requirements.txt -r requirements_test.txt
python -m pip install --upgrade pip
pip3 install -e .
````

Close powershell and open a new powershell window as a standard user
Copy your configuration file to \esphome-dev (in my example solarpanel.yaml)

````
## Build
esphome ..\solarpanel.yaml compile 
## Upload to esp32 
esphome ..\solarpanel.yaml run

petermh87, 2021-02-11:
[image]

Hi,

I also tried this method within home assistant and set the ESPHome add-on repository to "martgras:epever"

The code I have been running with my EPSolar LS1024B solar charge controller validated and complied happily. This might be another option for those wanting ESPHome/Home assistant functionality.

petermh87, 2021-02-11:
[image]

Screenshot of a successful compile

petermh87, 2021-04-26:
This doesn't seem to run correctly in HASSOS anymore, like my previous posts showed.....Do you know if the epsolar branch has been renamed or merged with the dev branch of esphome?

martgras, 2021-04-27:
It hasn't been merged. 
But I have created a generic modbus component that works not only with epever 
This is now the default branch in my repository - probably that causes problems with your add setting
The "old" code is still available in my epever branch (https://github.com/martgras/esphome/tree/epever) but I'm not sure how a branch can be specified in the addon settings

I'm running my new code for some time without problems. 
Apart from being more generic the new approach uses a cleaner configuration using sensors, binary sensors and switches
The config for epever is here: https://github.com/martgras/esphome/blob/modbus_component/esphome/components/modbus_controller/testconfig/epever.yaml

I'm going to submit a new pull request  once I cleaned up the code a bit (but of course no idea if it will be accepted here) 
If you have more questions or comments  feel free to open an issue at https://github.com/martgras/esphome/issues or use the discussions https://github.com/martgras/esphome/discussions/11
Any feedback is very welcome

kitt2011, 2021-07-22:
Everytime i try to compile ESPHOME show:  
  Platform not found: 'sensor.epsolar'. :(

martgras, 2021-10-05:
Have a look at https://github.com/martgras/esphome/blob/testing/esphome/components/modbus_controller/examples/epever/epever-all-in-one.yaml

closed 2022-06-29 by nagyrobi