esphome/feature-requests #673: SGP30: update_interval should be 1s, otherwise raise a log warning message

PieBru, 2020-04-08:
**Describe the problem you have/What new integration you would like**

As per SGP30 datasheet:
"The on-chip baseline compensation algorithm has been optimized for this sampling rate (1 Hz). The sensor shows best performance when used with this sampling rate."

**Please describe your use case for this integration and alternatives you've tried:**

With update_interval 10s the SGP30 shows strange behaviours, such as swap eCO2/tVOC baselines and/or return only one of the two real values.

**Additional context**

The ESPHome documentation may suggest to use median and/or SWMA filters to slow down the update frequency, while obtaining more mediated values.

PieBru, 2020-04-08:
The `INFO` log message should be output only when baseline values updates.
```
[I][sgp30:112]: Current eCO2 baseline: 0x8C65, TVOC baseline: 0x8965
```

Alternatively, that log message should be de-classed to `DEBUG`, otherwise the `INFO` log level will be spammed at 1 Hz rate.

PieBru, 2020-04-08:
See also #668

nagyrobi, 2022-07-07:
Fixed in https://github.com/esphome/esphome-docs/pull/982

closed 2022-07-07 by nagyrobi