esphome/feature-requests #677: Support for Remote Receiver raw/codec value sensor

dortamur, 2020-04-11:
**Describe the problem you have/What new integration you would like**

Rather than having to add a `binary_sensor` for every possible code, I would like to have a normal `sensor` that passes on the value received. 
It could still be limited to a platform, and for those supported, could optionally be limited to an address or similar.

For example, a single Sensor could be:
```
sensor:
  - platform: remote_receiver
    name: "Panasonic Remote Input"
    panasonic:
      address: 0x4004
```
...and the value returned by the sensor would be the `command`.

**Please describe your use case for this integration and alternatives you've tried:**

This would make it easier to handle automation logic in a platform such as Node Red using MQTT.
Instead of having to subscribe to every single IR code via MQTT `binary_sensor`, I could subscribe to a single MQTT `sensor`, and then handle the branching logic based on the value within Node Red.

This also has the advantage that adding support for new codes would not require reflashing the esphome device.

**Additional context**

dortamur, 2020-04-11:
I've found another alternative that does what I want, which uses the automation triggers, and doesn't require the `binary_sensor` definition:

```yaml
text_sensor:
  - platform: template
    name: "${device_title} Remote - Sony"
    id: remote_sony
    state_topic: esphome/${device_name}/ir/sony

remote_receiver:
  dump: all
  pin: 
    number: D7
    inverted: True

  on_sony:
    then:
      - text_sensor.template.publish:
          id: remote_sony
          state: !lambda |-
            char temp[20];
            sprintf(temp, "%d", x.data);
            ESP_LOGI("main", "Sony: %s", temp);
            return temp;
```

While it would be cleaner to have more direct sensor support, this does what I want for now.
Maybe it would be useful to include an example like this on the Remote Receiver page?

glmnet, 2020-04-15:
> This method also has the problem that the sensor value stays at the last received value, rather than defaulting to blank after a short delay.

you can add it right away!
````yaml

  on_sony:
    then:
      - text_sensor.template.publish:
          id: remote_sony
          state: !lambda |-
            char temp[20];
            sprintf(temp, "%d", x.data);
            ESP_LOGI("main", "Sony: %s", temp);
            return temp;
      - delay: 5s
      - text_sensor.template.publish:
          id: remote_sony
          state: ''
````

> Maybe it would be useful to include an example like this on the Remote Receiver page?

indeed, ESPHome does a lot in yaml but c++ lambdas allows you to do what is not supported in yaml directly, many times making yaml very confusing, this case you are sending the remote code in decimal format, but you can easily modify that code to use hex, or add a counter in the front, or use a numeric sensor instead of a text sensor, this way possibilities are endless

closed 2020-04-15 by glmnet