esphome/feature-requests #678: Dallas auto-discovery and setup sensors

krahabb, 2020-04-13:
**Describe the problem you have/What new integration you would like**

I would like to add the auto-discovery and setup of dallas sensors to the appropriate component api. Now auto-discovery is already implemented and you can see the log output during device initialization. After collecting sensor addresses you can follow up adding those id/addresses to the yaml node config and rebuild/reflash again. I would like to avoid explicitly setting sensor addresses in config file and let the esphome automatically discover and setup those when it starts

**Please describe your use case for this integration and alternatives you've tried:**

This could be useful if you want to skip entirely the 2 steps firmware compile/upload and also to allow connecting new hardware sensors to the same bus without the need to recompile every time

**Additional context**

This feature is really simple to setup (I've already modified the 'dallas.cpp' code to allow this) and looks innocent enough to not disrupt existing functionalities. We could still leave the actual option to explicitly set sensor addresses in *.yaml by leaving everything as is and just adding some options under the 'dallas:' hub configuration entry. Also, the name of the discovered sensor could be templatized based on 'device_name' and/or sensor address in order to guarantee uniqueness in ha/mqtt environment when sensors are moved between devices

I'm available to setup my dev environment and submit a PR on this (this would me my first on github so it could take some time).

glmnet, 2020-04-15:
so you have e.g. 

````
dallas:
  pin: xx
  auto_discovery: true
  device_name: esp01
````

then you'll have in ha e.g. 
* sensor.esp01_temperature_94823749287439123487123
* sensor.esp01_temperature_92874912084712304123411

krahabb, 2020-04-15:
Hello,

Exactly that. I’ve also provided under ‘dallas:’ the same options as for DallasTemperatureSensor (more or less) in order to properly configure every other property (resolution, icon, etc)

 

Also, the ‘device_name’ would be a format string in my idea in order to provide some customization:

 

Device_name: ‘%s_temperature_%s’

 

Where the parameters pushed to the proper snprintf  would be the espnode device name and the sensor address

 

Davide

glmnet, 2020-04-15:
It's ok for me, so long it keeps everything working as before too, I think this PR will be welcomed

krahabb, 2020-04-15:
Ok, give me a couple of days to do more testing and figure out how to publish it

 

Davide

nagyrobi, 2022-07-05:
Impemented, but reverted later: https://github.com/esphome/esphome/pull/1189

closed 2022-07-05 by nagyrobi