esphome/feature-requests #679: Support RF + IR Remote Receivers on one board

rudshep, 2020-04-13:
**Describe the problem you have/What new integration you would like**
I have a custom ESP8266 board with an IR receiver as well as an RF receiver.
ESPHome only allows one instance of the remotereceiver component at a time and gives a duplicate error if you try to define multiple.

**Please describe your use case for this integration and alternatives you've tried:**

I have a diy hub in the living room that can respond to IR codes for the LED strip, IR codes for the entertainment system and RF codes for motion sensors in the area.

The IR receiver and RF receiver are on different pins.
I will try and combine them on one physical pin and see what happens but it is not ideal because the one is inverted and the other one isn't.

To further complicate matters one IR remote is 38Khz and another is 40kHz and ideally there should be multiple receivers.

**Additional context**

It would be ideal to have multiple instances of the Remote Receiver each with its own id and on its own pin.

glmnet, 2020-04-15:
It's indeed supported, try

````yaml
remote_receiver:
  - id: receiver_one
    pin: D5
    dump: all
  - id: receiver_two
    pin: D6
    dump: all

````

closed 2020-04-15 by glmnet

rudshep, 2020-04-17:
@glmnet That worked thanks!

For the sake of noobs like me to follow... this was the mistake I made...

This does **not** work:

```
remote_receiver:
  id: ir_receiver
  pin: 
    number: GPIO14
    inverted: true
  dump: all

  id: rf_receiver
  pin: 
    number: GPIO3
    inverted: true
  dump: all
  # Settings to optimize recognition of RF devices
  tolerance: 50%
  filter: 250us
  idle: 4ms
  buffer_size: 2kb

Result when compiling:
[32mINFO Reading configuration gc_rfirled.yaml...
[31mERROR Error while reading config: Invalid YAML syntax. Please see YAML syntax reference or use an online YAML syntax validator:

duplicate key: "id"
  in "gc_rfirled.yaml", line 542, column 0
```

**This works:** 
Notice the dashes in front of the two id's and the correct indentation. YAML takes some getting used to.

```
remote_receiver:
  - id: gc_rfirled_irrx
    pin: 
      number: GPIO14
      inverted: true
    dump: all

  - id: gc_rfirled_rfrx
    pin: 
      number: GPIO3
      inverted: true
    dump: all
    # Settings to optimize recognition of RF devices
    tolerance: 50%
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
```