esphome/feature-requests #681: [REQUEST] ESPHome firmware on the Sonoff Zigbee Bridge (Sonoff ZBBridge)

Hedda, 2020-04-16:
**Describe the problem you have/What new integration you would like**

Please consider making ESPHome work to the new Sonoff Zigbee Bridge (Sonoff ZBBridge) including using the Zigbee radio to control Zigbee devices

https://www.itead.cc/sonoff-zbbridge.html

Itead has just launched Sonoff ZBBridge as an inexpensive Sonoff Zigbee Bridge which is based on a similar design as the Sonoff RF Bridge 433 (which I believe firmware already has support for?).

https://www.itead.cc/sonoff-zbbridge.html

According to the teardown on notenoughtech.com it sounds as if it is based on Silicon Labs EFR32MG21 (EFR32 Mighty Gecko) for Zigbee 3.0 radio module support and ESP8266 (ESP8266EX) for WiFi and bridge/gateway/controller software

https://notenoughtech.com/home-automation/sonoff/sonoff-zigbee-bridge-preview/

From the images it even looks like they are reusing the same injection moulds as for the Sonoff RF Bridge 433 housing/enclosure 

https://sonoff.tech/product/accessories/433-rf-bridge

https://www.itead.cc/sonoff-rf-bridge-433.html

https://www.itead.cc/wiki/Sonoff_RF_Bridge_433

Tasmota does something similar with Zigbee2Tasmota but by connecting an ESP8266 to a Texas Instrument CC2530 module instead however it too is using a serial communication protocol, see:

https://tasmota.github.io/docs/Zigbee/

EZSP (EmberZNet Serial Protocol) interface that Silicon Labs uses is also well documented and already used by open source projects, see example Home Assistant's ZHA integration component via zigpy and bellows:

https://github.com/zigpy/bellows  

https://www.home-assistant.io/integrations/zha/  

https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2017/05/25/build_an_ezsp-spiho-2VE8

https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

The alternative today is to build your own Zigbee2Tasmota bridge with a ESP8266 and a TI CC2530 module, but that is not a complete hardware solution and the Zigbee module included in Sonoff ZBBridge is much more powerful than TI CC2530, plus the EFR32 based module in the Sonoff ZBBridge supports Zigbee 3.0 while  TI CC253x only supports Zigbee Home Automation (HA) 1.2

https://tasmota.github.io/docs/Zigbee/

Hedda, 2020-04-20:
Please note that #681 and #688 are two sperate requests with different goals.

- Request #681 ask for ESPHome to be a smart Zigbee gateway which translate Zigbee to MQTT.
- Request #688 ask for ESPHome to be a dumb Zigbee gateway which only forwards the serial port.

Hedda, 2022-06-13:
FYI, there are now several ESP32 based gateways which contain either Silicon Labs EFR32MG21 or Texas Instruments CC2652P Zigbee Coordinator radio modules sold and many of them are open-source hardware with schematics for DIY builds. See example:

https://zig-star.com/projects/zigbee-gw-lan/

https://github.com/mercenaruss/zigstar_gateways

https://www.tubeszb.com

https://github.com/tube0013/tube_gateways/

Those have benefits of their makers being active members of the Home Assistant community and their support so far is great!

https://community.home-assistant.io/t/zigstar-zigbee-coordinators-and-routers/338586/

https://community.home-assistant.io/t/tubes-zb-coordinators-and-routers-was-zigbee-router-on-steroids/280896

ITead have also released a new ESP32 based product based ZBBridge-P (a.k.a. "ZB Bridge-P") with TI CC2652P Zigbee radio:

https://itead.cc/product/sonoff-zigbee-bridge-pro/

That "Sonoff Zigbee Bridge Pro" has already been hacked by the Tasmota community which can now use it with Zigbee2Tasmota:

https://github.com/arendst/Tasmota/discussions/14419

https://tasmota.github.io/docs/Zigbee/

Hedda, 2022-09-30:
FYI, Espressif themselves now seems to focus on ESP's one-stop Zigbee Gateway solution, i.e., the two chips solution using an ESP32-S3 Wi-Fi SoC for the Zigbee Gateway/Bridge application part paired together via UART-serial to an ESP32-H 802.15.4 SoC as a dedicated radio chip for Zigbee (or Thread/OpenThead), where a typical use case example would be a Zigbee to Matter Bridge (or Thread/OpenThead to Matter Bridge). 

* https://www.espressif.com/en/news/ESP32_H2

* https://github.com/espressif/esp-idf/tree/master/examples/zigbee/
  * https://github.com/espressif/esp-matter/tree/main/examples/zigbee_bridge
  * https://github.com/espressif/esp-idf/tree/master/examples/zigbee/esp_zigbee_gateway
  * https://github.com/espressif/esp-idf/blob/master/examples/zigbee/esp_zigbee_rcp

and

* https://github.com/espressif/esp-zboss-lib

or

* https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp

and for optionally adding Matter support on top of that see Project-CHIP:

* https://github.com/project-chip/connectedhomeip/tree/master/examples/
  * https://github.com/project-chip/connectedhomeip/tree/master/examples/bridge-app/
  * https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool

Note that even with that type of combination solution with a dedicated 802.15.4 radio chip for Zigbee (or Thread/OpenThead), there is still an option to use an alternative third-party 802.15.4 SoC such as Nordic Semiconductor nRF5/nRF52 (nRF52840), Texas Instruments CC2652/CC1352, or Silicon Labs EFR32MG21/EFR32MG24, instead of the new ESP32-H 802.15.4 SoC.

lachesis, 2023-05-30:
What about the ESP32-C6 RISC-V with built in Zigbee?

Hedda, 2023-06-20:
> What about the ESP32-C6 RISC-V with built in Zigbee?

Check out this separate feature request for ESP32-C6 support (which requires the upcoming ESP-IDF 5.1 to be released) -> https://github.com/esphome/feature-requests/issues/2176

Hedda, 2023-06-20:
> > What about the ESP32-C6 RISC-V with built in Zigbee?
> 
> Check out this separate feature request for ESP32-C6 support (which requires the upcoming ESP-IDF 5.1 to be released) -> #2176
> 
> Also check this other feature request for Zigbee (and Thread) support on ESP32-H2 which has the same 802.15.4 radio -> #1397
> 
> The hurdle with those is they require support for ESP ZBOSS 3.0 binary libraries supporting the Zboss Zigbee 3.0 stack for ESP32.
> 
> Or for Home Assistant's ZHA integration pass-through support need a ZBOSS radio library for zigpy (e.g. zigpy-zboss), check out:
> 
> https://github.com/kardia-as/zigpy-zboss
> 

FYI, that is now a new ESP32-H2 + ESP32-C3 combo development board kit and ESP Thread Boarder Router SDK that can be used as reference hardware. It has as been certified as a Thread Border Router (TBR) / OpenThread Border Router (OTBR) which can be used for Matter based on the Thread 1.3 standard, alternatively be used as an as Zigbee gateway/hub or bridge, and they also got an optional Ethernet Sub-Board

https://www.cnx-software.com/2023/06/20/espressif-esp-thread-border-router-board-combines-esp32-h2-esp32-c3-wireless-chips/

https://github.com/espressif/esp-thread-br

https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/

https://www.aliexpress.us/item/1005005688193617.html

PS: An earlier two-board ESP Thread Border Router solution was previously available as an example project for reference:

https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br

https://openthread.io/guides/border-router/espressif-esp32

[image]

and optional Optional Ethernet Sub Board

[image]