esphome/feature-requests #686: Add state to BinaryOutput

apeeters, 2020-04-19:
**Describe the problem you have/What new integration you would like**

I would like to be able to use the state of a (binary) output in lambda code.

**Please describe your use case for this integration and alternatives you've tried:**

Currently the BinaryOutputs do not expose their state. It would be nice to have the state available for use in lambda code.

**Additional context**

glmnet, 2020-04-25:
It might sound strange but output doesn't have internal state in ESPHome, if we'd going add this feature we will have to  add an extra variable just for this.

An output is just a hardware abstraction, which output you want to know the state?
I mean, is it a switch, is it a light?

If it's a switch, you can access id(switch).state if it's another component, more or less all have a way to retrieve their state

apeeters, 2020-04-25:
The use case is visualising all input and output states on a SSD1306 display, regardless of the exact kind of output.

Thanks for the tip. In my environment I have a combination of lights and cover. It would be rather complex to convert cover state to output states.

wpmjones, 2024-07-17:
I know that this is an old issue, but since it's open, I'll add my perspective.  I have a device to which I have added a buzzer (ledc output).  I want my touchscreen to only function if the buzzer is on.  I'd love to be able to say `if (id(my_buzzer).is_on())` or `if (id(my_buzzer).state == "on")`.  I know that there is also a closed issue under this integration for a similar request.  There are definitely times when it would be beneficial to know the state of an output.

nagyrobi, 2024-07-17:
- You could implement a template switch on top of the output and monitor that;
- You could use a template output before the physical output and with `write_action` you could split the "state" information per your needs.