esphome/feature-requests #689: Mitsubishi Heavy for ir climate

mSapik, 2020-04-21:
**Describe the problem you have/What new integration you would like**

Hello please, add support for Mitshubishi Heavy AC for IR climate remote.
(Mitsubishi Heavy SRKxxZJ-S (Remote control P/N RKX502A001C))

**Please describe your use case for this integration and alternatives you've tried:**

Now i use ESPEasy (https://espeasy.readthedocs.io/en/latest/Plugin/P088.html#p088-page)
But ESP Easy custom IR, not have climate integration for home assistant.
Im use rest command for control AC like (http://my_ip/control?cmd=heatpumpir,mitsubishi_heavy_zj,1,3,0,19,1,1)

**Additional context**

Sory for bad english

marcotestaccount, 2021-06-18:
Same need here. Found a github where someone already completely analyzed the IR codes using an IR analyzer:
https://github.com/SierraLimaOscar/RKS502A503-AC-Remote-IR-codes

matphillips, 2021-10-16:
Just to add a +1 here.
I have an SRK22TP unit, branded 'Beaver' which is a Mitsubishi name apparently.
Turns out Mitsubishi Heavy Industries and Mitsubishi Electric are different companies, and the SRK (MHI) units dont respond to the codes being sent by the climate.mitsubishi integration.
In theory.... it's just a case of changing the codes???

Anyway, would appreciate someone knowledgable looking at it; I've been trying to get my head around it and failing.

sasukebinbin, 2021-11-23:
Please help add. ESP8266 IR Remote library include Mitsubishi Heavy

mSapik, 2023-01-10:
Found description of the IR protocol of Mitsubishi Heavy air conditioners IR remote control protocol.

[https://github.com/joedirium/Mitsubishi_Heavy_HVAC_IR](https://github.com/joedirium/Mitsubishi_Heavy_HVAC_IR)

May be useful.

karllinder, 2023-01-19:
Hi,

give this PR a test, its both receive IR and send. Support for Preset, ECO mode will set to ECO, Activity will set A/C to night mode/low temperature and BOOST to Boost.

`external_components:
  source:
    type: git
    url: https://github.com/karllinder/esphome-climate-mhi
    ref: Add_ECO_Preset

remote_receiver:
  id: rcvr
  dump: aeha
  pin:
    number: GPIO25
    inverted: true
    mode:
      input: true
      pullup: true
  # high 55% tolerance is recommended for some remote control units
  #tolerance: 55%

remote_transmitter:
  pin: GPIO26
  carrier_duty_percent: 50%

climate:
   - platform: mhi
     name: "Vardagsrum MHI"
     receiver_id: rcvr

`

mSapik, 2023-01-20:
> Hi,
> 
> give this PR a test. It based on https://github.com/Dennis-Q/esphome-climate-mhi work. My is tested with Remote RLA502A700K and Mitsubishi heavy SRK25ZSX-WB, should work with all SRKxxZSK-Wx Both receive IR and send. Support for Preset, ECO mode will set to ECO, Activity will set A/C to night mode/low temperature and BOOST to Boost.
> 
> ```
> external_components:
>   source:
>     type: git
>     url: https://github.com/karllinder/esphome-climate-mhi
>     ref: Add_ECO_Preset
> 
> remote_receiver:
>   id: rcvr
>   dump: aeha
>   pin:
>     number: GPIO25
>     inverted: true
>     mode:
>       input: true
>       pullup: true
>   # high 55% tolerance is recommended for some remote control units
>   #tolerance: 55%
> 
> remote_transmitter:
>   pin: GPIO26
>   carrier_duty_percent: 50%
> 
> climate:
>    - platform: mhi
>      name: "Vardagsrum MHI"
>      receiver_id: rcvr
> ```

Thats great! But not work for my AC unit with remote PJA502A704AA remote control.

karllinder, 2023-01-20:
To bad, if you have a remote_receiver you can, with above setting, see the codes.
What ESPHome have is also support for the aeha protocol, see [https://esphome.io/components/remote_transmitter.html#remote-transmitter-transmit-aeha](url) if you want you can send me the debug and i can see if there is any easy changes that could be done.

mSapik, 2023-01-20:
I fork https://github.com/Dennis-Q/esphome-climate-mhi and edit code with values from https://github.com/ToniA/arduino-heatpumpir/blob/master/MitsubishiHeavyHeatpumpIR.cpp and https://github.com/ToniA/arduino-heatpumpir/blob/master/MitsubishiHeavyHeatpumpIR.h 

For my AC unit i use MitsubishiHeavyZJHeatpumpIR part of code

Result here https://github.com/mSapik/esphome-climate-mhi/tree/My

Transmitter work great! But i dont understand how work reciver part of code.

karllinder, 2023-01-21:
Great!
On the receiver, it´s seen in the debug view what happens with the receiver, from line 96 ESP_LOGD..
on row 133 the received byte should be decoded and then update. row 72 to 226 is for recive part, have no effect on the transmit part.
Look at the log when using the remote, and for there try to find out what´s wrong.

The use is that if you change with the remote, then the same settings will be reflected into ESP and forward that to HomeAssistant

mSapik, 2023-01-24:
I understood receiver part of code. And checked with a remote, all commands are recognized correctly except temperature.

for you AC unit:
transmit: (~((uint8_t)temperature - 17) & 0x0F)
receive: (~bytes[7] & 0x0F) + 17

for my AC unit:
transmit from arduinoheatpump: (~((temperatureCmd - 17) << 4)) & 0xF0
and i dont know how to invert this to receive...

I dont know C++, i just copy past from one file to other file...
Can you help me?)

timbba80, 2023-05-02:
> Horizontal and vertical swing not working correctly... I try to fix it...

Were you able to find a way to add manual option to set swing modes (e.g. up and left)? 

And btw, big thanks for the fork, it did help me to get rid of Sensibo cloud solution and use esphome instead with my MHI unit!

mortNZ, 2024-06-08:
> Hi,
>
> give this PR a test. It based on https://github.com/Dennis-Q/esphome-climate-mhi work. My is tested with Remote RLA502A700K and Mitsubishi heavy SRK25ZSX-WB, should work with all SRKxxZSK-Wx [...]

Hi there, sorry I know this is from a year and a half ago but I'm in the process of setting up my heat pump that isn't natively supported and wanted to try this; 
I've included that pull request in my esphome config external_components, and when I go to install the climate lines I get the following error on install:

```
INFO ESPHome 2024.5.5
INFO Reading configuration /config/esphome/atom-2.yaml...
Failed config

climate.mhi: [source /config/esphome/atom-2.yaml:84]
  
  Platform not found: 'climate.mhi'.
  platform: mhi
  name: AC Livingroom
  receiver_id: rcvr
```

esphome config:
```
external_components:
  - source: github://Jorre05/remote_receiver
    components:
      - remote_receiver
  - source:
      type: git
      url: https://github.com/karllinder/esphome-climate-mhi
      ref: Add_ECO_Preset
```
atom.yaml:
```
climate:
  - platform: mhi
    name: "AC Livingroom"
    receiver_id: rcvr
```

Any suggestions greatly appreciated.

mani94321, 2024-10-17:
> Hi there, sorry I know this is from a year and a half ago but I'm in the process of setting up my heat pump that isn't natively supported and wanted to try this; I've included that pull request in [...]

i am having the similar issue.

mani94321, 2024-10-18:
> > I understood receiver part of code. And checked with a remote, all commands are recognized correctly except temperature.
> > for you AC unit: transmit: (~((uint8_t)temperature - 17) & 0x0F) receive: (~bytes[7] & 0x0F) + 17
> > for my AC unit: transmit from arduinoheatpump: (~((temperatureCmd - 17) << 4)) & 0xF0 and i dont know how to invert this to receive...
> > I dont know C++, i just copy past from one file to other file... Can you help me?)
> 
> I will answer myself: ((~bytes[9] & 0xF0) >> 4) + 17 Now temperature work fine.
> 
> Horizontal and vertical swing not working correctly... I try to fix it...

can u plz share your final work. i am struggling with receiver part.

mSapik, 2024-10-18:
> can u plz share your final work. i am struggling with receiver part.

https://github.com/esphome/esphome/pull/6858 look here, maybe it will help

karllinder, 2024-11-17:
> can u plz share your final work. i am struggling with receiver part.

Hi
sorry for the delay, please see if this will work?

https://github.com/karllinder/esphome-climate-mhi/blob/main/README.md

/Karl

mSapik, 2025-04-23:
@karllinder Hello, I don't want to be pushy but could you help me check a custom component for different MHI models? As far as I understand, in your case it's the ZM model. Here's an example:
```
external_components:
  source:
    type: git
    url: https://github.com/mSapik/esphome-climate-mhi
    ref: main
  refresh: 0s

climate:
  - platform: mhi_multi_ir
    name: "Mitsubishi Heavy AC"
    model: "ZM"
```