esphome/feature-requests #691: Ability to change Wifi TX power

kbouchard111, 2020-04-21:
**Describe the problem you have/What new integration you would like**

Would like the ability to set tx power under the wifi component options.
Either implemented as a one time hardcoded value(set at compile time/through YAML) or exposing the value to homeassistance for on the fly changes.

**Please describe your use case for this integration and alternatives you've tried:**

With the growing use of my ESPHome install (and forecasted growth), I've setup multiple access points to limit the amount of nodes connecting to a single AP.
Although these nodes transmit very little to the AP, 802.11N standards leverage CDMA/CD as a medium access method. This has practical limitations on it's scalability (I really look forward to the day that TDMA on the 802.11 standard becomes mainstream).

Given the limitations of the ESP8266 and ESP32 platform (mainly only 2.4ghz, no 5ghz), this translates to a max of three non-overlapping channels that can be used (1,6,11).
One interference mitigation technique is to reduce the TX power of both the AP and clients to decrease the background noise levels. This would enable to re-use of similar channels with reduced physical separation due to other segments not seeing as much noise from the neighboring AP.

Also for those worried about their wifi exposure (and the increasing wireless devices in places like bedrooms), this would provide some reduced exposure.

This would also provide some power savings for nodes being battery powered (though how much is open for debate). Peak draws (usually happening during transmit operations) should also see a decrease making battery operated nodes last a bit longer on lower battery levels.

I currently only have esp8266 nodes on the network so I cannot comment on the ESP32 abilities. However for the esp8266, I have tried the WIFI option power_save_mode with a value of LIGHT.
Results were very unstable(usually disconnecting a minute or two after reboot) and staying unreachable despite the AP still showing an active connection.
The measured signal at the AP also indicated that the TX power was not reduced.
Given results at even LIGHT, I have not attempted to use the value HIGH.
This was with a node that was un-obstructed to the AP and was receiving around a -50db signal so disconnects were not signal strength related (and all other sensors, including this one worked fine when power_save_mode was set to NONE).
The specific purpose of power_save_mode is not exactly the same as reducing tx power. So although related, even working as intended would not be solution.

**Additional context**

I can muddle around code, however I dont know where to start for testing changes/pull requests.
I have however found at line 129 on https://github.com/esphome/esphome/blob/dev/esphome/components/wifi/__init__.py
appears to set the TX power of the ESP8266 to 20dbm.
This matches measured readings I got from one of my nodes at around 18dbm.
Exposing this value (presuming I am correct) and allowing it to be changed within the YAML config should meet the above request.

Thanks.

glmnet, 2020-04-26:
Are you aware of output_power as used in this post https://github.com/esphome/issues/issues/455#issuecomment-552976310

I did not find it documented though.

kbouchard111, 2020-04-26:
Nope didn't come up in my search. That's exactly what I was looking for. Thanks!

The amp on my ESP8266 module is far from linear however setting output_power to 10db (lowest setting) does get the desired outcome of about a 8db change. Would have been nice to be able to set it even lower.

Here are the measurements at the AP:
20db tx      -49dBm
15db tx	-50dBm (expected -54dBm)
10db tx	-57dBm (expected -59dBm)

Kind of new to this, how do we go about requesting the documentation gets updated for others? Discord?

kbouchard111, 2020-04-26:
...spoke too soon. It appears that once set, it reduces the TX power. However it settles back to the original transmit power after some time.

I saw mention of calling WiFI.setOutputPower(0) in one of the issues linked, I'll investigate this option.

glmnet, 2020-04-26:
> how do we go about requesting the documentation gets updated for others

there is a link at the botton of the page: "Edit in github"

kbouchard111, 2020-04-27:
pull request submitted to update documentation(hope I did that correctly).

As for TX power reverting back to default value, it appears to happen around the 30min mark.
Uptime counter is not being reset which indicates its not a restart of the ESP8266.
I'll do some more testing across all nodes to see if the time interval varies and if setting log level: VERY_VERBOSE provides any indication on what is going on.
If anyone else has the ability to check signal levels at their AP and have a few moments to confirm this is not an isolated case, it would be appreciated.

closed 2022-07-01 by nagyrobi