esphome/feature-requests #692: web_server: Expose Prometheus /metrics-Endpoint

margau, 2020-04-23:
**Describe the problem you have/What new integration you would like**
Expose available data on /metrics in web_server-component to allow prometheus scraping data from the esphome node.

**Please describe your use case for this integration and alternatives you've tried:**
Prometheus (https://prometheus.io/) is one of the mostly used Monitoring and TSDB-Solutions around infrastructure. Data is collected by prometheus scraping an HTTP-Page - mostly under /metrics - periodically.
If esphome exposes it's metrics directly, esphome-nodes could be easily integrated into existing Prometheus-Environments, using both the TSDB and Alerting-Features. The need for an additional exporter (like going via MQTT or Home Assistant) would be dropped, reducing complexity.

**Additional context**
An example format could be:
```
light_brightness{id="neopixel_light",platform="neopixelbus"} 255
light_color{id="neopixel_light",channel="r",platform="neopixelbus"} 255
```
and the following for an sensor:
```
sensor_value{id="bathroom_temperature",platform="xiaomi_lywsdcgq"} 22.1
```

randybb, 2020-04-23:
What is wrong on Prometheus integration in HA?

margau, 2020-04-24:
Nothing wrong, but additional Overhead. If you only Wanna Track temperatures and other readings, you won't need Home Assistant at all - the task could be performed by esphome directly. Especially for more "industrial-like" use cases (thinking of WISP Hubs etc.), that could be great.

randybb, 2020-04-24:
The primary purpose of ESPHome is to connect remote sensors/actuators to Home Assistant. You may even have self sufficient nodes with all the logic programmed, but there should be always HA somewhere for remote monitoring and management. Otto mentioned multiple times (for example [rest calls](https://github.com/esphome/ESPHome/issues/55) and this is exactly the same) if something can be achieved through connected HA, then we don't need such implementation on nodes itself. 

In this prarticular case you have two options _node - HA - prometheus_, or _node - mqtt - prometheus_. I totally understand you want to eliminate the thing in the middle...

I use influxdb and grafana to visualize values from all sensors and never had any problems with that.

margau, 2020-04-24:
I do not agree in stating making REST Calls and having a prometheus target is exactly the same, and I didn't really see the need to increase complexity for some use cases by requiring an HA.

Asked the other way around: If I'd prepare an PR with this feature, would it be merged?

randybb, 2020-04-24:
Prometheus uses just a simple [REST API](https://github.com/prometheus/pushgateway/blob/master/README.md), so it is exactly the same.

Nobody know where @OttoWinter is, thus it is completely in @glmnet hands.

margau, 2020-04-24:
Well, I think there's a misunderstanding here. I do not talk about the pushgateway (which would the node pushing data to Prometheus), I'm talking about the node providing an Web Server target, which prometheus scrapes as HTTP Client (the "default"-prometheus way).

glmnet, 2020-04-24:
so what you'd like is a GET request to /metric to return all the node state in json format at once?

entities states is already encoded in json format to be sent to the web ui for realtime update of the web page, so may be this same state can be put together in an array and returned in /metric

Actually I did this already for sending all the initial data at once to the web page on first load (because a change on the web server library)

So if this is the case, it's quite simple.

margau, 2020-04-24:
Yes, except it's not in JSON Format. The Prometheus Format is described here: https://prometheus.io/docs/concepts/data_model/

glmnet, 2020-04-24:
I believe a PR will be welcomed then

margau, 2020-04-24:
Currently trying something out, and WIP-PR will probably come today.

glmnet, 2020-04-24:
It’d be nice to make it somewhat optional, thus it will depend on the web server component. We can discuss how to make this optional. What are your ideas?

May be a new 

`prometeus:` a root component which depends on web_server, this might need to make web server extensible 

Or by adding a optional prometeus to the web_server component. However this way the prometeus code will be always bundled even if the user does not enable it. This is also ok so long the code has not a big footprint.

closed 2020-06-10 by glmnet