esphome/feature-requests #697: turn_on() turn_off() functions for status_led, once the device is connected

Diramu, 2020-04-26:
I'm looking for a simpler method to the led control by status_led, so it can be used a generic output led once the device is connected.

If there were  turn_on/off/toggle() function for the  gpio defined in the _status_led:_ block, that were non functional until the device is connected to the api or mqtt,  then this task would be much easier. 

I can actually get the behaviour I want with the following code:
```
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO1                                                                                                                                                                                                
    id: button
    on_press:
      - switch.toggle: relay
  - platform: status
    name: "${short_name} Status"
    id: connect

switch:
  - platform: gpio
    name: ${short_name}
    pin: GPIO14
    id: relay
  - platform: template
    id: relay_mirror
    lambda: |-
      if (id(connect).state > 0 ) {
        if (id(relay).state > 0 )                                                                                                                                                                                            
        {                                                                                                                                                                                                                    
          id(led).turn_off();                                                                                                                                                                                                
        } else {                                                                                                                                                                                                             
          id(led).turn_on();                                                                                                                                                                                                 
        }                                                                                                                                                                                                                    
      }                                                                                                                                                                                                                      
      return id(relay).state;                                                                                                                                                                                                

status_led:
  pin:
    number: GPIO13
    inverted: yes

output:
  - platform: gpio
    id: led
    pin:
      number: GPIO13
      inverted: True
```
**Additional context**

The above code could simply to this with the addition of these functions to the status_id component
```
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO1                                                                                                                                                                                                
    id: button
    on_press:
      - switch.toggle: relay
 
switch:
  - platform: gpio
    name: ${short_name}
    pin: GPIO14
    id: relay
  - platform: template
    id: relay_mirror
    lambda: |-
     if (id(relay).state > 0 )                                                                                                                                                                                            
        {                                                                                                                                                                                                                    
          id(led).turn_off();                                                                                                                                                                                                
        } else {                                                                                                                                                                                                             
          id(led).turn_on();                                                                                                                                                                                                 
        }                                                                                                                                                                                                                                                                                                                                                                                                                                        
      return id(relay).state;                                                                                                                                                                                                

status_led:
  pin:
    number: GPIO13
    inverted: yes
```

glmnet, 2020-04-27:
This is a hack, and if it works for you is ok, but this will result in a complicated explanation of the docs of what will happen if you use this...
As your code is written if wifi is lost led will start blinking, also if e.g. some i2c device loses comm then it will blink too. (Status led is not just connected)

Also you are just saving 6 lines of yaml with this feature!!

closed 2020-04-27 by glmnet

Diramu, 2020-04-28:
I believe my original cod is the hack.  I do want led to revert back to 'led_status' when connection is lost. 

Actually a better method would to  have
```
output:
  - platform: gpio
    id: led
    pin:
      number: GPIO13
      inverted: True
   status:
```
a status option output: