esphome/feature-requests #698: Web Server - expose entities defined as internal

theone11, 2020-04-26:
**Describe the problem you have/What new integration you would like**
I want to see internal data/status of the ESPHome entity but **without** exposing it to HA - don't need to create uneeded HA entities that now need to be hidden/disabled.

I have defined several sensors for internal use:
* text_sensors: IP, Connected SSID, Connected BSSID, (soon MAC Address), version
* sensors: uptime, wifi_signal
* Status
* Internal binary_sensors, switches, outputs, etc...

**Please describe your use case for this integration and alternatives you've tried:**
The current method I use is to connect using the "Show Logs" in ESPHome HASS addon and waiting for the information to be seen in log.

**Additional context**

glmnet, 2020-04-27:
I would have suggested to add a config option to `web_server` and mark this issue with *good first issue* but this will require a lot of changes on the core itself

theone11, 2020-04-27:
A config option would be good.
I assumed it would need changing the core code.

nagyrobi, 2022-07-07:
Implemented on version 2, use `include_internal` option: https://esphome.io/components/web_server.html

closed 2022-07-07 by nagyrobi