esphome/feature-requests #699: 868MHz emitter/transmitter

flouck, 2020-04-26:
**Describe the problem you have/What new integration you would like**

It seems to me there is no support for a 868 MHz tranceiver to receive and emit custom signals.

**Please describe your use case for this integration and alternatives you've tried:**

I have 868MHz remote controllers that I would like to copy and be able to simulate directly from Home Assistant. Use case is to open and close window shutter.
Besides, my water meter HRF-C-G2 emits a 868MHz and it can be interrogated anytime by sending a specific signal to it. It would help me to monitor nearly real time the water consumption.

**Additional context**

glmnet, 2020-04-27:
RF transmitters and receivers do the modulation by themselves. If the protocol is different then it won't work, but the carrier frequency is not handled by ESPHome in any RF case.

flouck, 2020-04-27:
Hi @glmnet,
I am sorry, I am still learning and discovering.
Does it mean ESPHome cannot be used on an ESP if I want to handle RF? Or does it mean I can use this ESP to be managed by ESPHome and I. The same time manage myself the tranceiver ?

glmnet, 2020-04-27:
I made a google search and find no much info about 868mhz nor your water meter.

Please provide more technical information linked.

flouck, 2020-04-27:
Alright, thank you @glmnet.

The water meter is designed by Homeridersystems, the model is HRF-c-G2 v1.6 and it emits in 868MHz. This page (sorry, it is in French) has been able to hack into a similar water meter in 433MHz by simulating the signal to wake up the water meter. By reading the comments, it seems to be working on Domoticz.
http://www.lamaisonsimon.fr/wiki/doku.php?id=maison2:compteur_d_eau:compteur_d_eau

My other use case is regarding the water shutters. The remote reference is CE868-2 or RCY-1 94V-0 or 41677 /B and it seems from what I can read online that it sends a 868MHz to open or close the shutter. Here is a link (again in French, so sorry about that): 
http://www.sav-volet.fr/category/piece-detachee-volet-bubendorff-carte-electronique-radio

glmnet, 2020-04-27:
seems to use UART too, so far we don't have any RF protocol which talks UART.

Also you are missing the hardware transceiver, i.e. I own a 433mhz and 315mhz one, they look like this https://www.banggood.com/3sets-433MHz-100M-Wireless-Transceiver-Module-Kit-Transmitter-Receiver-Copper-Spring-Antenna-p-1187020.html?cur_warehouse=CN

You'll need stuff like this for 868mhz

glmnet, 2020-04-27:
To get this working with ESPHome will require a lot of research and effort, and those things aren't much popular yet.

flouck, 2020-04-27:
Oh ok I understand.. Of course, I will need to buy hardware and I don't mind that. 
Do you have any clue on how I can achieve what I am trying to do? Maybe a tip which I can use to I look for the solution to my problem.