esphome/feature-requests #701: Shelly 1/1 PM addon / sensor module

stingbyte, 2020-04-26:
**Describe the problem you have/What new integration you would like**

Shelly 1/1 PM got an addon to support Ds18b20 temperature and some combined temp/humidity sensors (DHT). I was not able to get the sensors detected by ESPHome as it looks like the Shelly's are using a special variant of I2C to support this sensors via the addon to the GPIO ports. I did find an implementation for it over at Tasmota's GitHub. It would be very helpful to me and probably other Shelly Users if this could be intehrazed in ESPHkme too.

**Please describe your use case for this integration and alternatives you've tried:**

Use case would be temperature sensor integration, as with standard DS18B20 module.

**Additional context**

Tasmota's Integration can be found here:
https://github.com/arendst/Tasmota/issues/7469

duckfullstop, 2020-09-13:
Any interest in implementing this from anyone? @nekomimiscience is tempted to take a look at porting the Tasmota code at some point in the next week or two, but I figured it was worth checking just on the slim chance anyone has put the legwork in already.

Looks a fairly straightforward port.

iangregory, 2020-09-23:
I had a look at the differences in the Tasmota drivers for a standard DS18x20 and the Shelly version. The diff is here: https://www.diffchecker.com/0MukRm6v

The changes are minimal - the Shelly version of the Tasmota driver removes the support for parasite power (as it's not required) and defines different pins for the input and output. Otherwise it's the same code.

I've not looked in detail at the esphome codebase, but I suspect adding optional `input_pin` and `output_pin` configuration items to the existing one wire support might be a fairly small change (keeping the existing `pin` configuration but overriding it if the new config is provided). This would allow users to provide the new configuration for shelly devices, but retain backwards compatibility for other use cases.

persuader72, 2020-10-23:
I have made a custom component with changes need on one wire module to make it able to work with the shelly temperature sensor. You can find it on https://github.com/persuader72/esphome-components/tree/master/shelly_dallas. 
Do you think can be usefull a patch to the dallas module in order to integrate this changes in the esphome core?

stafbulp, 2020-10-31:
@persuader72 Could you describe how to implement this custom component in ESPHome?

persuader72, 2020-10-31:
@stafbulp You have to create a custom_components folder in the same folder where your yaml config file resides and then copy the entire shelly_dallas folder in there.
Next, you can configure the component using shelly_dallas as a platform in your config file. (See this example https://github.com/persuader72/esphome-components as reference).
To find the ID of probes you can follow the same procedre in https://esphome.io/components/sensor/dallas.html
Regards

stafbulp, 2020-10-31:
@persuader72 Thank you so much, got it working perfectly now!

Erd86, 2020-11-01:
Is ist possible to read more than three DS18B20 sensors with this custom component?
With original shelly firmware, the number of sensors is limited to three devices..

persuader72, 2020-11-01:
@Erd86 I only tested it with three probes but I checked the code and there is not any limitation about the maxmimum number of sensors that the component can discover or use. So I think that it is possible to use more than three sensors with this component (shelly_dallas) or with the original one (dallas).

trollsoft7, 2020-11-21:
@persuader72 Thanks for the great custom component - its working great. Besides the DS18B20 I'd need it also for the DHT22 - could I achieve using DHT22 with Shelly1 + Temp Add On with configuration only? Or does it take programming? Thanks again for your contribution!

persuader72, 2020-11-22:
@trollsoft7 You are welcome.
About the DHT22 I think it's theoretically possible to make it works with DHT22, but not together with DS18B20, you have to use one or another. 
To make working the shelly addon with the DHT22 sensor a patch to the dht component must be done. If you want, I can try to make it but I can't test it because my shelly addon is currently installed and not easy to access. So you have to test the patch by yourself. 
Regards

trollsoft7, 2020-11-22:
@persuader72 Thanks for your reply! I have a Shelly+AddOn+DHT22 installed and could immediately test it. Until now I ran Tasmota on it, but meanwhile I wish to migrate everything to ESP-Home. So yes, if you could spent your precious time, I promise to immediately try it out and provide you logs or whatever you need. Greetings from Austria!

persuader72, 2020-11-22:
@trollsoft7 Ok , let me some days (a week I think) I will send you a notification when the patch is ready to test.
Regards

persuader72, 2020-11-22:
@trollsoft7 I was more simple than I thought.... (If it works)
You can find the component in  https://github.com/persuader72/esphome-components/tree/master/shelly_dht you have to configure the component like the following lines (the other options are the same as the original component):

sensor:
  - platform: shelly_dht
    pin: GPIO3
    pin_a: GPIO0

Some hints: You can't mix the original dht with this component in the same configuration, and you can't mix DS18B20 and DHT22 in the same bus.

Let me know if it is working or not.
Regards

trollsoft7, 2020-11-22:
@persuader72 Wow so quick, just tried it - installation and compilation worked flawlessly. But it obviously does not the find the sensor. 
Config is like this:`# Device Specific Config

sensor:
  - platform: wifi_signal
    name: $devicename WiFi Signal Sensor
    update_interval: 60s

  - platform: shelly_dht
    pin: GPIO3
    pin_a: GPIO0
    model: 'AM2302'
    temperature:
      name: $devicename temp
    humidity:
      name: $devicename hum
    update_interval: 5s
`
Output is like this:
[image]

I tried different model-selections, but it always says "DHT22 or equivalent". Error message is always the same. Could I activate some more detailed debugging log?

persuader72, 2020-11-22:
@trollsoft7  I have found one error. Please pull and try again

trollsoft7, 2020-11-22:
Whohoo, its working perfectly - thank you very much! Hope you have a good any healthy time! best regards!
[image]

persuader72, 2020-11-22:
@trollsoft7 glad to hear that, I'll try this sensor too in the next ì days.
Regards

shallax, 2021-11-14:
Any chance of getting this component adopted into esphome? Seems like a fairly easy win as the code is already done and working.

applechrist, 2021-12-18:
Is it possible to use other sensors like the XKC-Y25-NPN as well?

persuader72, 2021-12-20:
Electrically I don't think so because the esp8266 is not 5v tolerant. I don't know if the Shellu1 PM addon (the black box in front of shelly) is 5v tolerant. If it is not you have to put a level shifter (5v to 3.3v) between the sensor and the shelly addon.
Furthermore, the sensor wants a power supply of 5v (from 5v to 24v) and the shelly doesn't provide the minimum voltage required.

At the software level, I think it is possible because the sensor is only a digital output line,  you can configure GPIO3 as binary_sensor to read the status of the sensor, to avoid bus conflicts you have to configure also the GPIO0 as binary_sensor even if you don't use it.

P.S. I did not know this sensor before, I used the following page as a reference (https://wiki.dfrobot.com/Non-contact_Liquid_Level_Sensor_XKC-Y25-T12V_SKU__SEN0204)

jowgn, 2022-01-13:
@persuader72 is there any chance to repair the code found in pr https://github.com/esphome/esphome/pull/1346 to make it work with the current version of esphome and @OttoWinter is there a particular reason the pr wasn't merged back than?
Is there anything that can be done to help here?

persuader72, 2022-01-25:
@jowgn I don't know the reason because the PR is not reject nor accepted.  Have you tested the code of this custom component? (https://github.com/persuader72/esphome-components/tree/dev) I recently updated it to support la versions of esphome.
Any way if @OttoWinter  is interested I can fix the PR and make  it capable of being merged again

shallax, 2022-01-25:
> @jowgn I don't know the reason because the PR is not reject nor accepted. Have you tested the code of this custom component? (https://github.com/persuader72/esphome-components/tree/dev) I recently [...]

I have a PR open to fix it again for more recent versions of esphome, which have removed to_string methods.

colmbuckley, 2022-06-06:
Hey folks -

Just wondering if this PR was ever merged? Can we count on the add-on sensor support with the 1PM now?

closed 2022-07-01 by nagyrobi