esphome/feature-requests #705: Document all supported YAML extensions

iBobik, 2020-04-27:
**Describe the problem you have/What new integration you would like**
I miss proper docs on what features I can use in YAML files on top of the specification (anchors, overrides, includes, …). This docs should include examples for common use cases.

**Please describe your use case for this integration and alternatives you've tried:**

I would like to have common settings in shared config files, so each device's file has only it's specific settings plus customizations and overrides. 

**Additional context**

I've found this chunks what could be base for proper docs:
- https://github.com/esphome/esphome/pull/888
- https://next.esphome.io/components/switch/gpio.html?highlight=yaml%20anchor
- https://next.esphome.io/guides/configuration-types.html#config-substitutions
- https://github.com/esphome/issues/issues/886

glmnet, 2020-04-27:
Also here https://next.esphome.io/guides/faq.html#tips-for-using-esphome it refers to (most) HA extensions being supported

iBobik, 2020-05-03:
I miss examples how to extend structures from included files.

E.g. how to do this Python code in esphome's YAML:
```python
common_yml = {
  "binary_sensors": [
    {
      "platform": "gpio",
      "pin": {
        "number": 3,
        "mode": "INPUT_PULLUP",
        "inverted": true
      },
      "name": "Button on RX/GPIO2"
    }
  ]
}

device1_yml["binary_sensors"][0]["on_press"] = {
  "then": [
    {
      "light.toggle": "light_1"
    }
  ]
}
```

iBobik, 2020-05-04:
Here are some hints on how it could be possible to use anchors and aliases with include (not tested with esphome): https://stackoverflow.com/questions/44910886/pyyaml-include-file-and-yaml-aliases-anchors-references