esphome/feature-requests #709: Implement Auth support in http_request

danps1, 2020-04-29:
**Describe the problem you have/What new integration you would like**

Please consider adding authentication support to the new http_request component - https://next.esphome.io/components/http_request.html
**Please describe your use case for this integration and alternatives you've tried:**

This would allow for authenticated ESP to ESP communication using the Rest API - https://esphome.io/web-api/index.html#rest-api
The Rest API is part of the http_server component, and currently has support for authentication.
Using http_server without auth would not be a great idea from a security perspective.
Adding auth to http_request would allow for security in node to node communication.
**Additional context**

One example of where this may be useful is for automations that could still work even if the Home Assistant server fails, such as 2 wall mounted light switches both controlling the same light, as commonly seen at the top and bottom of stairs. 
A button could be pressed on the downstairs ESP, which would then use the http_request component to send an authenticated request to the Rest API on the Smart Bulb ESP.

brandond, 2020-04-29:
You can already do this, you just need to craft the auth header yourself. You can get the base64 string from `echo -n  "username:password" | base64` - if there are special characters in either you'll need to urlencode them first.

```yaml
on_...:
  - http_request.get:
      url: https://esphome.io
      headers:
        Content-Type: application/json
        Authentication: Basic dXNlcm5hbWU6cGFzc3dvcmQ=
      verify_ssl: false
```

danps1, 2020-04-30:
Thanks @brandond - that's really good to know
I must be doing something elementary wrong, but when I try and test this out, I get config errors. I should be using the ```dev``` branch right?...
```
% esphome version
Version: 1.15.0-dev

% esphome DownstairsSwitch.yaml run
INFO Reading configuration DownstairsSwitch.yaml...
Failed config

binary_sensor.gpio: [source DownstairsSwitch.yaml:25]
  platform: gpio
  pin: 
    number: 3
    inverted: True
  id: button_2
  internal: True
  on_press:  [source DownstairsSwitch.yaml:33]
    - [source DownstairsSwitch.yaml:33]
      
      Unable to find action with the name 'http_request.post'.
      http_request.post:  [source DownstairsSwitch.yaml:34]
        url: http://192.168.X.Y/light/stairs/toggle
        headers: 
          Content-Type: application/json
          Authentication: Basic XXXXXXXXXXXXX
        verify_ssl: False
```

brandond, 2020-04-30:
Please read the http_request docs that you linked to - it sounds like you forgot to create a http_request component? If you have and it's still not working, post your full configuration yaml.

danps1, 2020-04-30:
Awesome - that was exactly what was wrong - I need to RTFM! - thanks @brandond 
Unfortunately a different problem popped up with the authentication.
It seems like the http_server component actually only supports Digest authentication, not Basic, as is mentioned in the documentation at https://next.esphome.io/components/web_server.html#configuration-variables
I have checked this using [Postman](https://www.postman.com/) - and authenticating to the http_server component only works with Digest, and not Basic. I can also see this in the headers returned by the other ESP.
I will edit this feature request to be a request specifically to implement DIGEST auth in http_request - not sure how involved that is. Obviously the auth is more complex as it involves a back and forth, but it may be something available in the underlying libraries used by http_request.
I'll also log a bug with the documentation for http_server so that the documentation correctly says it is Digest.
Thanks again @brandond

readylan, 2020-10-21:
Hello !
I try to send sensor values to a cloud platform wich needs "Authorization" header (basic)
This is the affected section in my yaml file:
```
- http_request.post:
                url: https://api.mydevices.com/things/***DEVICE_ID**/data
                headers:
                  Content-Type: application/json
                  Authorization: Basic YjZhNDg4MTAtZjcyOC0 <<<<< b64 encoded username/password >>>>>M2Q1OWJiOWM0MWY0N2U4ZWUwZDhlMmRiNjU2N2I5MTVlZmM=
                verify_ssl: false
                json: |-
                    root["channel"] = 3;
                    root["value"] = x; 
                    root["type"] = "co2"; 
                    root["unit"] = "ppm"; 
```
it compiles ok, but when the sensor reports data I can see this error message in logs:
```
[21:10:47][D][sensor:092]: 'Workshop eCO2': Sending state 413.00000 ppm with 0 decimals of accuracy
[21:10:47][W][http_request:048]: HTTP Request failed; URL: https://api.mydevices.com/things/***DEVICE_ID**/data; Error: connection refused
```
and off course, the data is not posted.
The same data can be posted from an online utility that is sending in raw:
```
POST /things/***DEVICE_ID**/data HTTP/1.1
Authorization: Basic YjZhNDg4MTAtZjcyOC0 <<<<< b64 encoded username/password >>>>>M2Q1OWJiOWM0MWY0N2U4ZWUwZDhlMmRiNjU2N2I5MTVlZmM=
Host: api.mydevices.com
Content-Type: application/json
Content-Length: 96

[
   {
      "channel": 3,
      "value": 450.1,
      "type": "co2",
      "unit": "ppm"
   }
]
```
So, can I use basic Authentication in my use case?

Regards,

Eduardo

Jibbonator, 2022-08-15:
I Would like to do the same. Direct connection from espnode to espnode.
if i make a post call from my smartphone with digest authentication it works.

Buit the following sample on esp8266 doesn't work :

headers:
  Content-Type: application/json
  Authentication: Digest ABCDEFGH

I cant undestand why this issue is closed, or did i miss something obvious?

bkbartk, 2023-04-28:
I really would like to have this feature too for digest.
for the exact some reason an @Jibbonator mentioned.
off course, I can remove authentication from the webinterface but that is a concession I do not want to do.

nagyrobi, 2023-06-03:
Cookbook: https://esphome.io/cookbook/http_request_sensor.html

Jibbonator, 2023-06-04:
I´ve tested with POST requests and it works. Thanks for the solution

bkbartk, 2023-06-12:
just been able to test and this works great, I thought it wouldn't because nonce etc are always different but it does work

JAKAMI99, 2025-05-09:
For everyone wondering:
"Authentication: Digest" is way more sophisticated as Method "Basic".
Sadly, you can not set the Auth to Basic for the web server, so other esp nodes could access it with their own http request lib since the Authentication **Digest** needs **sophisticated calculations** to prevent replay attacks.

From what I have seen, the http_request component has a function to calculate this header, but it doesn't seem to be callable from simple configurations currently
https://github.com/esphome/esphome/blob/1da8e99d27ee73900f55caea43feda61a9e07cb3/esphome/components/http_request/httplib.h#L5355