esphome/feature-requests #710: Warn user when trying to use a dht sensor on input only pin

TessyPowder, 2020-04-30:
**Describe the problem you have/What new integration you would like**
I wasted 3 hours yesterday, because i didn't know GPIO35 on my node-mcu-32 only supports Input -> my DHT11 Sensor didn't work and in the log a generic error message got logged:
```
[W][dht:126]: Requesting data from DHT failed!
[W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
```

This is not a bug, because it was my fault, but a more specific error message would have saved me a lot of time.
I found out by accident, that it works on GPIO2 ...

**Please describe your use case for this integration and alternatives you've tried:**

Users like me may not know that certain pins are Input or output only.
When using components, that require input/output and the selected pin doesn't support that, a error message in the log may be extremely helpful.

**Additional context**

Maybe every developer here probably knows, that certain pins are inut/output only.
But i started using ESP 2 days ago and because i didn't know this problem from arduino, wich i used for a longer time, i was quite confused (i got the dht11 sensor to work on my arduino mega instantly)
Maybe this could happen with other sensor components too.

randybb, 2020-04-30:
You have learned that you have to check documentation or at least pinout diagram and this is not waste of time. This is case of every MCU/SoC... and it is even worse on MCUs with limited peripherals.
In our case some pins are reserved for internal SPI flash, or connected to usb-to-serial and using them for other purpose as serial port could cause problems, some pins doesn't support internal-pullup (the safest approach is to use external pull-up / pull-down resistors),... there are many situation when we would need have warning/error. Then we need cover every platform/board... it sounds like a loot of work if this is not defined in platformio board database.

rradar, 2020-05-04:
The documentation even features a picture of your board @TessyPowder 

[image]

https://esphome.io/devices/nodemcu_esp32.html

TessyPowder, 2020-05-04:
> Edit: it looks like we have some warnings implemented in ESPHome Dashboard:
> [image]

I think a dht Sensor is the second thing  many beginners try after a blinking Led. And because It is a Sensor you might think a Input Pin is enough. I know software can't deal with every edge case user error, but a warning about input pins seams possible.

closed 2022-07-06 by nagyrobi