esphome/feature-requests #711: PCA9954 port expander

valepe, 2020-04-30:
**Describe the problem you have/What new integration you would like**

Is it possible to add the PCA9954 to the I2C port expander integrations?
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

pedrobsm, 2020-11-23:
Any news on this? This chip is used on M5Stack I/O expander

randybb, 2020-11-23:
I will do it, just need to have it in my hands - still waiting for it with bunch of other M5 modules.

pedrobsm, 2020-11-23:
@randybb, I have no experience on developing custom modules, but let me know if you need any help on testing it. I have the module M5 module.

pedrobsm, 2021-01-25:
@randybb, did you do it? :)

geelenbert, 2021-08-13:
I'm looking for support for this i2c expander as well.

Creating some sort of custom component would also be okay, but i lack the knowlage for this....

viotemp1, 2022-10-17:
Hi, is this chipset working with esphome now? I would like to use it with [M5Stack Extend Serial I/O Unit (PCA9554PW)](https://shop.m5stack.com/products/official-extend-serial-i-o-unit)

fonix232, 2025-06-01:
This component has been available since [22 December 2022](https://github.com/esphome/esphome/commit/c0a4e07e5a35e561dce4623b56ac2e6722d42a0d). I think this FR can be closed

closed 2025-06-01 by valepe