esphome/feature-requests #712: Expose FastLED power management feature

arvind-balaji, 2020-04-30:
**Describe the problem you have/What new integration you would like**

Allow setting a power limit when configuring a FastLED light

**Please describe your use case for this integration and alternatives you've tried:**

It would be nice to be able to always have the led's at max brightness without overloading the power supply. 

**Additional context**

As far as I can tell, `color_correct` is not a sufficient solution (#82). 
FastLED uses the current LED state to dynamically determine the max  brightness bases on the value of each pixel. [FastLED Docs](http://fastled.io/docs/3.1/group___power.html#gafff15ebb8adee6a6c76fd46eb980397c)
I don't think this behavior is currently replicable in ESPHome.

frog32, 2020-11-26:
My use case for this is a ws2811 powered Christmas light. During normal operation I use a flicker effect to light up single pixels at a time and want them to be at maximum brightness. However when setting the whole thing to white it would exceed the power supply and voltage would drop to a level that shuts down the esp. Then I have to unplug the whole thing to have pixels revert to black.

top4ek, 2021-09-12:
+1 for that feature. Got dead PSU today 😁

Eenoo, 2022-08-02:
+1 This feature would be really handy. Potentially alongside an estimated current draw for power calcs in HA.

digitydogs, 2022-09-06:
Another vote to add this feature.

I have recently converted all my blinds into smart blinds using esphome. All of my windows currently have addressable leds lining the interior for use in holiday light shows. (xlights e131)   Naturally I went ahead and had the blinds also drive the LEDs eliminating the need for another controller per window.  

As it is now I am having to limit all three channels to 40%. now when compared to the wled displays used in my shows the windows look lackluster and the color matching is noticeably off. 

I really don't want to have to get another 20 ESPs to run wled in parallel to espome when the latter should be able to do it all :)

monocycler, 2022-10-09:
I would like to vote for this feature. I was using WLED but switched to ESPHOME because I want to expose a GPIO binary sensor to Home Assistant on the same ESP that is controlling a LED strip.
I have attached an 89 LED WS1812B strip to 5V and GND pins of my ESP32 running esphome firmware. I found I had to use brightness: 50% to avoid drawing too much current and crashing my ESP32.

tbaumann, 2023-06-05:
This is an extremely useful feature in WLED and allows some pretty spectacular power output with theoretically under-powered PSU.
I ran into brown outs as soon as I tried a longer 5V strip with ESPHome. I am looking into a bigger PSU, but I doubt I have enough space in my housing.
If I had this function I could avoid all that without a visible degradation.

total1ty, 2024-01-30:
Bumping this because I'd love to have this feature too. I need a lot of the features and flexibility of ESPHome and don't want to switch to WLED for this feature alone..

distante, 2024-02-23:
Oh, is he only thing preventing me to make full use of  my ESP32, with WLED I can limit the output so my board doesn't crash and the PSU is overloaded, but I can not use any output pin for anything.

Seems like there is a good starting point here : https://github.com/Makuna/NeoPixelBus/issues/310