esphome/feature-requests #716: Support WX08ZM new mijia temp/humidity sensor

pplucky, 2020-05-01:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

ahpohl, 2020-05-03:
@pplucky Thanks for opening a separate feature request for the WX08ZM. The state property really calls for a binary sensor to publish the on/off state properly. I need a good example how to publish some float values like battery and mosquito tablet alongside with the state. I think binary sensor has also support for float values, but I'll have to read up on this.  Meanwhile you can test the code as is in [ahpohl/esphome:xiaomi_complete](https://github.com/ahpohl/esphome/tree/xiaomi_complete)

pplucky, 2020-05-03:
> The state property really calls for a binary sensor to publish the on/off state properly

Yep, that's true. Let me know if you need any testing there.

> Meanwhile you can test the code as is in [ahpohl/esphome:xiaomi_complete](https://github.com/ahpohl/esphome/tree/xiaomi_complete)

Looking good:
[image]
1 LYWSD03MMC,2 WX08ZM and 1 LYWSDCGQ

ahpohl, 2020-05-04:
Great its still working! Yesterday I worked on the documentation and updated the pull requests. Its nearly finished.

Magalex2x14, 2020-05-04:
HA has a separate sensor entity class for this. All I had to do was to convert the state of the sensor to a Boolean type. The device_class for this entity I select from the dictionary, based on the type of device. However, in the case of my custom component, there is one thin point related to the delay in changing the state of an entity in HA due to the periodic nature of the work of my platform (details [here](https://github.com/custom-components/sensor.mitemp_bt/issues/76), if interested).

sermayoral, 2020-05-06:
I have a question. Can this device be turned on/off using BLE?

Magalex2x14, 2020-05-06:
If this can be done from the MiHome application, then it can. The only thing that it is usually done through the BLE connection, which it is not implemented in ESPHome, if I remember correctly (correct me if I am mistaken).

ahpohl, 2020-05-15:
@pplucky I updated the WX08ZM to a hybrid `sensor` and `binary_sensor` component. Now the power status in HA looks much nicer (on/off rather than 0/1) and the color of the symbol changes as well. You can pull the updated code from [ahpohl/eshome:xiaomi_wx08zm](https://github.com/ahpohl/esphome/tree/xiaomi_wx08zm) until is has been merged with ahpohl/eshome:dev.

In order to compile, you need to have both `sensor` and `binary_sensor` in your config:
```
sensor:

binary_sensor:
  - platform: xiaomi_wx08zm
    name: "WX08ZM Mosquito Repellent"
    mac_address: "74:a3:4a:b5:07:34"
    tablet:
      name: "WX08ZM Tablet Resource"
    battery_level:
      name: "WX08ZM Battery Level"
```

pplucky, 2020-05-15:
> @pplucky I updated the WX08ZM to a hybrid `sensor` and `binary_sensor` component. Now the power status in HA looks much nicer (on/off rather than 0/1) and the color of the symbol changes as well. [...]

Looking good:
[image]

As expected, the binary sensor doesn't update immediately once device is turned on/off, but it is quite acceptable I think.

ahpohl, 2020-05-16:
@pplucky Thank you very much for testing the hybrid binary_sensor / sensor code.

I merged the changes with [ahpohl/eshome:dev](https://github.com/ahpohl/esphome) and updated the documentation. The preview is [here](https://deploy-preview-576--esphome.netlify.app/).

pattyland, 2021-02-04:
I guess this issue can be closed: https://esphome.io/components/sensor/xiaomi_ble.html#wx08zm

closed 2021-02-05 by pplucky