esphome/feature-requests #717: Native support for buzzer

MarcoFranchi, 2020-05-03:
**Describe the problem you have/What new integration you would like**
using PWR to manage Buzzer in not easy, new output component for passive Buzzer it could be a good idea

**Please describe your use case for this integration and alternatives you've tried:**
Support for passive buzzer with parameters like Duration, Tone and Volume
or better like this : https://www.letscontrolit.com/wiki/index.php?title=Buzzer_(RTTTL)

randybb, 2020-05-03:
Something like this https://github.com/esphome/esphome/pull/817. And yes, it is not documented :D

rradar, 2020-05-08:
@randybb if I'm right esphome/esphome#817 was never merged but the stale bot ate it :robot:

randybb, 2020-05-08:
@OttoWinter Is alive and I believe he will fix it soon :)

muxa, 2021-07-30:
@MarcoFranchi I believe the feature you were after was implemented. Please close this ticket.

OttoWinter, 2021-08-01:
For reference: implemented here: https://esphome.io/components/rtttl.html

closed 2021-08-01 by OttoWinter