esphome/feature-requests #718: IPv6 support

l9i, 2020-05-03:
**Describe the problem you have/What new integration you would like**
ESPHome should be usable (== addressable) on an IPv6-only network.

**Please describe your use case for this integration and alternatives you've tried:**
My HomeAssistant used to live on an IPv6 only network until it needed to communicate with an ESPHome node.

**Additional context**

Based on my cursory look, Espressif IDF supports IPv6 with SLAAC.

myfreeweb, 2020-06-24:
Looks like there is already work on mDNS-IPv6 in the arduino core: https://github.com/esp8266/Arduino/pull/7335

Unfortunately enabling IPv6 in lwIP currently breaks ESPAsyncTCP: https://github.com/me-no-dev/ESPAsyncTCP/issues/148

ski7777, 2020-11-15:
Is there any progress on this?

leonelsr, 2021-01-11:
Still nothing? Have anyone at least started porting this to IPv6-enabled LWIP?
I am looking for any examples (from forks of this or any other entirely different projects) so I can understand how calls to dual-stack `IPAddress` structs should look like and how should any interface deal with it, so I/we can start at least some hacky working draft for this...

RHAD1969, 2021-03-22:
What is the status of the IPv6 support? If I implement a new device architecture, I have to be sure that it is future proof.

grischard, 2021-03-22:
Enabling IPv6 in lwIP still breaks ESPAsyncTCP, and the issue was unfortunately closed because no one was working on it: me-no-dev/ESPAsyncTCP#148.

That project seems dead - no commits since 2019, and eight pull requests collecting dust.

In any case, it's a problem to be fixed there, and not in esphome.

jack9603301, 2021-08-03:
If possible, I also expect esphome to support IPv6 protocol stack, and I think this is the trend in the future

Ruakij, 2021-08-15:
v6 is the future and will also make many things easier e.g. static-ips without dhcp (faster link-local-addresses)

Hedda, 2021-09-23:
Matter (formerly Project CHIP) Application Layer support requirements might be a very strong argument why IPv6 support is needed.

Created separate feature request discussion for Matter/CHIP Application Layer support -> https://github.com/esphome/feature-requests/issues/1430 but looks like IPV6 is a requirement.

Hedda, 2021-09-23:
hmm, do these documents from ESP-IDF Programming guide for ESP32 v4.3.1 make it sound like IPv6 is supported on ESP32?

https://docs.espressif.com/projects/esp-idf/en/v4.3.1/esp32/api-guides/lwip.html?highlight=ipv6

ESP-IDF version of lwIP (esp-lwip) list IPv6 support under FEATURES:

https://github.com/espressif/esp-lwip/blob/2.1.2-esp/README

* **_IP (Internet Protocol, IPv4 and IPv6) including packet forwarding over multiple network interfaces_**

https://github.com/espressif/esp-lwip

grischard, 2021-09-23:
The blocker here is IPv6 support on the esp8266. It would be unfortunate if esphome only supported ipv6 on the esp32.

Hedda, 2021-09-23:
> The blocker here is IPv6 support on the esp8266. It would be unfortunate if esphome only supported ipv6 on the esp32.

OK then that is understandable. Espressif and the Matter/CHIP only support Matter/CHIP Application Layer on ESP32 anyway.

HeMan, 2021-12-22:
I looked a bit into this, and created an updated ESPAsyncTCP that works with IPv6, https://github.com/HeMan/ESPAsyncTCP/tree/ipv6 and I have a local version of https://github.com/HeMan/esphome/tree/ipv6 that could at least start `web_server` on an ESP8266.
It seems that the Arduino for ESP32 doesn't have IPv6 support, but I'll try to get it to work with esp-idf.

HeMan, 2022-02-20:
ESPHome 2022.2.0 was released with IPv6 support for esp-idf framework, https://esphome.io/changelog/2022.2.0.html#new-features

grischard, 2022-02-21:
Note that the esp-idf framework doesn't support some components (yet), for example the web server or the captive portal.

Nardol, 2022-07-01:
@nagyrobi can we really consider this as completed as it is only available with ESP-IDF which still not support all component and is only for ESP32?

nagyrobi, 2022-07-01:
AFAIK the long-term goal is to move everyting to ESP-IDF.

Nardol, 2022-07-01:
And what about ESP8266?
I thought ESP-IDF was only for ESP32, which is the case in ESPHome.

About not supported components I was referring for example to captive portal which is still not compatible with ESP-IDF, because of web server which is not yet compatible if I am not wrong. But I suppose in long term it will be compatible...

wtremmel, 2022-07-01:
Also I cannot compile esp-idf on Raspberry PI4

HeMan, 2022-07-01:
I have a working IPv6 implementation for ESP8266 that needs some work. Some of the components doesn't support IPv6, and I have a created a PR to solve one (https://github.com/OttoWinter/ESPAsyncTCP/pull/4).
IPv6 on the Arduino version for ESP32 needs a newer Arduino than what Platform.io had when I last looked into it, but that could have changed now.

MaZe3D, 2023-02-26:
Tasmota has gained great success in implementing IPv6 for ESP32 and ESP8266. Details can be found in the related issue https://github.com/arendst/Tasmota/issues/2521 and their release notes https://github.com/arendst/Tasmota/releases/tag/v12.4.0. Maybe their implementation is helpful or also compatible with esphome.

HeMan, 2023-04-29:
Finally got some time to work on IPv6 support and have created two PRs, https://github.com/esphome/esphome/pull/4750 and https://github.com/esphome/esphome/pull/4759. The second one needs https://github.com/OttoWinter/ESPAsyncTCP/pull/4 to be merged before it compiles.

HeMan, 2023-05-11:
Got IPv6 to work with Ethernet as well, https://github.com/esphome/esphome/pull/4807

Hedda, 2023-05-12:
_Originally posted by @HeMan in https://github.com/esphome/esphome/issues/4807#issuecomment-1544766968_

> This just adds dynamic IPv6 via SLAAC or DHCPv6, in par with IPv6 for ESP-IDF (#2953). Adding static addresses needs a bigger internal change that starts with #4729.

Off-topic but an indirectly related question; dynamic IPv6 addresses should be compatible with Matter standard/protocol, right?

Question is related to this other ESPHome feature request for Matter standard/protocol support -> https://github.com/esphome/feature-requests/issues/1430

As I understand the core Matter protocol should in theory be able to communicate over any type of IPv6 transport and network.

HeMan, 2023-05-12:
> Off-topic but an indirectly related question; dynamic IPv6 addresses should be compatible with Matter standard/protocol, right?
>
> Question is related to this other ESPHome feature request for Matter [...]

I think so. Dynamic addressing, especially on IPv6, is a more convenient way to maintain a network. Static addresses have it uses so I'm still going to implement it, but I think the main part is going to use dynamic addresses.

wtremmel, 2023-05-12:
Keep in mind there are two kind of dynamic address assignment in IPv6: SLAAC and DHCPv6. For IoT devices, address assignment via DHCPv6 might not be necessary (Android devices for example also cannot use DHCPv6 to get an IPv6 address).

If you want to read up about IPv6, there is a nice public book here on Github: https://github.com/becarpenter/book6

HeMan, 2023-08-24:
I now have basic IPv6 working on all platforms and have started to rewrite the internals to be more address agnostic. It also need to understand that it could have multiple addresses, since IPv6 could have a few and also since it will be dual stack it could have one IPv4 address. DNS-resolving for MQTT for example need to look for both A- and AAAA-records, and happy eyeballs should perhaps be implemented as well.

wtremmel, 2023-08-24:
I just enabled IPv6 on one of my sensors - but debug output did not show any IPv6 addresses being assigned. Checking my firewalls NDP table I only see a link-local address for this device. Do I need to enable any additional debug to see IPv6 related messages?

HeMan, 2023-09-13:
Yey, PR for enabling IPv6 on ESP8266 and RP2040 (Raspberry Pi Pico) was merged yesterday! 🥳

HeMan, 2024-02-23:
I think this could be solved by https://github.com/esphome/esphome/pull/5449. Could you people test it when it gets merged or from my branch now?

kquinsland, 2024-02-23:
> I think this could be close after [esphome/esphome#5449](https://github.com/esphome/esphome/pull/5449) is merged. Could you people test it when it gets merged or from my branch now?

Thank you for all the work you've done on this! <3.

I was planning on taking some time this weekend to update all my devices now that 24.02 is out but I can try pulling in your branch.

The `yaml` should look like this:

```yaml
external_components:
  - source: github://pr#2639
    components: [ wifi_info ]
```

Or does `components` need the full suite: `ethernet, ethernet_info, mqtt, network, socket, wifi, wifi_info` ?

HeMan, 2024-02-23:
> Thank you for all the work you've done on this! <3.
>
> I was planning on taking some time this weekend to update all my devices now that 24.02 is out but I can try pulling in your branch.
>
> The `yaml` [...]

You need to clone my repo and check out the branch. It can unfortunately not be used as an external component.

HeMan, 2024-02-27:
It is now merged so you should be able to use the dev-branch.

HeMan, 2024-03-21:
It's now released as part of 2024.3, https://esphome.io/changelog/2024.3.0.html#ipv6. There are some more things I'd like to add, like support for IPv6 only networks, but it should be usable in dual stack environments for now. Could you people test?

wtremmel, 2024-03-21:
It did not work in one of my IPv6 only-networks. Kept disconnecting (like if it does not have an IPv4 address it did not consider itself connected).

Also (in a dual stack network) it does not add an IPv6 DNS address. (using RA only. No DHCPv6).

HeMan, 2024-03-21:
> It did not work in one of my IPv6 only-networks. Kept disconnecting (like if it does not have an IPv4 address it did not consider itself connected).
Yeah, that is expected since it still deserves IPv4 to consider started. 
> 
> Also (in a dual stack network) it does not add an IPv6 DNS address. (using RA only. No DHCPv6).

Did you mean mDNS? Regular DNS could not be updated with SLAAC. And mDNS works best with esp-idf version 5.0.x or above unfortunately.

HeMan, 2024-03-21:
> > Did you mean mDNS? Regular DNS could not be updated with SLAAC.
> 
> Perhaps something worth working on. I really dream about getting rid of IPv4 completely for my internal networks...

mDNS could be worth working on. SLAAC does only send out information regarding the network, it doesn't get anything back, which means that it can't update a DNS. DHCPv6 could do that, but lwip doesn't support that part of DHCPv6.

aaliddell, 2024-03-21:
> mDNS could be worth working on. SLAAC does only send out information regarding the network, it doesn't get anything back, which means that it can't update a DNS. DHCPv6 could do that, but lwip [...]

DNS is optionally present in the RA packets used for SLAAC: https://datatracker.ietf.org/doc/html/rfc8106

So when you receive RA required to do SLAAC, you could take the IPv6 DNS info if it is present without having to use DHCPv6.

Nardol, 2024-03-21:
> DNS is optionally present in the RA packets used for SLAAC: https://datatracker.ietf.org/doc/html/rfc8106
>
> So when you receive RA required to do SLAAC, you could take the IPv6 DNS info if it is [...]

Yes, named RDNSS if I am not confusing with something else.
I don't know if lwip supports it.

HeMan, 2024-03-21:
RDNSS is to anounce which DNS server to use, not to update the DNS-server. RDNSS is supported by lwip.

HeMan, 2024-03-21:
Oh, now I see what you mean! RDNSS do set the DNS server on the ESP, but it's not shown in the log. Just tried to remove my DNS setting in the DHCP server and just propagated it via SLAAC and resolving of addresses worked. Will try to figure out how to display it.

Nardol, 2024-03-21:
For MDNS and ESP8266, does esphome/esphome#5359 change how IPv6 is supported?

kquinsland, 2024-03-21:
> Yeah, that is expected since it still deserves IPv4 to consider started.

Is there a list of $things in ESPHome that still rely on IPv4?

HeMan, 2024-03-22:
ESPHome networking components (ethernet and wifi) needs an IPv4 address to assume a connection. I haven't looked at what other components that communicates over IP that still needs IPv4.

Hedda, 2024-05-31:
I guess one future scenario for IPv6-only devices is how and when native support for OpenThread/Thread is added to ESPHome?

* https://openthread.io/guides/thread-primer
  * https://openthread.io/guides/thread-primer/ipv6-addressing
  * https://openthread.io/codelabs/openthread-border-router

_[Thread](http://threadgroup.org/) is an **IPv6-based networking protocol** designed for low-power Internet of Things devices in an IEEE 802.15.4-2006 wireless mesh network, commonly called a Wireless Personal Area Network (WPAN). Thread is independent of other 802.15 mesh networking protocols_

FYI, that includes ESP32-H2 and ESP32-C6 (and newer SoCs), but note there is a separate feature request for OpenThread here:

* https://github.com/esphome/feature-requests/issues/1397

PS: Off-topic but FYI, there is also an OpenThread proof-of-concept project for ESPHome discussed in community forum here:

https://community.home-assistant.io/t/announcement-esphome-running-on-adafruit-52840-over-thread-mesh-networking-using-zephyrrtos/363712

* https://github.com/natelust/esphomeZephyr

alexhaydock, 2024-12-03:
One of the missing pieces of the puzzle for IPv6 support here seems to be the ESPHome Dashboard.

In the Docker container at least, it seems to default to listening on `0.0.0.0` rather than all interfaces.

This can be worked around with this addition to a Compose file:
```yaml
command: "dashboard --address '' /config"
```

Unless I'm mistaken, this might be as simple as a one-line change in `__main__.py` to change the default for the address argument away from being `0.0.0.0`?

https://github.com/esphome/esphome/blob/a37ff2dbd97d17c5b9b522c2a4515fd8311b7003/esphome/__main__.py#L920

That default value for `args.address` seems to be getting passed into `start_web_server()` here:

https://github.com/esphome/esphome/blob/a37ff2dbd97d17c5b9b522c2a4515fd8311b7003/esphome/dashboard/dashboard.py#L142

mhalano, 2025-03-08:
Any news about that? I think IPv6-only support is the missing piece to create a Matter stack for ESP32.

HeMan, 2025-03-08:
> Any news about that? I think IPv6-only support is the missing piece to create a Matter stack for ESP32.

There is a decent IPv6 support in ESPHome now, but I have no idea how it could be used in a Matter context.

mhalano, 2025-03-08:
@HeMan They mention IPv6 support here #1430. That's because IPv6 is part of the Matter implementation, not IPv4, even there are devices that implement just IPv4 or dual-stack. The problem is: it should work if you have IPv6-only configured, as Matter standard requires it.

Hedda, 2025-03-10:
> _[@HeMan](https://github.com/HeMan) They mention IPv6 support here [#1430](https://github.com/esphome/feature-requests/issues/1430). **That's because IPv6 is part of the Matter implementation, not [...]

FYI and for the record; Home Assistant’s lead Matter developer also mentioned this fact in today’s Matter certification announcement (under a sub-section named ”_**The Future Matters**_” which high-lighted this need for IPv6 support):  

* https://www.home-assistant.io/blog/2025/03/10/matter-certification/

  * https://www.home-assistant.io/blog/2025/03/10/matter-certification/#the-future-matters

# _The future Matters_ 

_As we continue improving our Matter implementation, the Matter standard itself is also evolving. Even with the progress it has made in the past year, there are still challenges to overcome. Matter is completely built on IPv6, and its rollout is uncovering the poor state of IPv6 support in today’s network hardware. Also, as it’s a complicated standard, it can be difficult to explain concepts like Thread and Fabrics to the average user, but hopefully users won’t need to be experts in the future to get the most out of Matter in their homes._

HeMan, 2025-03-10:
I know that Matter is depending on IPv6, but I don't know how to use the current implementation with it.