esphome/feature-requests #719: Check that a manually default Default Gateway is on the same subnet as the IP address

WarmChocolateCake, 2020-05-04:
**Describe the problem you have/What new integration you would like**
Is it possible to verify that the default gateway is in the same subnet as the manual static_ip
 
**Please describe your use case for this integration and alternatives you've tried:**

I am segmenting my home network to put "IoT" devices on their own SSID - which means a different subnet and needs a default gateway.

Unfortunately, I modified the device's IP address (6 devices) and forgot to change their Default Gateways to the new subnet.

The only way to get things back to normal was to shutdown the new IoT Wifi, wait for all of the devices to enable their fallback APs and reconfigure using their captive portals (many thanks for that feature!)

If Manual_IP then
  is Gateway set AND is Gateway the same as (static_ip AND subnet mask)
else
  warn the end user that they could be doing something wrong
end

**Additional context**
Obviously this isn't an issue for DHCP users, but I'm not sure if the logic works for all use cases - but this would've saved me ~2 hours work today :)

Thanks for the great work

randybb, 2020-05-04:
I thought that IoT is mainly about auto-configuration, so any static network configuration is only for people that like make their life harder for unknown reason. Btw, fallback AP for IoT is not a good idea - if your WiFi will be down, for example during software upgrade, you will lost all nodes. They will not join back to your WiFi automatically, thus it should not be used in production.

Next people would say, they need to be notified when they are using domain names without configured DNS :) Or if you change settings on more than 2 devices without testing it first you should get a popup if are you sure about it..

WarmChocolateCake, 2020-05-04:
Well, I come from a background of industrial control and "automatic" is a bad word :)

(OT, but if I could totally disable Bonjour / Avahi / whatever else it's called - I would... or at least reconfigure it to not use ".local")

Also, consider some other ESP devices (not ESPHome specific) that have to wake up, find the wifi, use DHCP, then use DNS, possibly even deal with internet (S)NTP and _then_ transmit their MQTT values and finally go back to sleep.
A static configuration simplifies it to: wake up, transmit, sleep... and I save actual real money on batteries! :)

Fallback AP worked flawlessly for me - I just disabled the "IoT wifi", all the devices enabled their local fallback APs / captive portals and I reconfigured... then re-enabled the wifi and they all reconnected... no problem at all.

So, I think that the yaml parser does a great job of checking the syntax, so a 2nd parser for def. gateway is _relatively_ straightforward and would be a great feature.

randybb, 2020-05-04:
I am network engineer...it is just my experience that users are ******, many of them mess up with network settings and then complain that something is not working :)

You can avoid NTP and MQTT with enabling HA API as this is the main purpose of ESPHome.

Didn't know that fallback AP scans neighbor SSIDs and will reconnect back. Good to know...

But back to the "issue": It should be easy to implement it. We do IP validations in this file https://github.com/esphome/esphome/blob/dev/esphome/config_validation.py - In case you are experienced with python (I don't), then fell free to update it :)

WarmChocolateCake, 2020-05-04:
I hear that... I'm also doing a lot of networking... we also do not use autoneg speed & duplex or audo MDIX, which will probably amaze you ;)
 
OK, I am not experienced in python, but I have been looking at various programming languages for long enough, so I'll take a look at those links and see if I can break something - at least enough to help with my request :)
Thanks

nagyrobi, 2022-07-05:
> If Manual_IP then
> is Gateway set AND is Gateway the same as (static_ip AND subnet mask)
> else
> warn the end user that they could be doing something wrong
> end

This check could be implemented in the Dashboard's config validation.