esphome/feature-requests #720: Adafruit STEMMA / Seesaw i2c Capacitive Soil Sensor support?

heinowalther, 2020-05-04:
**Describe the problem you have/What new integration you would like**

I would like to have support for the Adafruit Soil Sensor which is not like the analog sensor because it uses i2c.
It presents Temperature and a values for the Capacitive sensor.

**Please describe your use case for this integration and alternatives you've tried:**

I of cause tried to hook it up to an D1 mini, but I can only have it detect the i2c addresses as show below... I searched though the different supported sensors without finding anything that was very useful.
I can see that the module has an Atmel SOC on it, so I guess they write their own code for the i2c responses...
Adafruit has some code that could be useful, as it is of cause possible to use the module with either Arduino or RaspberyPI...

https://learn.adafruit.com/adafruit-stemma-soil-sensor-i2c-capacitive-moisture-sensor/arduino-test

I am willing to help test this integration if I am pointed in the right direction :-)

**Additional context**

[22:34:27][D][api.connection:583]: Client 'ESPHome v1.14.3 (10.10.10.118)' connected successfully!
[22:34:27][I][app:100]: ESPHome version 1.14.3 compiled on May  4 2020, 22:32:46
[22:34:27][C][wifi:415]: WiFi:
[22:34:27][C][wifi:283]:   SSID: [redacted]
[22:34:27][C][wifi:284]:   IP Address: 10.10.30.101
[22:34:27][C][wifi:286]:   BSSID: [redacted]
[22:34:27][C][wifi:287]:   Hostname: 'test02'
[22:34:27][C][wifi:291]:   Signal strength: -48 dB ▂▄▆█
[22:34:27][C][wifi:295]:   Channel: 6
[22:34:27][C][wifi:296]:   Subnet: 255.255.255.0
[22:34:27][C][wifi:297]:   Gateway: 10.10.30.1
[22:34:27][C][wifi:298]:   DNS1: 10.10.30.1
[22:34:27][C][wifi:299]:   DNS2: (IP unset)
[22:34:27][C][i2c:028]: I2C Bus:
[22:34:27][C][i2c:029]:   SDA Pin: GPIO4
[22:34:27][C][i2c:030]:   SCL Pin: GPIO5
[22:34:27][C][i2c:031]:   Frequency: 50000 Hz
[22:34:27][I][i2c:033]: Scanning i2c bus for active devices...
[22:34:28][I][i2c:040]: Found i2c device at address 0x3F
[22:34:28][I][i2c:040]: Found i2c device at address 0x5F
[22:34:28][I][i2c:040]: Found i2c device at address 0x6F
[22:34:28][I][i2c:040]: Found i2c device at address 0x77
[22:34:28][C][logger:175]: Logger:
[22:34:28][C][logger:176]:   Level: DEBUG
[22:34:28][C][logger:177]:   Log Baud Rate: 115200
[22:34:28][C][logger:178]:   Hardware UART: UART0
[22:34:28][C][captive_portal:169]: Captive Portal:
[22:34:28][C][ota:029]: Over-The-Air Updates:
[22:34:28][C][ota:030]:   Address: test02.local:8266
[22:34:28][C][api:095]: API Server:
[22:34:28][C][api:096]:   Address: test02.local:6053
[22:34:43][D][api:067]: Disconnecting ESPHome v1.14.3 (10.10.10.118)

commander-chris, 2020-05-17:
Hi - 

1. Include the adafruit/custom libs:

```
    - 'CE_SoilSensor.h'
    - 'Adafruit_seesaw.h'
    - 'Adafruit_seesaw.cpp'
```

2. Code the custom lib CE_SoilSensor.h

```
#include "esphome.h"
#include "Adafruit_seesaw.h"

class CE_SoilSensor : public PollingComponent, public Sensor {
 public:
  // constructor
  Adafruit_seesaw ss;

  Sensor *TemperatureSensor = new Sensor();
  Sensor *MoistureSensor = new Sensor();

  CE_SoilSensor() : PollingComponent(6000) { }

  void setup() override {
    // This will be called by App.setup()
    ss.begin(0x36);
  }

  void update() override {
    if (float temp = ss.getTemp()) TemperatureSensor->publish_state(temp);
    if (uint16_t capread = ss.touchRead(0)) {
      float moist = capread;//((capread - 337.0)/1586.0)*100.0; 1020 -> 0%
      MoistureSensor->publish_state(moist);
    }
  }
};
```

3. Define a custom sensor component

```
- platform: custom
    lambda: |-
      auto SoilSensor = new CE_SoilSensor();
      App.register_component(SoilSensor);
      return  { SoilSensor->TemperatureSensor, SoilSensor->MoistureSensor };
    sensors:
    -  name: "Multisensor_3 Soil Temperature I2C"
       unit_of_measurement: '°C'
       accuracy_decimals: 1
       filters:
        - median:
            window_size: 10
            send_every: 10
            send_first_at: 10
    -  name: "Multisensor_3 Soil Moisture I2C"
       filters:
        - median:
            window_size: 10
            send_every: 10
            send_first_at: 10
```

4. The sensor does not work properly for me, though, because it saturates at 1015 at the slightest touch, see, e.g., https://forums.adafruit.com/viewtopic.php?f=8&t=157971&hilit=stemma+1015
which is why I will try to return them and get a refund.

Good luck, 
Chris

nicke94, 2020-06-18:
Hi, 

I managed to get a sensor working with the provided code, Is it possible to add support for multiple sensors on the same device? 

The sensors have support for different addresses with two jumpers

>  There's also two I2C address jumpers. The default address is 0x36, but you can select between 0x36 and 0x39 inclusive by shorting AD0 (to add 1 to the address) and/or AD1 (to add 2 to the address)

Br,
nicke

Segran, 2021-04-02:
Hi,
I have gone through all the steps and can compile the code with out any errors, but when I run the code it stops with ERROR Serial port closed! just after "Setting up WiFi". when I comment out the carom sensor code it runs just fine. I am using an ESP-32, if that makes any difference... Please advise
/Segran

jesperklang, 2022-06-12:
> Hi -
>
> 1. Include the adafruit/custom libs:
>
> ```
>     - 'CE_SoilSensor.h'
>     - 'Adafruit_seesaw.h'
>     - 'Adafruit_seesaw.cpp'
> ```
>
> 2. Code the custom lib CE_SoilSensor.h
>
> ```
> #include "esphome.h" [...]

Hi!

I'm trying to make this work, but I get error messages when compiling. Any idea what could be wrong? I'm new to ESPHome, and spent a few days trying to fix it, but now it's time to ask for help. Below is the error messages a long with my config that's attached. And the CE_SoilSensor.h is like the above post in the esphome folder. Hope someone can get me in the right direction.

`Linking /data/lizardboard/.pioenvs/lizardboard/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/lizardboard/.pioenvs/lizardboard/src/Adafruit_seesaw.cpp.o:(.text._ZN15Adafruit_seesaw4readEhhPhht+0x4): undefined reference to `_ZN18Adafruit_I2CDevice5writeEPKhjbS1_j'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/lizardboard/.pioenvs/lizardboard/src/Adafruit_seesaw.cpp.o:(.text._ZN15Adafruit_seesaw4readEhhPhht+0xc): undefined reference to `_ZN18Adafruit_I2CDevice4readEPhjb'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/lizardboard/.pioenvs/lizardboard/src/Adafruit_seesaw.cpp.o: in function `_ZN15Adafruit_seesaw4readEhhPhht':
Adafruit_seesaw.cpp:(.text._ZN15Adafruit_seesaw4readEhhPhht+0x70): undefined reference to `_ZN18Adafruit_I2CDevice5writeEPKhjbS1_j'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: Adafruit_seesaw.cpp:(.text._ZN15Adafruit_seesaw4readEhhPhht+0xb2): undefined reference to `_ZN18Adafruit_I2CDevice4readEPhjb'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/lizardboard/.pioenvs/lizardboard/src/Adafruit_seesaw.cpp.o: in function `_ZN15Adafruit_seesaw5writeEhhPhh':
Adafruit_seesaw.cpp:(.text._ZN15Adafruit_seesaw5writeEhhPhh+0x44): undefined reference to `_ZN18Adafruit_I2CDevice5writeEPKhjbS1_j'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/lizardboard/.pioenvs/lizardboard/src/Adafruit_seesaw.cpp.o:(.text._ZN15Adafruit_seesaw5beginEhab+0x0): undefined reference to `_ZN18Adafruit_I2CDeviceC1EhP7TwoWire'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/lizardboard/.pioenvs/lizardboard/src/Adafruit_seesaw.cpp.o:(.text._ZN15Adafruit_seesaw5beginEhab+0x4): undefined reference to `_ZN18Adafruit_I2CDevice5beginEb'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/lizardboard/.pioenvs/lizardboard/src/Adafruit_seesaw.cpp.o:(.text._ZN15Adafruit_seesaw5beginEhab+0x8): undefined reference to `_ZN18Adafruit_I2CDevice8detectedEv'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/lizardboard/.pioenvs/lizardboard/src/Adafruit_seesaw.cpp.o: in function `_ZN15Adafruit_seesaw5beginEhab':
Adafruit_seesaw.cpp:(.text._ZN15Adafruit_seesaw5beginEhab+0x50): undefined reference to `_ZN18Adafruit_I2CDeviceC1EhP7TwoWire'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: Adafruit_seesaw.cpp:(.text._ZN15Adafruit_seesaw5beginEhab+0x60): undefined reference to `_ZN18Adafruit_I2CDevice5beginEb'
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: Adafruit_seesaw.cpp:(.text._ZN15Adafruit_seesaw5beginEhab+0x86): undefined reference to `_ZN18Adafruit_I2CDevice8detectedEv'
collect2: error: ld returned 1 exit status
*** [/data/lizardboard/.pioenvs/lizardboard/firmware.elf] Error 1
========================= [FAILED] Took 163.65 seconds =========================`

And here's my config file:
```
esphome:
  name: lizardboard
  libraries:
    - 'Wire'
    - 'Adafruit_seesaw'
  includes:
    - 'CE_SoilSensor.h'

esp8266:
  board: d1_mini_pro

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "NotShowingYouThis"

ota:
  password: "NotShowingYouThis"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "NotShowingYouThis"
    password: "NotShowingYouThis"

captive_portal:
    
sensor:
  - platform: custom
    lambda: |-
      auto SoilSensor = new CE_SoilSensor();
      App.register_component(SoilSensor);
      return  { SoilSensor->TemperatureSensor, SoilSensor->MoistureSensor };
    sensors:
    -  name: "Multisensor_3 Soil Temperature I2C"
       unit_of_measurement: '°C'
       accuracy_decimals: 1
       filters:
        - median:
            window_size: 10
            send_every: 10
            send_first_at: 10
    -  name: "Multisensor_3 Soil Moisture I2C"
       filters:
        - median:
            window_size: 10
            send_every: 10
            send_first_at: 10
```

closed 2022-07-05 by nagyrobi

port3306, 2022-07-19:
> Hi, I have gone through all the steps and can compile the code with out any errors, but when I run the code it stops with ERROR Serial port closed! just after "Setting up WiFi". when I comment out [...]

I have the same Problem. I has use this code: 
https://randombytes.substack.com/p/using-the-adafruit-stemma-soil-sensor

MorganEzra, 2022-07-21:
Hi the sensor was working yesterday with the exemple code, now it doesn't work anymore i just have "Error seesaw not found" do you have an idea what is the problem ?