esphome/feature-requests #722: Log collector

blademckain, 2020-05-06:
**Describe the problem you have/What new integration you would like**
It would be nice if it were possible to somehow collect the logs of all esp devices within your network.
For example, the Dashboard could collect the logs of all the devices configured inside.
So it can be used for debugging

**Please describe your use case for this integration and alternatives you've tried:**
it might be useful for debugging crashes that occur in a non-deterministic way>

brandond, 2020-05-07:
I have used https://github.com/jpmens/mqttwarn for this in the past. You can configure it to subscribe to the esphome mqtt log topics and forward messages to syslog, write to a file, etc.

Note that some crash info can only be collected from the serial port, but it's certainly better than nothing.

glmnet, 2020-05-07:
Yeah, I've been thinking about collecting logs better. Collecting over TCP / api does not make much sense when troubleshooting communications. 

What would be the easiest for this? I have arduinos and sd cards so I was planning setting up a TTL logger which will be there.
Of course a ESP8266 or ESP32 can do it too.

It'd be nice to put together a guide

rradar, 2020-05-08:
I'm doing my logging with the native api so I have them in home assistant. It works only for the dev version right now but my code is something like that:

```yaml
logger:
  level: DEBUG
  on_message:
    level: WARN # minimum log level to get logged
    then:
      - text_sensor.template.publish:
          id: logger_error_text 
          state: !lambda |-
                return "Level: " + to_string(level) + ", Tag: " + tag + ", Message: " + message ;
      - binary_sensor.template.publish:
          id: logger_error
          state: ON
      - delay: 10s
      - binary_sensor.template.publish:
          id: logger_error
          state: OFF

binary_sensor:
  - platform: template
    name: "${upper_devicename} Logger Error"
    id: logger_error
    device_class: problem

text_sensor:
  - platform: template
    name: "${upper_devicename} Logger Error Text"
    id: logger_error_text
```

[image]

Not very beautiful I need to admit but very useful

nagyrobi, 2022-07-06:
See https://github.com/esphome/feature-requests/issues/223

DjordjeMandic, 2025-05-09:
I think that this is what you're looking for esphome/esphome#8637