esphome/feature-requests #723: Add warning "danger of death" for devices where GND is live.

CarlosGS, 2020-05-07:
**Describe the problem you have/What new integration you would like**

I was about to happily use the GPIOs inside my POW R2, and thankfully I checked with a multimeter. The GND pin is directly connected to the live wire.

My suggestion is to add a clear "warning: danger of death" in [the documentation](https://esphome.io/devices/sonoff.html), specially for devices with power measuring chips since it turns out they completely lack insulation.

**Please describe your use case for this integration and alternatives you've tried:**

New users will find ESPhome a great way to integrate many sensors. Some modules have internal live connections that are not evident for inexperienced users. Specially since the circuit board does _look_ like it has separation between the live area and ESP area. But there is a direct connection between GND and LIVE.

**Additional context**

If there is a warning in the documentation, it is not clear. At least I didn't see it and almost made a dangerous mistake.

glmnet, 2020-05-07:
Usually when you disassembly electronics is like that.
It'd be nice to add a yellow / black warning banner with a link about safety. Feel free to work on it.

brandond, 2020-05-08:
Almost all of these devices have the dc gnd connected to ac live due to how their power supplies are designed. Tasmota does have a big warning on their page about it:

https://tasmota.github.io/docs/devices/Sonoff-Pow/

glmnet, 2020-05-08:
Wow! looking at that PCB with the slot cut the blue cap along the transformer one would inadvertedly think they are isolated properly... it's like a trap!!

CarlosGS, 2020-09-12:
There is a warning message in [the Sonoff 4CH doc page](https://esphome.io/devices/sonoff_4ch.html), if you want to use it as inspiration, or to follow the same scheme.

I'm still very uncomfortable with the [main sonoff doc page](https://esphome.io/devices/sonoff.html) as **it doesn't mention that GPIOs become live for some devices**.

closed 2021-05-17 by frenck