esphome/feature-requests #724: VL53L0X long range mode

rradar, 2020-05-08:
**Describe the problem you have/What new integration you would like**
Make the long range mode available in the VL53L0X component.

**Please describe your use case for this integration and alternatives you've tried:**
Right now the default mode is used which only allow readings up to about 1200mm.

**Additional context**
The long range mode is already supported by the [upstream code](https://github.com/pololu/vl53l0x-arduino/) used by esphome

> https://github.com/pololu/vl53l0x-arduino/blob/master/examples/Single/Single.ino

```
// Uncomment this line to use long range mode. This
// increases the sensitivity of the sensor and extends its
// potential range, but increases the likelihood of getting
// an inaccurate reading because of reflections from objects
// other than the intended target. It works best in dark
// conditions.

//#define LONG_RANGE
```

rspaargaren, 2020-05-08:
I do not own the sensor but compared the code and did a small edit. 
https://github.com/rspaargaren/esphome/tree/VL53-long-range/esphome/components/vl53l0x
If you set the singnal rate limit on 0.1 then you should have long range enabled with the adjusted code.
If this works I can make it a bit more sofisticated.

rradar, 2020-05-08:
@rspaargaren it works!

`[D][vl53l0x:242]: 'ToF Distance' - Got distance 1.361 m`

Thank you for your time and work. If there is anything I can help with like testing or editing the documentation just tell me :wink:

rradar, 2020-05-08:
I don't know if this is any of importance but this two 'options' are also missing in esphome if I'm right :yum: 
```
// Uncomment ONE of these two lines to get
// - higher speed at the cost of lower accuracy OR
// - higher accuracy at the cost of lower speed

//#define HIGH_SPEED
//#define HIGH_ACCURACY

```

rspaargaren, 2020-05-09:
Hi I have adjust a part of the code. Can you do some testing:
Adding the option LR : True to your YAML config should set it in long range. Removing is back in normal mode or LR : False. Can you check with latest code.
Did not have time to compile so can still contain bug(s):
https://github.com/rspaargaren/esphome/tree/VL53-long-range

rspaargaren, 2020-05-09:
> I don't know if this is any of importance but this two 'options' are also missing in esphome if I'm right yum
>
> ```
> // Uncomment ONE of these two lines to get
> // - higher speed at the cost of lower [...]

This is a maybe first finish your request...

rradar, 2020-05-10:
> Can you do some testing:
Adding the option LR : True to your YAML config should set it in long range. Removing is back in normal mode or LR : False. Can you check with latest code.

It **fails now** (despite if the yaml contains the `LR:` or not):

```yaml
ERROR Unable to load custom component vl53l0x.sensor:
Traceback (most recent call last):
  File "/home/billgates/.local/lib/python3.6/site-packages/esphome/config.py", line 140, in _lookup_module
    module = importlib.import_module('custom_components.{}'.format(domain))
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 674, in exec_module
  File "<frozen importlib._bootstrap_external>", line 781, in get_code
  File "<frozen importlib._bootstrap_external>", line 741, in source_to_code
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/billgates/Documents/esphome/custom_components/vl53l0x/sensor.py", line 19
    cv.Optional(CONF_LR, default=False): cv.boolean,
     ^
SyntaxError: invalid syntax
Failed config

sensor.vl53l0x: [source tofd1.yaml:28]
  
  Platform not found: 'sensor.vl53l0x'.
  platform: vl53l0x
  name: ToF D1 Distance
  update_interval: 10s
  LR: True
```

rspaargaren, 2020-05-10:
> > Can you do some testing:
> > Adding the option LR : True to your YAML config should set it in long range. Removing is back in normal mode or LR : False. Can you check with latest code.
> 
> It **fails now** (despite if the yaml contains the `LR:` or not):

Hi can you test again. did a small change and compiles now...

rradar, 2020-05-10:
> The option can now be set as:
long_range: True or long_range: False or without = Flase

works exactly as described :1st_place_medal:

rspaargaren, 2020-05-10:
> > The option can now be set as:
> > long_range: True or long_range: False or without = Flase
> 
> works exactly as described 1st_place_medal

That is good to hear. Ok some clean up and I will push it for DEV

rradar, 2021-01-25:
Merged: https://github.com/esphome/esphome/pull/1055

closed 2021-01-25 by rradar