esphome/feature-requests #725: Status LED pwm/brightness support

rradar, 2020-05-08:
**Describe the problem you have/What new integration you would like**
I like to have a [pwm](https://esphome.io/components/output/esp8266_pwm.html) support for the [status led](https://esphome.io/components/status_led.html) to make it possible to dim it.

**Please describe your use case for this integration and alternatives you've tried:**
The blue onboard led of the wemos d1 mini is very (very) bright and if there is a error while the surrounding is dark (let's take night for a example) the light is extremely annoying and feels like the police is coming :oncoming_police_car: 

**Additional context**
A minimum option would be to have the ability to set a fixed brightness for the status led

JohannCR, 2022-05-14:
Same for any esp32 board. Those LEDs sure are annoying

GilDev, 2025-07-28:
You can now use the [Status LED Light](https://esphome.io/components/light/status_led) component with an Output Template, here’s an example using an [LEDC Output](https://esphome.io/components/output/ledc.html) PWM channel:

```yaml
output:
  - platform: ledc
    pin: 4
    id: my_status_led

  - platform: template
    id: status_output
    type: binary
    write_action:
      - if:
          condition:
            lambda: return state > 0;
          then:
            - output.set_level:
                id: my_status_led
                level: 10%
          else:
            - output.turn_off:
                id: my_status_led

light:
  - platform: status_led
    output: status_output
```