esphome/feature-requests #729: TM1637 4-Segment Display support

Vxider, 2020-05-13:
**Describe the problem you have/What new integration you would like**
TM1637 is a 4 digit 7 segment driver with 2 pin control (data + clock)

**Please describe your use case for this integration and alternatives you've tried:**
I found TM1637 has added in [#20 ](https://github.com/esphome/feature-requests/issues/20), but still cannot found the document on homepage. I have tried to add the following configuration, but it cannot work. ( Platform not found: 'display.tm1637'.)

```
  - platform: tm1637
    clk_pin: D6
    dio_pin: D5
    update_interval: 500ms
    lambda: |-
        static int i = 0;
        i++;
        if ((i % 2) == 0)
          it.strftime("%H.%M", id(homeassistant_time).now());
        else
          it.strftime("%H%M", id(homeassistant_time).now());
```

**Additional context**

randybb, 2020-05-13:
Still only in dev branch https://next.esphome.io/components/display/tm1637.html

Vxider, 2020-05-14:
Thanks, I'll try this.

closed 2020-05-14 by Vxider