esphome/feature-requests #730: Scroll function for MAX7219 7 seg display

rspaargaren, 2020-05-13:
**Describe the problem you have/What new integration you would like**
The text display on the MAX7219 display is now static. More data can be shown when data scrolls (left).

**Please describe your use case for this integration and alternatives you've tried:**
More data can be displayed by scrolling text

**Additional context**
Same work is in progress for max7219 digit display.

rspaargaren, 2020-05-13:
https://github.com/rspaargaren/esphome/tree/max7219-seg7-scroll
First try -> to be tested

rradar, 2020-05-14:
> First try -> to be tested

tested:

```yaml
display:
  - platform: max7219
    cs_pin: D2
    num_chips: 1
    intensity: 3
    lambda: !lambda |-
      it.print("Hello it is me!");
      it.scroll_left();
```

and works :tada:

minsuke, 2020-05-15:
It's great !! 
 thx for your effort with awesome codes.

rspaargaren, 2020-05-21:
@minsuke I have updated the code as I was expecting some error when the string is smaller than the actual display. Can you check on the updated code?
It would be interesting to check the following tests:
- Long code -> will it still scroll
- Short code -> that fits within the screen 4 digits?
- Short code -> with offeset value 
- Code including a dot.

closed 2022-07-05 by nagyrobi