esphome/feature-requests #732: Add support for Xiaomi-Yeelight BLE nightlight (MJYD2S)

vevsvevs, 2020-05-14:
**Describe the problem you have/What new integration you would like**
Add support for the new Xiaomi BLE nightlight with day/night motion detection (MJYD2S).

**Please describe your use case for this integration and alternatives you've tried:**
Using this light as a motion sensor could be pretty convenient for bathroom light automation.

**Additional context**
In-shape copy of the MJYD02YL, but equipped with a BLE interface. It supports motion detection event broadcast even with a bright ambient light, which is awesome! In addition, it sends the battery level, ambient light level (I'm not sure how - as absolute values or just as dark/light flag), and presumably "no-motion" event after 2 minutes of the last movement detection. Here some ADV packets captured in the VERY_VERBOSE mode and with duplicate frames filtered-out (MAC is 50:EC:50:CD:32:02):
```
In-dark events:
[05:01:36][VV][esp32_ble_tracker:365]:     Data: 58.59.F6.07.1D.02.32.CD.50.EC.50.9F.8C.98.5B.5C.04.4E.00.00.FB.B7.16.94 (24)
[05:01:38][VV][esp32_ble_tracker:365]:     Data: 48.59.F6.07.1E.96.AC.4E.ED.FA.2F.2E.4E.00.00.7F.63.4B.2F (19)
[05:01:44][VV][esp32_ble_tracker:365]:     Data: 10.59.F6.07.1F.02.32.CD.50.EC.50 (11)
[05:02:31][VV][esp32_ble_tracker:365]:     Data: 58.59.F6.07.20.02.32.CD.50.EC.50.6C.D8.CE.B0.5F.86.4E.00.00.F0.0E.7C.8E (24)
[05:02:35][VV][esp32_ble_tracker:365]:     Data: 58.59.F6.07.21.02.32.CD.50.EC.50.03.CA.7D.76.4E.00.00.EE.0B.B4.04 (22)
[05:02:37][VV][esp32_ble_tracker:365]:     Data: 10.59.F6.07.22.02.32.CD.50.EC.50 (11)
[05:03:38][VV][esp32_ble_tracker:365]:     Data: 48.59.F6.07.23.6F.54.F0.CD.70.AF.F6.4E.00.00.24.8C.25.C0 (19)

In-bright events:
[05:05:30][VV][esp32_ble_tracker:365]:     Data: 58.59.F6.07.25.02.32.CD.50.EC.50.E9.23.FE.99.4B.23.4E.00.00.B8.6B.97.E8 (24)
[05:05:33][VV][esp32_ble_tracker:365]:     Data: 48.59.F6.07.26.03.1D.33.C5.8F.01.FC.4E.00.00.AD.5B.46.F7 (19)
[05:05:34][VV][esp32_ble_tracker:365]:     Data: 10.59.F6.07.27.02.32.CD.50.EC.50 (11)
[05:07:31][VV][esp32_ble_tracker:365]:     Data: 58.59.F6.07.28.02.32.CD.50.EC.50.3D.24.E3.0E.D2.11.4E.00.00.68.A1.3A.DF (24)

In-dark events with keep moving within 1 min:
[05:09:12][VV][esp32_ble_tracker:365]:     Data: 58.59.F6.07.2D.02.32.CD.50.EC.50.23.31.0D.03.DB.EF.4E.00.00.5F.F5.09.BB (24)
[05:09:16][VV][esp32_ble_tracker:365]:     Data: 48.59.F6.07.2E.7F.28.2B.9A.33.F3.D4.4E.00.00.C0.7E.BB.51 (19)
[05:09:22][VV][esp32_ble_tracker:365]:     Data: 10.59.F6.07.2F.02.32.CD.50.EC.50 (11)
[05:12:18][VV][esp32_ble_tracker:365]:     Data: 48.59.F6.07.30.1D.12.24.A7.A9.92.D4.4E.00.00.FB.AE.D9.28 (19)
[05:12:32][VV][esp32_ble_tracker:365]:     Data: 58.59.F6.07.32.02.32.CD.50.EC.50.BC.95.BF.F2.C2.C1.4E.00.00.BC.AF.50.C0 (24)
```

ahpohl, 2020-05-14:
@vevsvevs I pushed initial support for the MJYD2S nightlight to [ahpohl/esphome:xiaomi_mjyd2s](https://github.com/ahpohl/esphome/tree/xiaomi_mjyd2s). Did you already extract the bindkey?

Here is a config template:
```
binary_sensor:
  - platform: xiaomi_mjyd2s
    device_class: motion
    name: "MJYD2S Night Light"
    mac_address: "7A:80:8E:19:36:BA"
    bindkey: "48403ebe2d385db8d0c187f81e62cb64"
    timeout: 5s
```
Decryption of the packets will fail with the wrong bind key. I see two types of data packets from the output above:
```
Data: 58.59.F6.07.1D.02.32.CD.50.EC.50.9F.8C.98.5B.5C.04.4E.00.00.FB.B7.16.94 (24) // 3 bytes value
Data: 58.59.F6.07.21.02.32.CD.50.EC.50.03.CA.7D.76.4E.00.00.EE.0B.B4.04 (22) // 1 byte value presumably battery status
```
Does the service data contain the usual "95 FE" magic bytes?

vevsvevs, 2020-05-14:
@ahpohl Sure, I got the bindkey but forgot to write it here, sorry. I've updated the issue context (just in case). So, battery status sends just fine! Now I should somehow to investigate which of these three bytes represents which data? I believe it's possible to see a decrypted data payload in the log?

vevsvevs, 2020-05-15:
I've collected some logs. Seems like the main encrypted payload has some structural differences.

Active scan events:
```
[03:22:20][D][xiaomi_ble:245]: decrypt_xiaomi_payload(): authenticated decryption failed.
[03:22:20][D][xiaomi_ble:246]:   MAC address : 50.EC.50.CD.32.02 (6)
[03:22:20][D][xiaomi_ble:247]:        Packet : 58.59.F6.07.33.02.32.CD.50.EC.50.C7.A4.16.1F.70.7F.63.00.00.8D.F9.45.23 (24)
[03:22:20][D][xiaomi_ble:248]:           Key : 71.A5.55.02.3F.95.D5.A6.2D.BD.BA.B3.CA.6B.C9.1D (16)
[03:22:20][D][xiaomi_ble:249]:            Iv : 02.32.CD.50.EC.50.F6.07.33.70.7F.63 (12)
[03:22:20][D][xiaomi_ble:250]:        Cipher : C7.A4.16.1F (4)
[03:22:20][D][xiaomi_ble:251]:           Tag : 00.00.8D.F9 (4)
[03:22:20][D][esp32_ble_tracker:544]: Found device 50:EC:50:CD:32:02 RSSI=-50
[03:22:20][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[03:22:25][D][xiaomi_ble:265]: decrypt_xiaomi_payload(): authenticated decryption passed.
[03:22:25][D][xiaomi_ble:267]:   Plaintext : 0A.10.01.64 (4), Packet : 53
[03:22:25][D][xiaomi_ble:279]: Got Xiaomi MJYD2S (50:EC:50:CD:32:02):
[03:22:25][D][xiaomi_ble:288]:   Battery Level: 100%
[03:22:25][D][sensor:092]: 'nlight_batt_bath': Sending state 100.00000 % with 0 decimals of accuracy
```

Motion-on events:
```
[03:25:06][D][xiaomi_ble:245]: decrypt_xiaomi_payload(): authenticated decryption failed.
[03:25:06][D][xiaomi_ble:246]:   MAC address : 50.EC.50.CD.32.02 (6)
[03:25:06][D][xiaomi_ble:247]:        Packet : 58.59.F6.07.39.02.32.CD.50.EC.50.CB.AC.FD.86.66.C9.63.00.00.79.A2.58.C5 (24)
[03:25:06][D][xiaomi_ble:248]:           Key : 71.A5.55.02.3F.95.D5.A6.2D.BD.BA.B3.CA.6B.C9.1D (16)
[03:25:06][D][xiaomi_ble:249]:            Iv : 02.32.CD.50.EC.50.F6.07.39.66.C9.63 (12)
[03:25:06][D][xiaomi_ble:250]:        Cipher : CB.AC.FD.86 (4)
[03:25:06][D][xiaomi_ble:251]:           Tag : 00.00.79.A2 (4)
[03:25:06][D][esp32_ble_tracker:544]: Found device 50:EC:50:CD:32:02 RSSI=-48
[03:25:06][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
```

Motion-off events:
```
[03:27:21][D][xiaomi_ble:245]: decrypt_xiaomi_payload(): authenticated decryption failed.
[03:27:21][D][xiaomi_ble:246]:   MAC address : 50.EC.50.CD.32.02 (6)
[03:27:21][D][xiaomi_ble:247]:        Packet : 58.59.F6.07.3E.02.32.CD.50.EC.50.58.4D.7A.A8.4D.E5.63.00.00.F2.09.3A.A5 (24)
[03:27:21][D][xiaomi_ble:248]:           Key : 71.A5.55.02.3F.95.D5.A6.2D.BD.BA.B3.CA.6B.C9.1D (16)
[03:27:21][D][xiaomi_ble:249]:            Iv : 02.32.CD.50.EC.50.F6.07.3E.4D.E5.63 (12)
[03:27:21][D][xiaomi_ble:250]:        Cipher : 58.4D.7A.A8 (4)
[03:27:21][D][xiaomi_ble:251]:           Tag : 00.00.F2.09 (4)
[03:27:25][D][xiaomi_ble:265]: decrypt_xiaomi_payload(): authenticated decryption passed.
[03:27:25][D][xiaomi_ble:267]:   Plaintext : 0A.10.01.64 (4), Packet : 64
[03:27:25][D][xiaomi_ble:279]: Got Xiaomi MJYD2S (50:EC:50:CD:32:02):
[03:27:25][D][xiaomi_ble:288]:   Battery Level: 100%
[03:27:25][D][sensor:092]: 'nlight_batt_bath': Sending state 100.00000 % with 0 decimals of accuracy
```

ahpohl, 2020-05-15:
@vevsvevs I added the illuminance value and pushed the update. Do you have a second sensor with MAC address 2F:FA:ED:4E:AC:96 configured? I think the 19 bytes long packet originates from a different sensor.

I would need to take a look at the full log. If you have only one sensor with MAC address 50:EC:50:CD:32:02 configured, then these packets should have been rejected before being passed to the decryption function. I enabled decryption of the 19 bytes long packet anyway just to see what it contains.

I found the config needs a dummy sensor entry otherwise it will fail to compile as this component is a hybrid between a "sensor" and a "binary sensor":
```
sensor:

binary_sensor:
  - platform: xiaomi_mjyd2s
    device_class: motion
    name: "MJYD2S Night Light"
    mac_address: "50:EC:50:CD:32:02"
    bindkey: "71a555023f95d5a62dbdbab3ca6bc91d"
    timeout: 5s
    illuminance:
      name: "MJYD2S Illuminance"
    battery_level:
      name: "MJYD2S Battery Level"
```
Do we actually need to give a timeout value in the config for this sensor (it originally comes from the mue4094rt night light code)?

vevsvevs, 2020-05-15:
@ahpohl do you use the Telegram?

ahpohl, 2020-05-15:
I've decrypted a couple of 19-byte long packets:
```
In-dark
48.59.F6.07.1E.96.AC.4E.ED.FA.2F.2E.4E.00.00.7F.63.4B.2F  Plaintext : 17.10.04.00.00.00.00 (7) 
48.59.F6.07.23.6F.54.F0.CD.70.AF.F6.4E.00.00.24.8C.25.C0  Plaintext : 17.10.04.78.00.00.00 (7)

In-bright
48.59.F6.07.26.03.1D.33.C5.8F.01.FC.4E.00.00.AD.5B.46.F7  Plaintext : 17.10.04.00.00.00.00 (7)

In-dark movement
48.59.F6.07.2E.7F.28.2B.9A.33.F3.D4.4E.00.00.C0.7E.BB.51  Plaintext : 17.10.04.00.00.00.00 (7)
48.59.F6.07.30.1D.12.24.A7.A9.92.D4.4E.00.00.FB.AE.D9.28  Plaintext : 17.10.04.78.00.00.00 (7)

New log
48.59.F6.07.41.DD.F2.DC.A1.D9.C0.39.67.00.00.D0.35.F2.8F  Plaintext : 17.10.04.00.00.00.00 (7)
48.59.F6.07.47.2C.53.C7.6B.5F.2D.32.67.00.00.73.76.C9.58  Plaintext : 17.10.04.78.00.00.00 (7)
```
Packets contain either a "0" or a "78". Could be "motion" or "no motion", depending on the time these packets have been sent.

@vevsvevs Any ideas?

vevsvevs, 2020-05-16:
@ahpohl I may only guess, but maybe these messages contain "seconds since last motion" value? If so, 0 obviously will be sent when motion detected at the first time, 120 - when the sleep-off timer is reached with no motion, and other values - when light being reactivated in some intermediate point (it has an adjustable light-off timer).

If it is possible to implement the live-decryption of these packets, I will reproduce different situations to check my guessings.

ahpohl, 2020-05-16:
@vevsvevs I am going to implement live decryption for these packets for you to test.

ahpohl, 2020-05-17:
I renamed the component and added an idle time since last motion:
```
  - platform: xiaomi_mjyd02yla
    device_class: motion
    name: "MJYD02YL-A Night Light"
    mac_address: "50:EC:50:CD:32:02"
    bindkey: "48403ebe2d385db8d0c187f81e62cb64"
    idle_time:
      name: "MJYD02YL-A Idle Time"
    illuminance:
      name: "MJYD02YL-A Illuminance"
    battery_level:
      name: "MJYD02YL-A Battery Level"
```
[image]

ahpohl, 2020-05-21:
@vevsvevs Thank you for contributing code and your vigorous testing efforts!

Code for supporting the MJYD02YL-A night light has been merged with [ahpohl/esphome:dev](https://github.com/ahpohl/esphome). For anyone who wants to try out this sensor, use this configuration (illuminance has been removed in favour of the light on/off status, default device class changed to motion):
```
sensor:

binary_sensor:
  - platform: xiaomi_mjyd02yla
    name: "MJYD02YL-A Night Light"
    mac_address: "50:EC:50:CD:32:02"
    bindkey: "48403ebe2d385db8d0c187f81e62cb64"
    idle_time:
      name: "MJYD02YL-A Idle Time"
    light:
      name: "MJYD02YL-A Light Status"
    battery_level:
      name: "MJYD02YL-A Battery Level"
```

leeuwa, 2020-10-29:
> @vevsvevs I pushed initial support for the MJYD2S nightlight to [ahpohl/esphome:xiaomi_mjyd2s](https://github.com/ahpohl/esphome/tree/xiaomi_mjyd2s). Did you already extract the bindkey?

How did you find the bindkey?

ramelito, 2020-12-31:
Hi, i am on ESPhome 1.15.3 and getting

[illuminance] is an invalid option for [binary_sensor.xiaomi_mjyd02yla]. Please check the indentation.

is illuminance supported in code? I see in the logs Yeelight is reporting Illuminance.

nagyrobi, 2022-07-05:
Implemented: https://esphome.io/components/sensor/xiaomi_ble.html#mjyd02yl-a

closed 2022-07-05 by nagyrobi