esphome/feature-requests #733: Local buffering of periodic sensor measurements during communication outages

heathkh, 2020-05-16:
**Describe the problem you have/What new integration you would like**

Currently, when the wifi network or the home assistant server is down, sensor measurements are lost.  I'd like if the ESP device could buffer periodic sensor measurements locally and submit them when it next has the ability to reach the home assistant server. 

**Please describe your use case for this integration and alternatives you've tried:**

I'm logging measurements of temperature from 10 devices once a minute for a long-term study of thermal properties of my home over the course of the year.  When there is a power outage at home or my server crashes, I lose important measurements and have gaps in my data.

**Additional context**

dougle03, 2020-05-16:
I'd like this too

ali-essam, 2024-11-11:
Another use case for this feature is to allow sensors to collect data while saving power by not connecting to wifi then periodically connect to wifi and update HA

Emmegiemme, 2024-11-29:
I would love to have it. 
WiFi is always unreliable and having a buffer for saving data, it will help to fill the gaps.

Floriszz, 2025-10-26:
It is quite an old issue.
I do see this; max_send_queue on esphome:
https://esphome.io/components/api/
But I'm not sure if this covers it.
Also I have read somewhere there is a parameter in esphome yaml config for it, but can not find it back.

Floriszz, 2025-10-30:
OK. 
Clear. That makes sense. Best way forward would be that the esp does record time stamps and use ntp to keep clock in sync. 
But in one case I use an esp for water meter and it just needs to register/count  pulses. So it could easily buffer the counts and send them through when connection is back. Good enough for a totalizer. OK, you will miss some real time flow during the network disconnect time. But the total will be good and averages can be calculated from it.