esphome/feature-requests #734: Light effect sync between multiple lights

Vxider, 2020-05-17:
**Describe the problem you have/What new integration you would like**
Can I sync the light effect between multiple lights?

**Please describe your use case for this integration and alternatives you've tried:**

I have 12V RGB trip with `RGB Light` components and ws2812B with `FastLED` components. Can I sync the light effect in two different kinds of lights? For example, if I use `Random effect`, the color should change synchronously. 

In my case, it can’t be controlled by one bus because they belong to different platforms. And for FastLED strips, sometime it needed to be displayed independently, so I cannot connect them as one device.

**Additional context**

oxan, 2021-06-04:
#1107 would also fix this (as you can then combine the lights into a single one using partitions).

MarshallAsch, 2021-11-27:
@oxan I dont think #1107 solves the problem, from what I understood from the light `partition` it will just create one long virtual strip. It will not allow you to apply the same effect to multiple strips. 

This is something I am also looking for. I have 2 led strips that I would like to be able to apply the _same_ effect to. Currently I need to set 2 `fastled_clockless` lights and then set both to show the same effect manually. 

I think this is also what is being requested in #585

oxan, 2021-11-27:
> @oxan I dont think #1107 solves the problem, from what I understood from the light `partition` it will just create one long virtual strip. It will not allow you to apply the same effect to multiple strips.

If you use non-addressable effects such as the random effect OP mentioned, each LED in the strip has the same color, so having them in one virtual strip achieves the same result. You're right that for addressable effects this doesn't cause them to work synchronously.

A `template` light would be the best, generic implementation for this, I think.