esphome/feature-requests #737: Improvements of pulse_counter component

m1ch, 2020-05-19:
**Describe the problem you have/What new integration you would like**

The pulse_counter component is lagging some features for my use case, that I also can't reproduce otherwise. 
These are the features that I have in mind:
* Implement a 'slow' mode for ESP32 where it behaves like an ESP8266
* Implement an option to send an MQTT message with the time in ms every time an impulse is detected. 
  * Alternatively send the count every given interval and a retaining zero after the first interval no impulse is detected. 

**Please describe your use case for this integration and alternatives you've tried:**

I have 12 S0 meters (10*Power and 2*Water-flow) for monitoring in my flat connected to an ESP32. [Wikipedia S0 (German)](https://de.wikipedia.org/wiki/S0-Schnittstelle) An S0 signal is just a 20 ms pulse every time a certain amount of energy, or water was used. I have a theoretic maximum of 3 pulses per second, but this would translate to a power consumption of 16A an all 3 phases (1imp/Wh). 
The pulse_counter component on ESP32 only allows me to monitor 8 inputs. 
I also would prefer to send the time in ms via MQTT each time a pulse was detected, or the count per second if there have been pulses and a retaining 0 in the first second no pulse was detected. This would lead to much fewer messages. 
Re-building the pulse_counter component by using only binary_sensor-gpio components could work somehow, but is rather tricky. 

**Additional context**

I like to start here a discussion on an update of the pulse_counter component and agree on the changes to be implemented. 
I am willed to do the update myself and create a pull request, once it is agreed how to implement the new features.

m1ch, 2020-05-25:
I have now a more, or less working solution using GPIO sensors, globals and timed lambdas:
https://github.com/m1ch/esphome-configs/blob/gpio-and-lambdas/powermeter_test.yaml

Still, I think it would be better to do it just with pulsecounters.

nagyrobi, 2022-07-07:
Any of these implemented since suit your request?
https://esphome.io/components/sensor/pulse_meter.html
https://esphome.io/components/sensor/pulse_width.html
https://esphome.io/components/sensor/pulse_counter.html