esphome/feature-requests #738: multiple VL53L0X sensors

OnlineGuz, 2020-05-20:
**Describe the problem you have/What new integration you would like**

I would like to support 2 or more VL53L0X sensors

**Please describe your use case for this integration and alternatives you've tried:**

There might be two way to do this:
1. **Put two (or more) VL53L0X sensors on the same i2c bus**: but currently this is not possible as they would have the same address. The VL53L0X supports address change, but this would require disabling one both sensors via the SXHUT pin, and then enabling one sensor at a time and changing its address before enabling the next one. Currently it looks like ESPHome does not support the XSHUT pin.
2. **Put two (or more) sensors on separate i2c buses**: I tried this but it seems like the current release doesn't support more than 1 instance of the sensor. I say this because when I try to read both sensors synchronously (same _update_interval_), I can only read one at a time, and when thry to read them asynchronously (different _update_interval_), they assume the same value even though they should not. Correct me if I am wrong.

**Additional context**

VL53L0X is not necessarily very precise but I see it is often used as a presence sensor or beam break sensor alternative.
I want to use two of them to make a "door direction sensor" (please have mercy, I don't know how they are called XD). Here is a "high quality" picture to explain what I want to do: after a person crosses the two sensors in one direction a message is sent to Home Assistant (or any automation hub) to notify that a person has "changed room".
[image]

This is my yaml: I made so that the sensor value is saved in a template sensor as a binary value. Sometimes the VL53L0X reads "0": in that case the lambda function does not update the value (I had issues with _return {};_)
```
esphome:
  name: door_sensor
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "SSID"
  password: "PASS"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Door Sensor Fallback Hotspot"
    password: "fbksfmcaOX2x"

web_server:
  
captive_portal:

# Enable logging
logger:
#  level: DEBUG
#  logs:
#    vl53l0x: ERROR
#    sensor: ERROR

# Enable Home Assistant API
api:

ota:

i2c:
   - id: bus_a
     sda: D2
     scl: D1
     scan: True
   - id: bus_b
     sda: D6
     scl: D7
     scan: True
     
sensor:
  - platform: template
    name: "Template Sensor 1"
    id: template_sens_1
    update_interval: 100ms
    on_value:
      - then:
        - logger.log:
            format: "sens1 = %d"
            args: ['id(template_sens_1).state']
            tag: 'AAA'
      
  - platform: template
    name: "Template Sensor 2"
    id: template_sens_2
    update_interval: 100ms
    on_value:
      - then:
        - logger.log:
            format: "sens2 = %d"
            args: ['id(template_sens_2).state']
            tag: 'BBB'
          
  - platform: vl53l0x
    name: "Sensor 1"
    id: sens_1
    i2c_id: bus_a
    address: 0x29
    update_interval: 70ms
    signal_rate_limit: 1
    on_value:
      - sensor.template.publish:
          id: template_sens_1
          state: !lambda |-
            if (id(sens_1).raw_state == 0) {
              return id(template_sens_1).state;
            } else if (id(sens_1).raw_state > 0) {
              return 1;
            } else {
              return 0;
            }
    
  - platform: vl53l0x
    name: "Sensor 2"
    id: sens_2
    i2c_id: bus_b
    address: 0x29
    update_interval: 90ms
    signal_rate_limit: 1
    on_value:
      - sensor.template.publish:
          id: template_sens_2
          state: !lambda |-
            if (id(sens_2).raw_state == 0) {
              return id(template_sens_2).state;
            } else if (id(sens_2).raw_state > 0) {
              return 1;
            } else {
              return 0;
            }
```
Thanks for all your great work! ;)

randybb, 2020-05-25:
Another nice thing you can do with two sensors that are in defined (needs to be configurable) distance apart is to measure speed (distance / (time_2 - time_1) ) and object length (speed x (time_1_off - time_1_on)). The same principle is used on roads where they can with induction loops measure car count, define their type based on length and their speed. Would be good to have all these values calculated within this component.

Svrbinek, 2020-06-01:
Hello. I also join this request. I built the stair lighting according to this pattern https://www.youtube.com/watch?v=X8vHWPlUJSw and connected it to WLED and its usermods starway wipe and programmed the rest in NodeRED. It is simply not possible to use these sensors because ESPHome does not yet support them as MULTI. I like to get involved and help with testing, etc. I'm learning to program in Python, but I'm at the beginning, so I'm not able to edit this.
@OnlineGuz I just want to write that you can't use two buses for ESP8266, because this HW only supports one bus. Therefore, it shows what it shows. If you want to try two buses, you have to use ESP32 and find out which pins the buses are on. I don't know from my head now. Then write feedback whether it then worked.

randybb, 2020-06-01:
The cleanest way is to use a GPIO expander as is written in their [design sheet](https://www.st.com/resource/en/application_note/dm00280486-using-multiple-vl53l0x-in-a-single-design-stmicroelectronics.pdf), connect XSHUT and GPIO1 (INT will be needed for precise time measurements) and then we can change its address. But it needs to be supported first.

OnlineGuz, 2020-06-05:
> @OnlineGuz I just want to write that you can't use two buses for ESP8266, because this HW only supports one bus. Therefore, it shows what it shows. If you want to try two buses, you have to use [...]

Ok, I thought there was some sort of "Soft I2C" that would allow multiple buses even on 8266. Maybe I will try an ESP32 as soon as have some spare time.
Also, the speed stuff is cool, but I'm afraid we would get pretty unreliable results even on a "High Speed Profile". But maybe I'm wrong. @randybb have you already tried some sort of speed measurement using VL53L0X without ESPHome?

randybb, 2020-06-05:
I have only one sensor, will order couple more.

rspaargaren, 2020-06-27:
Hi since you are owning the VL53L0X sensor are you able to make a final test for the PR: https://github.com/esphome/esphome/pull/1055. This is related to Feature request: https://github.com/esphome/feature-requests/issues/724.
I might be able to have a look at your feature request as well... Is there already existing arduino code enabling this feature request for the esp?

rradar, 2020-06-29:
I have two TOF sensors (VL53L0X) so if there is something to test please tell me :wink:

Svrbinek, 2020-06-29:
Hi, I can test in a week. I have enough sensors. What do you need to test specifically?

OnlineGuz, 2020-07-02:
> Hi since you are owning the VL53L0X sensor are you able to make a final test for the PR: [esphome/esphome#1055](https://github.com/esphome/esphome/pull/1055). This is related to Feature request: #724.
> I might be able to have a look at your feature request as well... Is there already existing arduino code enabling this feature request for the esp?

Hi, i tested the _long_range_ feature and I can confirm it works, as in here: https://github.com/esphome/esphome/pull/1055#pullrequestreview-438789029 .
With my module I can consistently get measurements up to ~1.25m

Here is a link for a code that uses 2 vl53l0x sensors. Unfortunately it uses the Adafruit library, and not the Pololu one:
https://github.com/adafruit/Adafruit_VL53L0X/tree/master/examples/vl53l0x_dual

For those who want to try the pip installation here is the command to install the test version:
`pip3 install https://github.com/rspaargaren/esphome/archive/VL53-long-range.zip`
and here is the file i tested:
[vl_test.zip](https://github.com/esphome/feature-requests/files/4866818/vl_test.zip)

Thanks,
keep up the good work

kkellner, 2020-07-10:
I coded a solution to allow for multiple vl53l0x devices on the same i2c bus.   I have tested it with two vl53l0x devices on an ESP8266.   I could use other testers to help verify its working.  I also implemented a timeout as the current sensor code would lock-up the startup in an infinite loop if it was unable to talk to the sensor.

Since all vl53l0x devices default to address 0x29 at power up, it's necessary to connect the XSHUT pin of the vl53l0x to a unique output pin of the ESP so that all vl53l0x devices can be shutdown, then one at a time enabled, so that the address can be changed from 0x29 to whatever address you have assigned.  You need 1 output pin on the ESP per vl53l0x device.

Here is the repo/branch with the enhancement: https://github.com/kkellner/esphome/tree/vl53l0x-change-address

This is the configuration tested on an D1Mini (ESP8266).  In this example the D7 pin of D1Mini is connected to XSHUT pin of vl53l0x device 1 and the D6 pin of D1Mini is connected to the XSHUT pin of vl53l0x device 2.

Note the new `enable_pin` and `timeout` conf properties.

```
sensor:
  - platform: vl53l0x
    name: "VL53L0x Distance1"
    id: distance1
    i2c_id: bus_a
    address: 0x41
    enable_pin: D7
    timeout: 500us
    update_interval: 2000ms

  - platform: vl53l0x
    name: "VL53L0x Distance2"
    id: distance2
    i2c_id: bus_a
    address: 0x42
    enable_pin: D6
    update_interval: 2000ms

i2c:
  sda: D1
  scl: D2
  scan: True
  id: bus_a
```

OnlineGuz, 2020-07-10:
I just made a little test with the new features, it look like it is working. I'm also using a D1 mini.

The only "issue" I found is when I was using D3 and D4 as enable pins: I think that when the codes wants to use GPIO0 very close to startup it freaks out and doesn't work. I don't know if GPIO0 is not meat to be used but that's it.

Also (maybe I'm wrong) I saw that the timeout defaults to 0ms. Does that mean that in practice the ESP tries to communicate with the sensor only once, or that it continuously tries to find it? Is there an "average" or minimum time that best to use?

I will keep tinkering with it and report any issue if found.

kkellner, 2020-07-10:
D1mini pins - https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/
D3 - Connected to FLASH button. 
D4 - Connected to D1Mini LED.

So, this means that when configured for use as an enable pin for vl53l0x -- it will "reset" the D1Mini when it changes D3 state.  D4 should work but might be odd since its connected to the on-board LED so I would avoid both of those pins.

As for timeout.  The 0us default (no timeout) is the current behavior of the esphome vl53l0x sensor code. This means that it will try indefinitely to talk to the vl53l0x sensor -- which might sound good however this means it will get stuck at startup in the setup method and effectively your esphome will never finish booting (e.g., no WiFi). I would like to make the default something more reasonable but I didn't want to effect the current behavior. In other words, with my updates, if you don't change anything in your yaml (e.g., add the enable_pin or timeout properties), your old code should run exactly as it did before.

@OnlineGuz Where you able to test with multiple vl53l0x sensors?

kkellner, 2020-07-10:
I just tested multi-vl53l0x support on ESP32 (using same i2c bus) and it looks good.  Here is the ESP32 config I used:

```
sensor:
  - platform: vl53l0x
    name: "distance1"
    id: distance1
    i2c_id: bus_a
    address: 0x41
    enable_pin: GPIO16
    timeout: 200us
    update_interval: 2000ms
    unit_of_measurement: "m"

  - platform: vl53l0x
    name: "distance2"
    id: distance2
    i2c_id: bus_a
    address: 0x42
    enable_pin: GPIO17
    timeout: 200us
    update_interval: 2000ms
    unit_of_measurement: "m"

i2c:
  - id: bus_a
    sda: GPIO33
    scl: GPIO25
    scan: True
```

kkellner, 2020-07-12:
PR has been submitted for support of multiple vl53l0x sensors on same i2c bus:  https://github.com/esphome/esphome/pull/1126

OnlineGuz, 2020-07-13:
> @OnlineGuz Where you able to test with multiple vl53l0x sensors?

The tests I did were with 2 sensors. I have 5 of them, I will try as soon as I can.

OnlineGuz, 2020-07-13:
Hi, just tested the code with five sensors. When everything is connected and working properly it works just fine (Hurray🥳):

[image]

Anyway, if I remove one sensor from the bus (in this example it's **Sensor 4**), it looks like the code continuously tries to interrogate it, even though the timeout is set to 500us:
[image]
I don't know if this is normal or it's an issue related to my code.

Also after a short time (no more than 2 minutes) the warnings change:
[image]
I tried switching the sensors position on my breadboard and that always happens (and always with address 0x41, but sometimes 0x45 appears), so I suppose it's not due to a sensor malfunctioning.

By the way here is the code I used:
[vl530lx_test.zip](https://github.com/esphome/feature-requests/files/4912017/vl530lx_test.zip)

If you need me do to some more tests please let me know.

kkellner, 2020-07-13:
@OnlineGuz  based on some discussion in the PR, I decided to make the default `10ms` on timeout vs infinite.  I also updated the code to add a missing `this->mark_failed();` when the sensor fails setup.   This should solve the NACK issue -- if the sensor is not available at startup (i.e., fails setup call), then it will not try to talk with it in the `loop()` method to get valid data.

Can you try your test again with the latest code?
https://github.com/kkellner/esphome/tree/vl53l0x-change-address

OnlineGuz, 2020-07-14:
Hi,
just tested the latest code. I can confirm that NACK issue after setup is solved.
One note that doesn't regard the code:
I have 2 different models of VL53l0x:
- 3 x CJVL53L0XV2
- 2 x GYVL53L0X

and I saw that the GYVL53L0X models are a little bit slower and unreliable. For example, after a while they both stop answering (I start getting NACKs on theirs addresses). This is an example of what I get:
[image]
where:
Sensor 1 --> address 0x41 --> CJVL53L0XV2
Sensor 2 --> address 0x42 --> CJVL53L0XV2 (not connected - setup failed as expected)
Sensor 3 --> address 0x43 --> GYVL53L0X
Sensor 4 --> address 0x44 --> GYVL53L0X
Sensor 5 --> address 0x45 --> CJVL53L0XV2

Have any one of you ever experience this behaviour?

kkellner, 2020-07-15:
@OnlineGuz have you seen this issue with GYVL53L0X on the old code base with just the single unit?  How long does it run before the NACK issue starts? 

As far as I can tell there is no `V2` of the sensor itself, so I think the `V2` refers to the carrier board.  I wonder if it has different pull-up resistors.

OnlineGuz, 2020-07-15:
> Also after a short time (no more than 2 minutes) the warnings change:
> [image]
> I tried switching the sensors position on my breadboard and that always happens (and always with address 0x41, but sometimes 0x45 appears), so I suppose it's not due to a sensor malfunctioning.

Yeah, I had the same issue here on the old code, I just did not realize it was due to the board until the "NACK issue" was solved. I moved the sensors around to verify that the problem was caused by the board and not the code, that's why the addresses are different in the latest comment.
I think that's just caused by a bad circuit that slows down the bus

To be clear, I'm referring to this as GYVL53L0X:
[image]

and this as CJVL53L0XV2:
[image]

The colours make no difference.

I don't own the Adafruit version, but I'm pretty sure it works just fine. Anyway I would suggest buying an CJVL53L0XV2 over GYVL53L0X, just for empirical reasons...

Svrbinek, 2020-08-08:
Hello. I would ask if it is possible for these sensors (I will have 4 used) to send output data only if motion is detected? So somehow set that if the scanning distance changes from 120cm to shorter by more than 20cm, an output signal will be sent. I want to use these sensors to illuminate the stairs and I need the sensors to respond immediately upon detection. If the test time is set, then there will be unnecessarily much data going or I may even miss the pass. My point is to adapt them so that they can work like here in this video https://www.youtube.com/watch?v=X8vHWPlUJSw Thank you for your reply.

rspaargaren, 2020-08-08:
> Hello. I would ask if it is possible for these sensors (I will have 4 used) to send output data only if motion is detected? So somehow set that if the scanning distance changes from 120cm to shorter [...]

Isn't an internal automation within the esp to trigger Boolean after a value fast enough? You can already program this in the yaml.

Svrbinek, 2020-08-08:
> > Ahoj. Chtěl bych se zeptat, zda je možné, aby tyto senzory (budu mít 4 použité) odesílat výstupní data, pouze pokud je detekován pohyb? Tak nějak nastavte, že pokud se skenovací vzdálenost změní ze 120 cm na kratší o více než 20 cm, bude vyslán výstupní signál. Chci tyto senzory použít k osvětlení schodů a potřebuji, aby senzory reagovaly okamžitě po detekci. Je-li testovací čas nastaven, bude zbytečně mnoho dat, nebo mi může chybět průchod. Mým cílem je přizpůsobit je tak, aby mohli fungovat jako v tomto videu https://www.youtube.com/watch?v=X8vHWPlUJSw Děkujeme za vaši odpověď.
> 
> Není interní automatizace v esp ke spuštění Boolean po hodnotě dostatečně rychle? To už můžete naprogramovat v yamlu.

I'm sorry, I probably wrote it wrong because I'm not a programmer. Maybe what I described can be done directly in ESP, but I don't know. However, as it now works, it sends too much data and unnecessarily via WIFI to HA. So if it works out so that I can modify the code in esphome to send only some kind of data, that was nice. Can you please advise me how to achieve this?

kkellner, 2020-08-08:
@Svrbinek Look at the documentation for `internal` https://esphome.io/components/sensor/index.html

Just mark all 4 sensors with `internal` then create another `binary_sensor` template that just `or` them all together.  Only the binary_sensor will be visible and sending data to HA and only when the VL53L0X falls within the range of the values specified in your template.

Something like this (untested so syntax may not be quite right):

```
binary_sensor:
  - platform: template
    id: detect_on_any_sensor
    name: "detect_on_any"
    lambda: !lambda |-
        return id(distance1).state < 1.0f ||
               id(distance2).state < 1.0f ||
               id(distance3).state < 1.0f ||
               id(distance4).state < 1.0f;
```

OnlineGuz, 2020-08-08:
Hi,
@Svrbinek take a look at my code if it can help: it does what the image in the first post describes.
[door_sensor_show.zip](https://github.com/esphome/feature-requests/files/5045609/door_sensor_show.zip)

The two sensors analog values (sens_1 and sens_2) are converted into boolean switches (switch_1 and switch_2). The sensors set each switch to 1 when the value is > 70cm and set them to 0 when the value is below 70cm or Out of Range.

The states of the switches are then interpreted by a _state_machine_ (the scheme is at the top of the code) and the template sensor _value_ is increased or decreased for each person that crosses the door.

Also the code sends an MQTT message when a person finishes crossing the door, and the information is used to turn on a light when a room contains at least 1 person, and off with no person.

As you can see, when I don't want a sensor to show up in the web page or mqtt, I just comment the "name" entry and they automatically become "internal".

Sorry if I was not very clear😅

Svrbinek, 2020-08-26:
Hi and thank you all for the answers. I had little time, so I didn't get to it until now. I already have the HW sensors ready and now I need to load the test KOD from github and also the code into ESPHOME. I'm so sorry, but I'm not sure what exactly I should download from github and whether it's stored in the CUSTOM_COMPONENTS folder. Thank you in advance for your willingness to advise.
Otherwise, I noticed that the sensors do not show exact distance values but are shifted by 5-10cm.

Svrbinek, 2020-08-26:
So I probably have everything uploaded, but when I try to save in ESPHOME, it reports an error, as if he didn't know the new features at all. Where can be the mistake?
[image]

kkellner, 2020-08-26:
The pull request has not been merged into any version of ESPHome yet.  https://github.com/esphome/esphome/pull/1126

You need to put the changed files for Vl53l0x into `custom_components/vl53l0x` directory to override the default version that does not have the feature.

Here are the files you need:  https://github.com/kkellner/esphome/tree/vl53l0x-change-address/esphome/components/vl53l0x

Here is the directory tree as it looks in Home Assistant:

[image]

Svrbinek, 2020-08-27:
> Žádost o vyžádání nebyla dosud sloučena do žádné verze ESPHome. [esphome / esphome # 1126](https://github.com/esphome/esphome/pull/1126)
> 
> Chcete- `custom_components/vl53l0x`li přepsat výchozí verzi, která tuto funkci nemá, musíte do adresáře umístit změněné soubory pro Vl53l0x .
> 
> Zde jsou soubory, které potřebujete: https://github.com/kkellner/esphome/tree/vl53l0x-change-address/esphome/components/vl53l0x
> 
> Zde je strom adresářů, jak vypadá v domovském pomocníkovi:
> 
> [image]

Hi, so exactly as you write I have it and anyway in the esphome there is a problem with the fact that he does not know the two new instructions enable_pin: D0 and timeout: 500us
I have everything uploaded in custom_components in the right folder, but even after restarting HA, nothing changes. I do not know what I am doing wrong.

kkellner, 2020-08-27:
Looks like its not finding the correct vl5310x sensor code.   From your screenshot above it shows `y:\custom_components\vl5310x`, I'm not sure where that is in relation to your esphome install.   Are you using Home Assistant's ESPHome add-on?   On my HA setup I have the normal `config` directory for HA, then the add-on uses `esphome` directory.  It is under the `esphome` directory that the `custom_components` should be placed.

[image]

Also, I noticed when I update the validation code, I need to refresh the browser window that has the esphome editor in order for the new validation to take effect.

Svrbinek, 2020-08-27:
The display you saw was via SAMBA in total commander, now it's straight from HA, but I think it's the way it's supposed to be and you're presenting.
[image]

Svrbinek, 2020-08-27:
I'm starting to suspect that there will be an error in the name and it is not clear whether it is vl53l0x or vl5310x or v15310x etc.

kkellner, 2020-08-27:
Nope, you have it in the wrong spot.  You have it under `\config\custom_components`.  That is where "Home Assistant" custom components go.   You need it in `\config\esphome\custom_components`.  Note the "esphome" in the directory tree.  That is where `esphome` custom components go.   As for the correct sensor name...   just use the name as it exists in github, which is `vl53l0x`  (cut and paste is your friend went it comes to ohs vs zeros and L vs ones).

Svrbinek, 2020-08-27:
Perfect, it works, all 4 sensors :-) Thanks a lot, I was really worried I didn't notice that there is another option custom_components. So I'm sorry for such clarity and harassment :-) Now I'm going to try to get the sensors to scan the stairs.

Svrbinek, 2020-08-27:
> dělá to, co popisu

Hi, so now I have the sensors running thanks to kkellner and now I'm going to start how to do what you have in that code, but I'll have a total of 4 sensors. Two will be downstairs and two upstairs, so it should be different. I programmed the logic with NodeRED, so it would be enough for each sensor to send information only when someone enters the stairs or leaves them. I'll take a look at it tomorrow, maybe it will work.

Svrbinek, 2020-08-28:
Hi, so I started working on it. Your kkellner code for a binary sensor works quite well, I just adjust it for each sensor separately. But I have a little problem with two things.
1. I need the sensors to detect even fast runs (if someone runs up the stairs quickly, some sensors may not respond and then everything is wrong), so ideally online monitoring.
2. to send data via wifi from esp8266 only when the motion detection condition is met. At the moment, I'm on an update interval of 250ms and it's a huge amount of unnecessary data.

When reduced to 50ms, the ESP8266 has already bitten. Probably not running wifi or memory.

I don't know if it's described clearly, but I'll add a description of what I want and ask if it's possible to do it somehow.

So there are a total of 4 sensors. 2 sensors on the stairs at the bottom and 2 at the top. Both pairs will be close together, only a few cm. This is so that it is possible to detect the direction of movement on the stairs and thus count the number of people on the stairs. I have already programmed it in nodered. Well, I need data from those sensors to go to HA only if someone passes. The staircase is about 120 cm long, so if the distance is below 1m so as to send from what sensor it was. Now I have reached the update interval of 250ms, but I also consider it insufficient, because with a very fast run on the stairs, the foot may miss this interval and it would all be poorly detected and automated. So I would need the sensors to monitor it continuously, ideally online or with 1ms but the data is only sent if the condition is met and the distance decreases below 1m

Svrbinek, 2020-08-28:
Done :-)

kkellner, 2020-08-28:
@Svrbinek: Does "Done" mean you have a solution?  

Based on the datasheet https://www.st.com/resource/en/datasheet/vl53l0x.pdf it looks like it takes 23ms to perform a mensurement -- so I wouldn't expect you to be able to go any faster then that for updates.

[image]

Also, in my experimenting, I found the esp8266 could not keep up with 4 sensors.  The symptom that it couldn't keep up was that Home Assistant would show the esp8266 as unavailable at least once every hour for a few seconds to a minute.   I've tested esp32 with 4 sensors and it works fine -- as the esp32 is much faster then esp8266.

As for updates -- mark all your vl53l0x sensors as internal so it doesn't send any network traffic on changes, then create a template sensor that detects what you want to send to HA.

Something like this 
```
...
  - platform: vl53l0x
    id: distance1
    name: "${friendly_name} distance1"
    internal: true
    i2c_id: bus_a
    address: 0x41
    enable_pin: D7
    update_interval: 300ms
    unit_of_measurement: "m"
    filters:
    - lambda: if (isnan(x) || x > 1.1) {  return 9999.0; } return x;      
    on_value:
      then:
      - lambda: |-
          if (id(distance1).state > 0.1 && id(distance1).state < id(1.0)) {
            id(detect1) = true;
          } else {
            id(detect1) = false;
          }

... (other vl53l0x sensors)

  - platform: template
    id: detect_any_sensor
    name: "${friendly_name} detect_any"
    lambda: !lambda |-
        return id(detect1) || id(detect2) || id(detect3) ||  || id(detect4);

```

Svrbinek, 2020-08-28:
Yes, I meant that I managed to get it in a bit functional condition. There is a bit of a problem with the fact that the sensor sometimes makes a fake signal on the sensor in the reflection, but I partially eliminated this in the delay off code with the filter. I still have the code created in nodered and when testing with real sensors it sometimes behaves strangely. I use a WLED system with 900 LED lights (15m).
ESP8266 I use with 4MB memory. When set to 50ms it was unusable, but then I tried to set the internal sensor as you type. Despite that, the problem was probably with flooded memory, so I changed the LOG from debugs to INFO and then it went great. I enclose my code. If it is not optimized (I am not a pro), feel free to write what can be improved. Otherwise, in case of interest, I can also send FLOW from NodeRED

i2c:
   - id: bus_a
     sda: D2
     scl: D1
     scan: True
     
sensor:
  - platform: vl53l0x
    name: "Sensor 1"
    id: sens_1
    i2c_id: bus_a
    address: 0x41
    update_interval: 50ms
    enable_pin: D0
    timeout: 500us
    internal: true
    
  - platform: vl53l0x
    name: "Sensor 2"
    id: sens_2
    i2c_id: bus_a
    address: 0x42
    update_interval: 50ms
    enable_pin: D5
    timeout: 500us
    internal: true
    
  - platform: vl53l0x
    name: "Sensor 3"
    id: sens_3
    i2c_id: bus_a
    address: 0x43
    update_interval: 50ms
    enable_pin: D6
    timeout: 500us
    internal: true
    
  - platform: vl53l0x
    name: "Sensor 4"
    id: sens_4
    i2c_id: bus_a
    address: 0x44
    update_interval: 50ms
    enable_pin: D7
    timeout: 500us
    internal: true
    
binary_sensor:
  - platform: template
    id: detect_on_any_sensor1
    name: "detect_on_any1"
    filters:
      - delayed_off: 100ms
    lambda: !lambda |-
        return id(sens_1).state < 1.0f;
               
  - platform: template
    id: detect_on_any_sensor2
    name: "detect_on_any2"
    filters:
      - delayed_off: 100ms
    lambda: !lambda |-
        return id(sens_2).state < 1.0f;
               
  - platform: template
    id: detect_on_any_sensor3
    name: "detect_on_any3"
    filters:
      - delayed_off: 100ms
    lambda: !lambda |-
        return id(sens_3).state < 1.0f;
               
  - platform: template
    id: detect_on_any_sensor4
    name: "detect_on_any4"
    filters:
      - delayed_off: 100ms
    lambda: !lambda |-
        return id(sens_4).state < 1.0f;

Svrbinek, 2020-10-07:
Does anyone please have experience with VL53L1X esphome sensors? Now two of them came to me I would try them. They have a faster sampling frequency and better resistance to UV interference. The VL53L0X switched on me in the house during the day, when the UV penetrated through the blinds into the room. But there were no direct sunbeams, only white light through the white blinds.

DutchDeffy, 2021-01-25:
> Does anyone please have experience with VL53L1X esphome sensors? Now two of them came to me I would try them. They have a faster sampling frequency and better resistance to UV interference. The [...]

I just started my very first ESPHome custom component with software to count the number of people in a room. I also ordered a set of VL53L1X. Here's the repo https://github.com/DutchDeffy/esphome_custom_component_VL53L1X_people_counter

kkellner, 2021-01-25:
@DutchDeffy FYI, this will work for your project, but it's not the correct way to handle i2c devices within ESPHome...  You are using the SparkFun library to handle your i2c communication.  That would directly conflict with ESPHome's i2c implementation.  ESPHome must "own" the i2c bus.   

As long as you are not using ESPHome's i2c implementation, then your solution will work, but that is a no-go for most of us and would not be accepted as a pull-request for ESPHome.

DutchDeffy, 2021-01-26:
@kkellner Thank you for pointing that out. Does that mean I would have to handle the i2c implementation comparable to the way it is done in this custom component? https://gist.github.com/mKeRix/4fc553574af0a2d8682734695160b859

kkellner, 2021-01-26:
@DutchDeffy That one doesn't look correct either.  Your class should use `i2c::I2CDevice`.  Best example would be to look at https://github.com/esphome/esphome/tree/dev/esphome/components/vl53l0x  or any of the i2c based components in https://github.com/esphome/esphome/tree/dev/esphome/components  (e.g., ina219, sht3xd, etc)

DutchDeffy, 2021-01-26:
Allright, despite it's the incorrect way of approaching this if it's ever to be merged to esphome. I will just try it out this way and see if I can get something working. Then, if I have more time on my hands, I might take a look at doing it the proper way.

Svrbinek, 2021-02-02:
Hi, so I'm writing after a long time. I already have the sensors and stairs done as I wished. Only one thing is a bit of a problem with the sensor. Because the two sensors are close to each other, it happens that during the passage I sometimes catch the second sensor earlier than the first, so that the stairs do not light up. This is because each sensor has its own timed scan frequency. It can also be seen on the IR camera, when it can be seen that the transmission beam of the two sensors is not synchronized but they flash differently. Therefore, it can happen that when I walk under the sensor, it is not in the cycle when he does not scan, but he scans the other and sees me earlier. Therefore, I would like to know if this could be somehow synchronized so that all sensors have the frequency timed the same. I have the scanning frequency set to 80ms. The solution could really be to use matrix sensors, ie the VL53L1X. Some solution?

pepe59, 2021-02-03:
So far, I am testing 2 sensors (detection of the passage to the kitchen). For now, I have them on separate nodemcu boards. How do you solve the counting of people? I used a counter in HA, but if it happens that the sensor skips, the logic stops and it is necessary to reset the counter manually.

kkellner, 2021-02-03:
I don't think you are using the correct sensor for your binary needs -- you are trying to detect if an object (person) passed by.  The VL53L0X is a distance measurement sensor which has minimum timing requirements to get that reading.  A fast moving person at the right time would not detect this.  You need an IR beam break type solution.  Once such solution is below.

https://www.inspectmygadgets.com/ir-beam-break-sensors-with-tasmota-and-home-assistant/

You can use ESPHome's binary sensor with the above (or any beam breaking sensor):
https://esphome.io/components/binary_sensor/index.html

For a "one sided" solution (where you don't need a transmitter and receiver side) that are also many options.  
Example:  https://www.amazon.com/dp/B0753Z5R3W

pepe59, 2021-02-11:
I have a VL53L0X + STM32 sensor, but I can't get it to work. The STM32 chip on this sensor may not be supported in ESP home. Is there a solution? Thank you

randybb, 2021-02-11:
@pepe59 STM32 provides UART (RX, TX) interface to the VL53L0X, so just ignore it and use only I2C (SDA, SCL) that is being supported.

Seba-VT, 2021-05-30:
@Svrbinek, How did you implemented the logic to detect the direction? I used Node-Red and get it working but I would like to run the logic inside the ESP32 and just receive the direction in to HA insted of the binary sensors data. Is this possible with ESPhome?

here my Node-Red flow in case someone finds it helpful.

`[{"id":"67487c43.97ecf4","type":"tab","label":"People Counter TEST","disabled":false,"info":""},{"id":"7cba67a5.05eb88","type":"change","z":"67487c43.97ecf4","name":"inside","rules":[{"t":"set","p":"topic","pt":"msg","to":"inside","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":510,"y":80,"wires":[["cfa760ae.b9ac6"]]},{"id":"c6315233.fe28","type":"change","z":"67487c43.97ecf4","name":"outside","rules":[{"t":"set","p":"topic","pt":"msg","to":"outside","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":520,"y":160,"wires":[["cfa760ae.b9ac6"]]},{"id":"51704eaa.0141e","type":"api-call-service","z":"67487c43.97ecf4","name":"","server":"13f57a0.20e1386","version":1,"debugenabled":false,"service_domain":"input_number","service":"increment","entityId":"input_number.people_in_lab","data":"","dataType":"jsonata","mergecontext":"","output_location":"","output_location_type":"none","mustacheAltTags":false,"x":1080,"y":80,"wires":[[]]},{"id":"42f045a6.dc79bc","type":"api-call-service","z":"67487c43.97ecf4","name":"","server":"13f57a0.20e1386","version":1,"debugenabled":false,"service_domain":"input_number","service":"decrement","entityId":"input_number.people_in_lab","data":"","dataType":"jsonata","mergecontext":"","output_location":"","output_location_type":"none","mustacheAltTags":false,"x":1090,"y":160,"wires":[[]]},{"id":"cfa760ae.b9ac6","type":"function","z":"67487c43.97ecf4","name":"","func":"//get stored values if present\ncontext.set(msg.topic,msg.payload);\n\nlet inside = context.get(\"inside\");\nlet outside = context.get(\"outside\");\n\n\n\nif(inside == false && outside == false){\n    msg.payload = 0;\n}\nelse if (inside == false && outside == true){\n    msg.payload = 1;\n}\nelse if (inside == true && outside == false){\n    msg.payload = 2;\n}\nelse {\n    msg.payload = 3;\n}\n\nreturn msg;\n\n","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":620,"y":120,"wires":[["e983029b.3566d","6cf27517.220dac"]]},{"id":"e983029b.3566d","type":"sequence-detector","z":"67487c43.97ecf4","name":"Sequence Detector","watch":"payload","sequence":"1\n3\n2\n0","timeout":"3000","matchMessage":"match","resetMessage":"reset","timeoutMessage":"timeout","x":810,"y":80,"wires":[["51704eaa.0141e"],[]]},{"id":"6cf27517.220dac","type":"sequence-detector","z":"67487c43.97ecf4","name":"Sequence Detector","watch":"payload","sequence":"2\n3\n1\n0","timeout":"30000","matchMessage":"match","resetMessage":"reset","timeoutMessage":"timeout","x":810,"y":160,"wires":[["42f045a6.dc79bc"],[]]},{"id":"b770922b.e3e7b","type":"server-state-changed","z":"67487c43.97ecf4","name":"","server":"13f57a0.20e1386","version":1,"exposeToHomeAssistant":false,"haConfig":[{"property":"name","value":""},{"property":"icon","value":""}],"entityidfilter":"binary_sensor.lab_inside_door_laser","entityidfiltertype":"exact","outputinitially":true,"state_type":"habool","haltifstate":"","halt_if_type":"num","halt_if_compare":"lt","outputs":1,"output_only_on_state_change":true,"for":0,"forType":"num","forUnits":"minutes","ignorePrevStateNull":false,"ignorePrevStateUnknown":false,"ignorePrevStateUnavailable":false,"ignoreCurrentStateUnknown":true,"ignoreCurrentStateUnavailable":false,"x":230,"y":80,"wires":[["7cba67a5.05eb88"]]},{"id":"683f32b5.f73d6c","type":"server-state-changed","z":"67487c43.97ecf4","name":"","server":"13f57a0.20e1386","version":1,"exposeToHomeAssistant":false,"haConfig":[{"property":"name","value":""},{"property":"icon","value":""}],"entityidfilter":"binary_sensor.lab_outside_door_laser","entityidfiltertype":"exact","outputinitially":true,"state_type":"habool","haltifstate":"","halt_if_type":"num","halt_if_compare":"lt","outputs":1,"output_only_on_state_change":true,"for":0,"forType":"num","forUnits":"minutes","ignorePrevStateNull":false,"ignorePrevStateUnknown":false,"ignorePrevStateUnavailable":false,"ignoreCurrentStateUnknown":true,"ignoreCurrentStateUnavailable":false,"x":240,"y":160,"wires":[["c6315233.fe28"]]},{"id":"13f57a0.20e1386","type":"server","name":"Home Assistant","legacy":false,"addon":true,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":true,"cacheJson":true}]`

Seba-VT, 2021-06-04:
I figured out how to make the ESP32 do the count instead.

Here the code for the curious ones: 

`esphome:
  name: lab_sensors
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: <deleted>
  password: <deleted>

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lab Sensors Fallback Hotspot"
    password: "YcSmQRKGWols"

captive_portal:

# Enable logging
logger:
  level: INFO

# Enable Home Assistant API
api:

ota:

i2c:
  sda: GPIO21
  scl: GPIO22
  scan: True
  id: bus_a

globals:
  - id: seq1
    type: int
    restore_value: no
    initial_value: '0'
  - id: seq2
    type: int
    restore_value: no
    initial_value: '0'
  - id: seq3
    type: int
    restore_value: no
    initial_value: '0'
  - id: seq4
    type: int
    restore_value: no
    initial_value: '0'
    
  - id: count
    type: int
    restore_value: no
    initial_value: '0'
    
  - id: inside_previous_value
    type: bool
    restore_value: no
    initial_value: 'false'
    
  - id: outside_previous_value
    type: bool
    restore_value: no
    initial_value: "false"

binary_sensor:
  - platform: template
    id: lab_inside_door_laser
    name: "Lab Inside Door Laser"
    lambda: !lambda |-
        if((id(sens_1).state < 0.7f) != id(inside_previous_value)){
          id(inside_previous_value) = (id(sens_1).state < 0.7f);
          if(((id(sens_1).state < 0.7f) == false) && ((id(sens_2).state < 0.7f) == false)){
            id(seq1) = id(seq2);
            id(seq2) = id(seq3);
            id(seq3) = id(seq4);
            id(seq4) = 0;
          }else if(((id(sens_1).state < 0.7f) == false) && ((id(sens_2).state < 0.7f) == true)){
            id(seq1) = id(seq2);
            id(seq2) = id(seq3);
            id(seq3) = id(seq4);
            id(seq4) = 1;
          }else if(((id(sens_1).state < 0.7f) == true) && ((id(sens_2).state < 0.7f) == false)){
            id(seq1) = id(seq2);
            id(seq2) = id(seq3);
            id(seq3) = id(seq4);
            id(seq4) = 2;
          }else if(((id(sens_1).state < 0.7f) == true) && ((id(sens_2).state < 0.7f) == true)){
            id(seq1) = id(seq2);
            id(seq2) = id(seq3);
            id(seq3) = id(seq4);
            id(seq4) = 3;
          }
          if((id(seq1) == 1) && (id(seq2) == 3) && (id(seq3) == 2) && (id(seq4) == 0)){
            id(count) ++;
            id(people_in_lab).publish_state(id(count));
          }else if((id(seq1) == 2) && (id(seq2) == 3) && (id(seq3) == 1) && (id(seq4) == 0)){
            id(count) --;
            if(id(count) < 0){
              id(count) = 0;
            }
            id(people_in_lab).publish_state(id(count));
          }
        }
        return id(sens_1).state < 0.7f;
        
        

  - platform: template
    id: lab_outside_door_laser
    name: "Lab Outside Door Laser"
    lambda: !lambda |-
        if((id(sens_2).state < 0.7f) != id(outside_previous_value)){
          id(outside_previous_value) = (id(sens_2).state < 0.7f);
          if(((id(sens_1).state < 0.7f) == false) && ((id(sens_2).state < 0.7f) == false)){
            id(seq1) = id(seq2);
            id(seq2) = id(seq3);
            id(seq3) = id(seq4);
            id(seq4) = 0;
          }else if(((id(sens_1).state < 0.7f) == false) && ((id(sens_2).state < 0.7f) == true)){
            id(seq1) = id(seq2);
            id(seq2) = id(seq3);
            id(seq3) = id(seq4);
            id(seq4) = 1;
          }else if(((id(sens_1).state < 0.7f) == true) && ((id(sens_2).state < 0.7f) == false)){
            id(seq1) = id(seq2);
            id(seq2) = id(seq3);
            id(seq3) = id(seq4);
            id(seq4) = 2;
          }else if(((id(sens_1).state < 0.7f) == true) && ((id(sens_2).state < 0.7f) == true)){
            id(seq1) = id(seq2);
            id(seq2) = id(seq3);
            id(seq3) = id(seq4);
            id(seq4) = 3;
          }
          if((id(seq1) == 1) && (id(seq2) == 3) && (id(seq3) == 2) && (id(seq4) == 0)){
            id(count)++;
            id(people_in_lab).publish_state(id(count));
          }else if((id(seq1) == 2) && (id(seq2) == 3) && (id(seq3) == 1) && (id(seq4) == 0)){
            id(count)--;
            if(id(count) < 0){
              id(count) = 0;
            }
            id(people_in_lab).publish_state(id(count));
          }
        }
        return id(sens_2).state < 0.7f;
        
        
  - platform: gpio
    pin: GPIO2
    name: "Lab PIR Sensor"
    device_class: motion

sensor:
  - platform: template
    id: people_in_lab
    name: "People in Lab"
    lambda: !lambda |-
      return id(count);
        

  - platform: vl53l0x
    name: "Sensor 1"
    id: sens_1
    i2c_id: bus_a
    address: 0x41
    update_interval: 50ms
    enable_pin: GPIO25
    timeout: 500us
    internal: true
    
    
  - platform: vl53l0x
    name: "Sensor 2"
    id: sens_2
    i2c_id: bus_a
    address: 0x42
    update_interval: 50ms
    enable_pin: GPIO32
    timeout: 500us
    internal: true`

Seba-VT, 2021-09-29:
@TheRealWaldo I do have my sensors working on the on the lastest version of ESPHome. did you set a different address for each sensor?

Here my yaml for the sensors setup:
```
sensor:
        

  - platform: vl53l0x
    name: "Sensor 1"
    id: sens_1
    i2c_id: bus_a
    address: 0x41
    update_interval: 50ms
    enable_pin: GPIO25
    timeout: 500us
    internal: true
    
    
  - platform: vl53l0x
    name: "Sensor 2"
    id: sens_2
    i2c_id: bus_a
    address: 0x42
    update_interval: 50ms
    enable_pin: GPIO32
    timeout: 500us
    internal: true
```
and don't forget the enable pins

TheRealWaldo, 2021-09-29:
I did; using precisely the same YAML (the only exception is enable_pins D6 and D7 as I'm using the D1 Mini), wired appropriately with pull-ups, I still only get Sensor 1 results.  The I2C scan shows two sensors, one at 0x41 and the other at 0x29 (which I am aware is the default), but attempts to access at 0x42 returning NACKs.

It seems as though the second sensor never gets assigned the new address.

I've verified the circuit using Arduino code but cannot get it functional with ESPHome.

Seba-VT, 2021-09-29:
Did you try using both sensors separately to make sure both units are good? can I see your YAML?

TheRealWaldo, 2021-09-29:
Yes, both work in ESPHome independently, and if you swap physically, the same results occur.

The following is from the most recent attempt:

```
i2c:
   scl: D1
   sda: D2
   scan: true
   id: bus_a

sensor:
   - platform: vl53l0x
     name: "Car 1 Distance"
     id: distance1
     address: 0x41
     update_interval: 5s
     long_range: true
     enable_pin: D6
     timeout: 500us
     i2c_id: bus_a

   - platform: vl53l0x
     name: "Car 2 Distance"
     id: distance2
     address: 0x42
     update_interval: 6s
     long_range: true
     enable_pin: D7
     i2c_id: bus_a
     timeout: 500us
```

Seba-VT, 2021-09-29:
I don't see anything wrong with your YAML. The only thing I can think of is the pull up you are using. In my case I didn't use any resistor, I connected the enable pins directly. Have you tried that?

TheRealWaldo, 2021-09-29:
Yes, without the pull-ups, even the first sensor doesn't consistently get assigned an address.  Your ESP or sensor board may have internal pull-up resistors, my boards do not, and specifications require pull-up resistors for the sensor and I2C. 

The circuit works well if I do not use ESPHome.  I guess I will have to dig deeper into the internals of ESPHome itself to see what's going on.

Seba-VT, 2021-09-30:
yeah... have you tried a different board? or with an ESP32? I'm using the ESP32 D1 Mini, and it's been running rock solid.