esphome/feature-requests #741: Support for GP2Y1010AU0F sharp dust sensor

theundefined, 2020-05-22:
**Describe the problem you have/What new integration you would like**
Support for cheap GP2Y1010AU0F dust sharp sensor.

**Please describe your use case for this integration and alternatives you've tried:**
Migrate from espeasy :-) https://www.letscontrolit.com/wiki/index.php?title=GP2Y10

andriuspetrauskis, 2020-06-29:
I have this sensor and so I would love to see it integrated :)

hypercube33, 2020-07-28:
I have its brother, the GP2Y1014AU0F

maskaz, 2020-10-01:
Hi, i had create custom Esphome sensor using Sharp dust sensor, i hope it will be useful.

https://github.com/maskaz/Esphome_custom_dust_sensor

malfroid, 2022-02-04:
I also created a custom sensor based on the work of @maskaz for my keyestudio Ks0196 sensor (contains Sharp GP2Y1010AU0F):

https://github.com/malfroid/esphome-KS0196

closed 2022-07-05 by nagyrobi