esphome/feature-requests #744: ESP32 internal temperature sensor

Semmu, 2020-05-24:
Hi!

First of all, thanks for this awesome project, I was genuinely surprised how easy and straightforward using this is! What an awesome DIY experience!

**Describe the problem you have/What new integration you would like**
I would love to be able to get the internal CPU temperature sensor value from an ESP32. I want to use it to track it's performance and load under different circumstances.

According to this (and some other) posts, it should be possible to get the value, because there is a function for it defined in an other lib: https://www.esp32.com/viewtopic.php?p=9286&sid=41486dfeb69cd567659dc070219b6dfe#p9286
But I'm not sure if ESPHome is using it and the library is linked. When I tried to use it in a sensor lambda, the linker could not find the implementation for it.

**Please describe your use case for this integration and alternatives you've tried:**
I have an ESP32 with a camera and I want to test different image qualities and framerates to find the optimal setup and also figure out what is the bottleneck (network traffic or CPU thermal throttling, etc.)

**Additional context**

marrold, 2020-05-24:
The Temperature sensor is only accessible in older ESP32 hardware as far as I know

prageeth, 2020-12-13:
I agree, this is an awesome project. I used to write everything from scratch using the Arduino IDE, and ESP home has made it so much better. Thanks again for this awesome project!

Now back to topic, I too have a use case where I need to read the internal CPU temperature. 

I have also read somewhere that not all EPS boards support this. So perhaps, we should have a component that can read the temperature, if the device supports it. 

Here's also an example of an Arduino sketch that I've used in the past: 
https://github.com/pcbreflux/espressif/blob/master/esp32/arduino/sketchbook/ESP32_int_temp_sensor/ESP32_int_temp_sensor.ino

Any help on this much appreciated!

Semmu, 2020-12-14:
i guess the hard part is detecting if the target board supports it or not, i'm not sure how that can be done

prageeth, 2020-12-15:
I'm not exactly sure how the components work in esphome (I haven't had the chance to dig in at the moment), but I assume it's using CPP native code to talk to the hardware.

A possible way would be that a simple "mock" is used when the native function (e.g. `temprature_sens_read` doesn't exist). With cpp, you can use `#ifndef` to do that. I'm not sure how this will work with the esphome architecture, but it's just an idea.

saso5, 2021-01-04:
Already address in https://github.com/espressif/arduino-esp32/issues/2422. I think this can be closed also.

xcray, 2022-06-25:
tasmota firmware supports this sensor quite well

ssieb, 2022-06-26:
If you want to test it, you can add this to your config:
```yaml
sensor:
  - platform: template
    id: temp
    lambda: return temperatureRead();
```
See what you get.

0neday, 2022-06-26:
> 

not support esp8266

```
Compiling .pioenvs/gas/src/main.cpp.o
/config/gas.yaml: In lambda function:
/config/gas.yaml:122:14: error: 'temperatureRead' was not declared in this scope
  122 |     lambda: return temperatureRead();
      |              ^~~~~~~~~~~~~~~
*** [.pioenvs/gas/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.66 seconds ==========================
```

jprouty, 2022-09-22:
I would like to petition that this issue be reopened (or maybe a new FR?) for an out of the box die temperature readout for the officially `Supported Targets | ESP32-C2 | ESP32-C3 | ESP32-S2 | ESP32-S3`. Official [ESP32C3 documentation is here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/temp_sensor.html#:~:text=The%20ESP32%2DC3%20has%20a,ranges%20with%20specific%20measurement%20errors.) with a [generalized example for all supported targets](https://github.com/espressif/esp-idf/tree/09f7589ef2/examples/peripherals/temp_sensor). An FYI: The driver api is changing in v5 of esp-idf.

Here is a working proof of concept custom driver I created following the [ESPHome Custom Sensor](https://esphome.io/components/sensor/custom.html) tutorial:
```
// This is it until v5.*
// https://github.com/espressif/esp-idf/blob/release/v4.4/components/driver/esp32c3/include/driver/temp_sensor.h
#include <driver/temp_sensor.h>

// Would preferably like to use the v5 latest temperature_sensor api:
// https://github.com/espressif/esp-idf/blob/09f7589ef2/components/driver/include/driver/temperature_sensor.h
// Example:
// https://github.com/espressif/esp-idf/tree/09f7589ef2/examples/peripherals/temp_sensor
// #include <driver/temperature_sensor.h>

#include "esphome.h"
class ESP32DieTemperatureSensor : public PollingComponent, public Sensor {
 public:
  ESP32DieTemperatureSensor() : PollingComponent(15000) {}

  void setup() override {
    ESP_LOGD("ESP32DieTemperatureSensor", "Configuring ESP32 to read die temperatures."); 
    ESP_ERROR_CHECK(temp_sensor_set_config(TSENS_CONFIG_DEFAULT()));

    // Enable and start the temperature sensor
    ESP_ERROR_CHECK(temp_sensor_start());
    ESP_LOGD("ESP32DieTemperatureSensor", "Done configuring.");
  }

  void update() override {
    float reading;
    ESP_ERROR_CHECK(temp_sensor_read_celsius(&reading));

    ESP_LOGD("ESP32DieTemperatureSensor", "Publishing temperature reading: %f", reading);
    publish_state(reading);
  }
};
```

nagyrobi, 2022-09-22:
Isn't @ssieb's example above working on ESP32-C2 | ESP32-C3 | ESP32-S2 | ESP32-S3? On plan old ESP32 it works.

jprouty, 2022-09-22:
Yes, on arduino core, @ssieb's example ought to continue to work as [per source here](https://github.com/espressif/arduino-esp32/blob/1a7962ece8a4c6ffa1d64c5a86ed8ee58dde10ba/cores/esp32/esp32-hal-misc.c#L59-L77). "Temp sensor" has also been recognized as supported for arduino but with no formal reference given for this peripheral. [See doc](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/libraries.html?highlight=temp#supported-peripherals) and [doc](https://docs.espressif.com/_/downloads/arduino-esp32/en/latest/pdf/). 

I am using plain esp-idf (no need for arduino, OTA is about 10x faster, and stability has been better on the new boards). I attempted the same code from my example in lambdas but ran into difficulty when it came to including `<temp_sensor.h>` from the `includes:` directive. 

Regardless of platform used, having a ready made Sensor component for people to discover and use from the official list means more people will find use for it. ie, "ESP32 Temp" or "ESP Die Temp" in the [Environmental section](https://esphome.io/#environmental) versus having to know they want the die temp, doing a web search for it, finding a thread like this, and then implementing a workaround.

nagyrobi, 2022-09-22:
So this request can re-qualify as _please add this to ESP-IDF platform in ESPHome_.

kquinsland, 2022-10-06:
> So this request can re-qualify as _please add this to ESP-IDF platform in ESPHome_.

Not necessarily. There's a way to do this in both platforms and the hardware. This isn't the first ESPHome component that would come with a `WARN:` in the docs about the platform and hardware making a difference here. E.g.: webserver only works on arduino and only ESP32 have dedicated PWM ... etc.

> Regardless of platform used, having a ready made Sensor component for people to discover and use from the official list means more people will find use for it. ie, "ESP32 Temp" or "ESP Die Temp" in [...]

I am _exactly_ in this boat ^, The only reason I know that (at least some) ESP32 devices have a built in temperature sensor is because that is automatically shown to the user when they flash Tasmota32. I got curious about why ESPhome didn't seem to have support for it because I am curious about some ventilation on an enclosure with an ESP32 and some _warm_ power electronics.

I would greatly appreciate a simple `platform: esp32_internal_temp` sensor that set up the metadata (`{device,state}_class`, `entity_category` ... etc) for me and was smart enough to use the right API behind the scenes / emit a `ERROR: $thisHardware does not support die temp sensor` line somewhere in the compile or run if the chip targeted isn't supported.

Electronlibre2012, 2023-01-08:
> https://github.com/espressif/esp-idf/blob/release/v4.4/components/driver/esp32c3/include/driver/temp_sensor.h

Hello,

someone could give the complete config please?

1-Config into Esphome Node like :
```
esp32:
  board: esp-wrover-kit
  framework:
    type: arduino
```
i guess

because i have these errors : 

```
Compiling /data/capteur-salon2/.pioenvs/capteur-salon2/src/main.cpp.o
Linking /data/capteur-salon2/.pioenvs/capteur-salon2/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/capteur-salon2/.pioenvs/capteur-salon2/src/main.cpp.o:(.literal._ZN24ESP32InternalTemperature6updateEv[ESP32InternalTemperature::update()]+0xc): undefined reference to `temp_sensor_read_celsius'
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/capteur-salon2/.pioenvs/capteur-salon2/src/main.cpp.o:(.literal._ZN24ESP32InternalTemperature5setupEv[ESP32InternalTemperature::setup()]+0x14): undefined reference to `temp_sensor_set_config'
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/capteur-salon2/.pioenvs/capteur-salon2/src/main.cpp.o:(.literal._ZN24ESP32InternalTemperature5setupEv[ESP32InternalTemperature::setup()]+0x18): undefined reference to `temp_sensor_start'
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/capteur-salon2/.pioenvs/capteur-salon2/src/main.cpp.o: in function `ESP32InternalTemperature::update()':
/config/esphome/.esphome/build/capteur-salon2/src/driver_temp_esp32.h:17: undefined reference to `temp_sensor_read_celsius'
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/capteur-salon2/.pioenvs/capteur-salon2/src/main.cpp.o: in function `ESP32InternalTemperature::setup()':
/config/esphome/.esphome/build/capteur-salon2/src/driver_temp_esp32.h:11: undefined reference to `temp_sensor_set_config'
/data/cache/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/esphome/.esphome/build/capteur-salon2/src/driver_temp_esp32.h:12: undefined reference to `temp_sensor_start'
collect2: error: ld returned 1 exit status
*** [/data/capteur-salon2/.pioenvs/capteur-salon2/firmware.elf] Error 1
```

thanks

nagyrobi, 2023-01-19:
Please ask such direct support questions our team on Discord plaform.

closed 2023-02-20 by jesserockz