esphome/feature-requests #745: VL6180

Highland7471, 2020-05-25:
**Describe the problem you have/What new integration you would like**
Integration of the VL6180 distance sensor. 

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**
https://www.sparkfun.com/products/12785

exxamalte, 2020-06-02:
Here is some custom component code that might help: https://github.com/exxamalte/esphome-customisations/tree/master/vl6180x

Highland7471, 2020-06-04:
Thank you. I will test it as sone as possible.

Highland7471, 2020-07-05:
Hellow

Finaly got my sensor. I have a problem ven i try to run with esphome.

In file included from .piolibdeps/vaxthus/Adafruit BusIO_ID6214/Adafruit_BusIO_Register.h:2:0,
                 from .piolibdeps/vaxthus/Adafruit BusIO_ID6214/Adafruit_BusIO_Register.cpp:1:
.piolibdeps/vaxthus/Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

compilation terminated.
*** [.pioenvs/vaxthus/libe7a/Adafruit BusIO_ID6214/Adafruit_BusIO_Register.cpp.o] Error 1
==================================================================== [FAILED] Took 31.70 seconds =============

Highland7471, 2020-07-06:
Solved it by adding #include <SPI.h> to vl6180x_sensor.h

Highland7471, 2021-04-18:
Hello

The sensor shows 0.0 lux all the time. Distans i working.

DidibabaQiu, 2021-12-06:
> Hello
> 
> The sensor shows 0.0 lux all the time. Distans i working.

Dear sir, 

Did you finally solve it?

if you did, could you share your epshome yaml? i try and try but no luck, i want to use vl6180x sensor to replace my HC-SR04 to detect my water tank

MichielioZ, 2022-01-26:
> The sensor shows 0.0 lux all the time. Distans i working.

A lot of these sensor boards only have the vl6180 chip (even if the board shows vl6180x), which only measures distance.
The vl6180x chip (so with the "x") also measures luminescence, but I haven't seen one in the wild yet...

closed 2022-07-01 by nagyrobi

ssieb, 2023-12-03:
An external component is nice, but not a reason to close a feature request, unless the feature wouldn't be accepted.

NonaSuomy, 2023-12-03:
This https://github.com/esphome/esphome/tree/dev/esphome/components/vl53l0x being based on https://github.com/pololu/vl53l0x-arduino

They have similar libraries for the other units as well

https://github.com/pololu/vl6180x-arduino

https://github.com/pololu/vl53l1x-arduino

Maybe if similar enough there could be a unified component for the units. Just to get basic distance then if others want the expanded features of the other units future additions could be added. 

Seems they are based on an API 

vl6180x https://www.st.com/en/embedded-software/stsw-img003.html
vl6180v1 https://www.st.com/en/embedded-software/stsw-img011.html

vl53l0x https://www.st.com/en/embedded-software/stsw-img005.html

vl53l1x https://www.st.com/en/embedded-software/stsw-img007.html
Ultra light api https://www.st.com/en/embedded-software/stsw-img009.html

These others show on Pololu with this warning and no supported code by them. There is however an Ultra Light Driver for the rest of them and may work ok on stuff like the ESP32 etc.

```
Note: this product is not recommended for use with 8-bit microcontrollers. Initializing the VL53L5CX and processing its readings require a significant amount of RAM and code space, making this sensor impractical for use with a typical 8-bit microcontroller. (ST’s API for the VL53L5CX typically uses over 90 KB of program memory.) For alternatives that are simpler to use and can work with 8-bit microcontrollers (but do not have the multi-zone or multi-target capabilities of the VL53L5CX), please consider the [VL53L1X](https://www.pololu.com/product/3415), [VL53L0X carrier](https://www.pololu.com/product/2490), or [VL6180X carrier](https://www.pololu.com/product/2489).
```

vl53l3cx https://www.st.com/en/embedded-software/stsw-img015.html

vl53L4cx https://www.st.com/en/embedded-software/stsw-img029.html

vl53l5cx https://www.st.com/en/embedded-software/stsw-img023.html

vl53l7-8cx https://www.st.com/en/embedded-software/stsw-img036.html

mikulik86, 2024-01-26:
I second this request.

NonaSuomy, 2024-05-30:
We just built a component for this but my sensor is a vl6180 (non x) anyone have a confirmed X version with the ALS (Ambient Light Sensor)?