esphome/feature-requests #746: New trigger on_connect and on_disconnect

m1ch, 2020-05-25:
**Describe the problem you have/What new integration you would like**

I would like to run some automation when MQTT, WIFI or Home-assistant is [dis-]connected. 

**Please describe your use case for this integration and alternatives you've tried:**

I am sending several measurements with a single MQTT transmission. To get them to be automatic recognized in Home-assistant I need to send the configuration MQTT messages. 

At the moment I am polling if the MQTT connection was lost and when it is back up. It would be easier to have a trigger to determine to send the update. 

I also think that there are lots of other applications for this feature. 

**Additional context**

theOzzieRat, 2021-05-19:
This is something that would be really useful. It would be great to be able to push updated sensor data when a new connection is made. I am personally more interested in a trigger when an API connection is made, but mqtt would be useful as well (mostly for devices that need to deep sleep and therefore rely on mqtt).

escoand, 2022-03-31:
You could use this work around:

```yaml
esphome:
  on_boot:
    - mqtt.publish:
        topic: /myesp/reconnect
        payload: reconnect
        retain: true

# recreate reconnect trigger regularly
# (when broker was restarted without persisted messages)
interval:
  - interval: 1h
    then:
      - mqtt.publish:
          topic: /myesp/reconnect
          payload: reconnect
          retain: true

mqtt:
  on_message:
    topic: /myesp/reconnect
    then:
      # do your reconnect stuff
```

nagyrobi, 2022-07-06:
Implemented in PR above.

closed 2022-07-06 by nagyrobi