esphome/feature-requests #751: [Hassio Add-on] Ability to filter by online/offline/update available status

santibur06, 2020-05-28:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

In the hassio add-on, it'd be great to have a dropdown to filter devices based on their status, or show offline devices first, update available next and then all online.

This could also be 'fixed' if the size of each row/device was smaller and maybe options (such as upload, show logs, etc) were put in a contextual menu. This way you'd be able to see a lot more devices without needing to scroll and be able to identify any issues at first glance.

However, I believe the best way to do this is by adding that filter, and, the smaller rows would also be great.

**Additional context**

ariedeleen, 2020-06-19:
That would be a nice feature.
Now if for example the connection is lost to Home Assistant, ESPHome the bulk update starts 
all over again after reconnecting. Even with the components that already are updated. And not on-line. When using the Update All button in the top right menu.

poldim, 2020-08-13:
This title should be updated to remove the Hassio Addon portion.  This change would happen in the dashboard UI and would be useful to all ESPHome users.