esphome/feature-requests #752: support for climate preset_modes

blademckain, 2020-05-29:
**Describe the problem you have/What new integration you would like**
I have seen that support for fan_modes and swing_modes has been added but support for preset_modes for the climate component is still missing

**Please describe your use case for this integration and alternatives you've tried:**
in this way the climate component would be more compatible with [HA's Climate Entity component](https://developers.home-assistant.io/docs/core/entity/climate/)

**Additional context**
today the preset away is treated as a true or false Boolean but from version 0.96 of HA, the correct method to set the preset away is through the support of the presets (ECO,AWAY,BOOST,COMFORT,HOME,SLEEP,ACTIVITY)

blademckain, 2020-06-03:
@glmnet 
could you add this features to [your changes on the climate component](https://github.com/esphome/esphome-docs/pull/415)?

fraschizzato, 2020-11-24:
Same here, trying to add support for eco and turbo mode on my AC, all works except the ability to use home-assistant presets

fraschizzato, 2020-11-26:
Had to be changed: EspHome Home Assistant Component, Aioesphomeapi, EspHome and components that are now using "away". 
Some hint can be found here:
https://github.com/fraschizzato/aioesphomeapi/commit/586e73a073391df3d65e792b977b3e521963d73c
https://github.com/fraschizzato/ha-esphome-climate
https://github.com/fraschizzato/esphome/commit/508fd293d44575176b1ea32f3c4ae0b2357bb2f4
Latest two have to be changed according to the first, are only an half way to test logic

Dennis-Q, 2020-11-28:
Would be nice if this will be available, allowing modes as HI-Power and Eco.

nagyrobi, 2022-07-01:
Implemented: https://esphome.io/components/climate/thermostat.html#presets

closed 2022-07-01 by nagyrobi