esphome/feature-requests #754: Anemometer sensor Ventus WS132

Wonkavator, 2020-05-30:
**Describe the problem you have/What new integration you would like**
I would love to integrate the Anemometer sensor Ventus WS132. As its not so pricy.
It has sensors for wind speed, wind direction, temparature and humidity.

**Please describe your use case for this integration and alternatives you've tried:**
Simple integration with a public scipt, bound to Mqtt :(
https://gist.github.com/micw/098709efc83a9d9ebf16d14cea4ca38e

**Additional context**
The script handels the data for winddirection and speed. Which would be absolutely fine.

kvvoff, 2020-06-02:
Sounds great.  Where did you buy this anemometer?  I can not find

Wonkavator, 2020-06-02:
Probably just the German market has it?

- [amazon](https://www.amazon.de/Ventus-Ersatz-Windmesser-W132-Funk-Wetterstation-W155/dp/B01NAM1E21)

- [reichelt (English translations)](https://www.reichelt.de/spare-wind-gauge-for-ventus-wireless-weather-station-ventus-w132-p132595.html?GROUPID=5992&START=0&OFFSET=16&SID=94XtZyyawQAT8AAF2jA4U7ef9f2e19bcb227da6922de3837d817b&LANGUAGE=EN&&r=1)

- [ebay](https://www.ebay.de/sch/i.html?_from=R40&_trksid=m570.l1313&_nkw=Ventus+W+132&_sacat=0&LH_TitleDesc=0&gclid=EAIaIQobChMIhK2n2rrj6QIVDrh3Ch04pgrGEAAYAiAAEgL6AfD_BwE&matchtype=b&adgroupid=77784020053&geo_id=33421&MT_ID=585306&cmpgn=6500825034&crlp=433288817424_&keyword=ventus+w&rlsatarget=kwd-11635218330&abcId=1141466&loc=1004363&sitelnk=&norover=1&mkrid=707-156513-467738-8&network=s&poi=&mkevt=1&adpos=&mktype=&_odkw=Ventus+W&device=c&mkcid=2)

nagyrobi, 2024-05-19:
For the Aliexpress model here's the config: https://github.com/esphome/feature-requests/issues/1103#issuecomment-2119325838